Marzogh/SPIMemory #7: Missing CLK Pulses

SFyffe, 2015-06-22:
Hello,

With a Salea Logic 4 analyzer I was able to determine that I was missing the clock pulses when trying to read\write to my Windbond SPI memory chip.

I have this connected to the SPI bus of my ATMEGA644A microcontroller.

Through further testing I figure out that if I swapped my crystal from 8MHz to 16MHz I can get the clock pulses to generate and I am able to work with the memory chip.

According to my chips datasheet it seems like 8MHZ would be just fine all the way up to 100MHz.

What could be reasons why an 8MHz crystal does not work with the demo program?

Thanks,

Stephen

Marzogh, 2015-06-22:
I suspect that might be because you are running at 8MHz. The SPI Clock prescaler sets the clock divider to SPI_CLOCK_DIV4 by default. My library is set to run on chips set to 16MHz - which when divided by 4 works out to 4MHz. In your case it would work out to 2MHz which is probably why you have missing clock pulses.

Try calling the following function after the constructor:
```SPI.setClockDivider(SPI_CLOCK_DIV2);```

That should provide a temporary workaround. I'll get around to making the library AVR chip agnostic ASAP. :)

SFyffe, 2015-06-22:
Hello,

I am not seeing the function to call you mention in your comment...

Thank You for replying.

Marzogh, 2015-06-22:
For further information on the SPI clockDivider: http://www.arduino.cc/en/Reference/SPISetClockDivider

Marzogh, 2015-06-22:
Try refreshing the page. My Markdown syntax was off earlier. I've just fixed it. :)

SFyffe, 2015-06-22:
Ok, interesting outcome.  I took one of my 16MHz boards and divided down by 2, 4, 8, and even 16.   All dividers seemed to work.   I could at least read the JEDEC anyway.   About the only thing it did do was alter the baud rate of my serial monitor.   For example, I divided SPI by 16 and instead of 9600 baud I had to set my serial monitor to 19200.   This test makes me feel a little bit better about my 8MHz crystal.

Any other thoughts?

Stephen

Marzogh, 2015-06-22:
That's interesting. Well, I learnt something new about SPI clocks and Serial baud rate  today. :)

How many clock pulses are you missing? Could you post a screen grab from your Logic Analyser results here so I can see what's happening?

I don't have a 644 so I'll test the library on a 328p and a 1284p at 8MHz once I get home from work and check what's happening.

Out of curiosity, which Arduino core are you using for your 644? And which version of the IDE?

SFyffe, 2015-06-23:
I may be a little better off with the clock divider set on the 8MHz board than I thought based on the screenshot of my analyzer:

[image]

This is the first time I have seen the clock pulses using this board setup.   I am missing now the MISO signal though...

I am using Arduino 1.6.4 The Core is 1.6.7

Stephen

Marzogh, 2015-06-23:
I've just been going through the 644 datasheet over lunch and I can't find a reason why this should be happening. The CS, clock and MOSI signal look like they should. Its pretty odd that the MISO signal is non-existent. 

Have you tried using a different flash chip?

I don't know if this has anything to do with it, but if you're using the Sanguino board-definitions for the 644, could you check your pins_arduino.h? The Sanguino definitions that I use have D10 mapped to RX1/INT0. This might cause problems. Have you tried mapping the cs pin to any pin other than 10? (That's what it is mapped to by default in my test code).

SFyffe, 2015-06-23:
Hi,

I had re-defined your chip select pin in the code to match the one I used on my controller.

Good news though.  Looks like there was something wrong with the memory chip.  I replaced it this afternoon and I now have all signals and things look to be working on the 8MHz board.

[image]

I will keep testing and see how things work this week.   Thanks for all your help.

Marzogh, 2015-06-23:
That's fantastic. I'm glad I could help. :)

Cheers,

Praj

closed 2015-06-23 by Marzogh