Marzogh/SPIMemory #8: readByte does not work

mtamassia, 2015-08-25:
The very first line of the readByte function should not have the braces and should return false.

The first line is the check for busy, out of range page/offset and should return false right after it.

If you compare the first two lines of the readByte and readBytes you will see the bug.

As it is today, the readByte functon only runs if the chip is busy or if you pass in an out of range page/offset.

Marzogh, 2015-08-25:
Hi, Thanks for pointing this out, but this is a bug I remember squashing before my last release. I checked and I don't find the bug you mentioned in the current one (v1.3.0). Which version of the library are you working with? :)

mtamassia, 2015-08-25:
Thank you for the quick response. I did not see the code changed in you Master branch and hence the message. I am using the library from your master branch and I just noted you have separated branch for the 1.3 release but it looks like it is WIP. Do you have a release more recent than what is in your master branch? How stable is the one on the WIP branch? Which one would you recommend I use?

Thanks,

Marcelo Tamassia
-- sent from mobile device

Marzogh, 2015-08-25:
No worries mate. I'd recommend working with the latest release that is available from the Arduino IDE library manager. :)

I wouldn't recommend using the one from the w.i.p unless you're interested in helping develop the library - if you are, please feel free to look at the w.i.p branch and submit pull requests on any changes you make. :) 

The w.i.p has got more than half a dozen bugs I'm tracking at the moment. I'm working on giving the AVR the ability to detect different Winbond chips and use the right settings for each. Im also working on getting the library to be compatible with more AVR chips - including the ATTinys.

Cheers,

Praj

mtamassia, 2015-08-26:
Thanks Praj, the latest version available from the Arduino Library Manager is 1.2.1 (screenshot below), which is the version that still has the bug. Do you have a more recent release that is not the one in progress?

[cid:image001.jpg@01D0DF9A.57FF2140]

Marzogh, 2015-08-26:
Argh! Silly me. I thought I'd pushed it through when I hadn't. I've just pushed through release v1.2.2 with the bug fix. It should propagate through to Arduino Library manager in under 30 mins - with the bug fixed. Thanks for pointing it out. I hadn't realised I was running 1.3.0w.i.p on my work computer and thought the bug was fixed in the latest release.

mtamassia, 2015-08-26:
Thank you Praj!

Marzogh, 2015-08-26:
Cheers!

closed 2015-08-26 by Marzogh

mtamassia, 2015-08-26:
Hi Praj,

You mentioned different Winbond chips.

I had to make a couple changes to get it to work with the W25X40CL chip.

1)      Reduced the number of pages from 4096 to 2048

2)      Fixed the _getAddress function, which had a bug on the int conversion since when you left shift by 8 any page over 256 would not fit in the unint16. In other words, I was only able to address pages 0 – 255 before  the change.

Original code:
uint32_t SPIFlash::_getAddress(uint16_t page_number, uint8_t offset) {
       uint32_t address = page_number << 8;
       address += offset;
       return address;
}

Fixed code:
uint32_t SPIFlash::_getAddress(uint16_t page_number, uint8_t offset) {
       uint32_t page = page_number;
       uint32_t address = page << 8;
       address += offset;
       return address;
}

In addition to these changes, I started implementing the Fast Read (0Bh) but I did not have a chance to test it out.
Thanks,
Marcelo

Marzogh, 2015-08-26:
Thanks for that! Would you like to fork the 1.3.0w.i.p branch and add them and create a pull request? I'd be happy to have you on as a fellow contributor to the project and attribute the improvements you make to you. :)

Marzogh, 2015-08-30:
Hey Marcelo, v1.3.0 is out and it has a whole bunch of improvements including the fix for the address bug you mentioned and support for a whole lot more Winbond chips. :)

mtamassia, 2015-09-01:
Awesome. Thank you for letting me know. I will definitely check it out.