Marzogh/SPIMemory #15: Problem with powerDown

joaquimortega, 2015-10-19:
Systematically powerDown() is not working for me (more precisely it returns a false)
powerUp() and other functions are working properly. The chip is a Winbond W25Q32BV

Marzogh, 2015-10-20:
Thanks for reporting this. I'll check it out and get it fixed asap.

Marzogh, 2015-10-24:
All fixed. Make sure you're running v2.1.1 :) Thanks for the heads up!

joaquimortega, 2015-10-24:
Thanks for your effort.
Unfortunately  I am still getting a false bool when I power.Down()  with
version v2.1.1.
power.Up() seems OK.

Marzogh, 2015-10-24:
Ok. Try this please. Save the code below as "recover_from_powerDown.ino" and run it.
```
#include<SPI.h>

#define RELEASE 0xAB
#define READSTAT1 0x05
#define POWERDOWN 0xB9

#define cs 10
#define xfer(n) SPI.transfer(n)

uint8_t stat1;
uint8_t devID;

void setup() {
  Serial.begin(115200);
  SPI.begin();
  SPI.setDataMode(0);
  SPI.setBitOrder(MSBFIRST);
  pinMode(cs, OUTPUT);

  digitalWrite(cs, HIGH);
  
  digitalWrite(cs, LOW);
  xfer(POWERDOWN);
  digitalWrite(cs, HIGH);
  delay(3);

  digitalWrite(cs, LOW);
  xfer(READSTAT1);
  stat1 = xfer(0);
  digitalWrite(cs, HIGH);
  Serial.println(stat1, BIN);

  digitalWrite(cs, LOW);
  xfer(RELEASE);
  (void)xfer(0);
  (void)xfer(0);
  (void)xfer(0);
  devID = xfer(0);
  digitalWrite(cs, HIGH);
  delay(3);
  Serial.println(devID, HEX);

  digitalWrite(cs, LOW);
  xfer(READSTAT1);
  stat1 = xfer(0);
  digitalWrite(cs, HIGH);
  delay(3);
  Serial.println(stat1, BIN);

  digitalWrite(cs, LOW);
  xfer(RELEASE);
  (void)xfer(0);
  (void)xfer(0);
  (void)xfer(0);
  devID = xfer(0);
  digitalWrite(cs, HIGH);
  delay(3);
  Serial.println(devID, HEX);
}

void loop() {
}

```

When it is run it should print 
```
11111111
15
0
15
```
to your serial stream at 115200 baud.

Once that is done, upload the code below as "powerDown_test.ino" to your Arduino board and run it.
```
#include<SPI.h>
#include<SPIFlash.h>

String testString = "This is a test String";
uint32_t stringAddress1, stringAddress2, stringAddress3, capacity;

SPIFlash flash(10);

void setup() {
  flash.begin();
  capacity = flash.getCapacity();
  Serial.begin(115200);
  Serial.println("Starting flash memory...");
  randomSeed(analogRead(A0));
  stringAddress1 = random(0, capacity);
  stringAddress2 = random(0, capacity);
  stringAddress3 = random(0, capacity);

  Serial.print("String write function");
  if (flash.writeStr(stringAddress1, testString))
    Serial.println(" successful");
  else
    Serial.println(" unsuccessful");

  Serial.print("Chip powerdown");
  if (flash.powerDown())
    Serial.println(" successful");
  else
    Serial.println(" unsuccessful");

  Serial.print("String write function");
  if (flash.writeStr(stringAddress2, testString))
    Serial.println(" successful");
  else
    Serial.println(" unsuccessful");

  Serial.print("Chip powerup");
  if (flash.powerUp())
    Serial.println(" successful");
  else
    Serial.println(" unsuccessful");

  Serial.print("String write function");
  if (flash.writeStr(stringAddress3, testString))
    Serial.println(" successful");
  else
    Serial.println(" unsuccessful");

}

void loop() {

}

```
When run, it should print 
```
Starting flash memory...
String write function successful
Chip powerdown successful
String write function unsuccessful
Chip powerup successful
String write function successful
```
to your serial monitor at 115200 baud.

If neither or both of them don't work, could you provide me with a copy of the code you are using and let me know which Arduino board you are working with?

joaquimortega, 2015-10-24:
"recover_from_powerDown.ino" gave the expected output. However, "powerDown_test.ino" produced the following output:
Starting flash memory...
String write function successful
Chip powerdown unsuccessful
String write function unsuccessful
Chip powerup successful
String write function successful

I attach the copy of the code that gave problems in the first place. The board is selfmade, with an ATMEGA328P running at 8000 Hz with no crystal to lower battery use. This implies that the serial monitor runs at most at 9600 bauds.

#include <UnFil.h>
#include <avr/sleep.h>
#include <avr/wdt.h>
#include <avr/eeprom.h>
#include <PCD8544.h>
#include <SFE_BMP180.h>
#include <Wire.h>
#include <dht.h>
#include <SPIFlash.h>
#include <SPI.h>  
#include <RTClib.h>

#define SPIFLASH true
#define LCD true
//Define DEBUG as true/false to choose among the debugging/release version
#define DEBUG true

char printBuffer[128];

struct Mostra {
  int voltsun;
  int voltbat;
  int myear;
  int mmonth;
  int mday;
  int mhour;
  int mminute;
  int msecond;
  double tempd;
  double tempb;
  double temph;
  double pres;
  double humi;
};

//Battery Voltage
const int batpin=A1;

//Solar panel Voltage
const int solarpin=A0;

// Initalize RTC
Chronodot RTC;

// Define a custom "degrees" symbol...
static const byte DEGREES_CHAR = 1;
static const byte degrees_glyph[] = { 0x00, 0x07, 0x05, 0x07, 0x00 };

//Initialize lcd
#if LCD
  static PCD8544 lcd;
#endif

// Dallas data wire is plugged into pin 14 on the ATmega328p. It is A2 on proto, it should be 8
const int ONE_WIRE_BUS = A2;

//Mosfet for a fake ground
const int fakegnd= 2;

//Initialize temperature chip ds18S20 i/o
[... 353 lines cut ...]
void ID() {
  Serial.println(F("----------------------------------------------------------------------------------------------------------------------------------"));
  Serial.println(F("                                                           Get ID                                                                 "));
  Serial.println(F("----------------------------------------------------------------------------------------------------------------------------------"));
  Serial.println(F("----------------------------------------------------------------------------------------------------------------------------------"));
  uint8_t b1, b2;
  uint16_t b3;
  uint32_t JEDEC = flash.getJEDECID();
  uint16_t ManID = flash.getManID();
  b1 = (ManID >> 8);
  b2 = (ManID >> 0);
  b3 = (JEDEC >> 0);
  clearprintBuffer();
  sprintf(printBuffer, "Manufacturer ID: %02xh\nMemory Type: %02xh\nCapacity: %02xh", b1, b2, b3);
  Serial.println(printBuffer);
  clearprintBuffer();
  sprintf(printBuffer, "JEDEC ID: %04lxh", JEDEC);
  Serial.println(printBuffer);
}
void clearprintBuffer()
{
  for (uint8_t i = 0; i < 128; i++) {
    printBuffer[i] = 0;
  }
}

joaquimortega, 2015-10-24:
I think that I found the problem. According to the datasheet:

"While in the power-down state only the Release from Power-down / Device ID instruction, which restores the device to normal operation, will be recognized. All other instructions are ignored. This includes the Read Status Register instruction, which is always available during normal operation."

Thus to check whether we are in powerdown mode we cannot read the status register 1. It returns 0.

Marzogh, 2015-10-25:
This is weird. I've just run my test code on an Uno, Leonardo, Mega and Due and all of them respond properly. The fact that you get the right results when you run recover_from_powerDown.ino means your flash chip is behaving normally.

The 'recover_from_powerDown.ino' sketch does the following:
  - Sets up SPI comms
  - Powers down the chip
  - Tries to read status register 1 and prints the returned byte as a binary (11111111 in this case)
  - Releases power down and prints the returned Device ID as hex (0x15 in your case)
  - Tries to read status register 1 and prints the returned byte as a binary (0 in this case)

So reading from the status register 1 when in powerDown returns 255 (decimal) and not 0. That is what the powerDown function is set to do:

```
bool SPIFlash::powerDown(void) {
	if(!_notBusy())
		return false;                                      // returns false if the chip is busy

	_cmd(POWERDOWN, NO_CONTINUE);
	#if defined (__AVR__)
	CHIP_DESELECT
	#endif
	_delay_us(3);						//Max powerDown enable time according to the Datasheet

	if (_readStat1() != 255)                     //If the status register 1 does not read 255 (binary 11111111)
		return false;
	return true;
}
```

So, if you are running v2.1.1, then, I suspect the false you are getting is because the chip is busy when you try and execute a powerDown() and not because it was of issues after powerDown. I've just double checked all my read/write functions and they all make sure that the chipSelect is pulled high before they sound the all clear, so they aren't what's locking up the chip in busy.

Have you tried to run Diagnostics.ino from the library examples and see if it passes everything?

if that works, could you try change the 
```
_delay_us(3);
```

in the powerDown function in your copy of the library to

```
_delay_us(6);
```
and see if that works? (I'm just trying to see if your lowered clock speed might be affecting the _delay_us() function - it technically shouldn't).

Marzogh, 2015-10-25:
Also try adding a ``` delay(15); ``` after you call writeAnything() and see if that helps?

joaquimortega, 2015-10-25:
The chip passes all the tests from Diagnostics.ino correctly.
I don't think that the powerDown function fails when checking if the device is busy, because I modified the code as follows:
```
bool SPIFlash::powerDown(void) {
    if(!_notBusy())
        return false;                                      // returns false if the chip is busy
    Serial.println("Not busy");
    _cmd(POWERDOWN, NO_CONTINUE);
    #if defined (__AVR__)
    CHIP_DESELECT
    #endif
    _delay_us(3);                       //Max powerDown enable time according to the Datasheet

    if (_readStat1() != 255)                     //If the status register 1 does not read 255 (binary 11111111)
        return false;
    return true;
}
```
And it always printed the string "Not busy". I also increased the delay_us without success.

In fact, I ran again recover_from_powerDown.ino moving the relevant part of the code into the loop to see the output repeatedly, and it didn't always answer the same. A typical output was the following:
0
15
0
15

10101110
15
0
15

11111011
15
0
15

11111111
15
0
15

As you can see, the output of the register while powered down doesn't seem to be predictable. This suggests that the powerDown function is powering down the device, but the device isn't answering the request to read the register. This is consistent with the behavior described in the datasheet.

Marzogh, 2015-10-26:
D'oh! Silly me. I should've checked to see what happens in a loop! Thanks for pointing it out, I'll fix it with the next release. Also, thanks for raising the issue in the first place, it let me catch a couple of serious bugs in the powerDown/powerUp functions :)

closed 2015-10-26 by Marzogh

Marzogh, 2015-10-28:
Its all been fixed in the latest v2.2.0-w.i.p branch. The code in the branch is stable as of now if you'd like to use it. The final release of v2.2.0 is a while away and I thought you might want to play around with a version of the library without the powerDown bug. :)

joaquimortega, 2015-10-28:
I will give it a try. A workaround that I found is to use a pull up
resistor (10k) on the MISO line. In this way when the memory is in sleep
mode there is no noise on the
data line and its driven high by the pullup resistor and you read all 1 to
the status request while sleeping.

Marzogh, 2015-10-28:
That is a great idea. I'll test it and include it in the notes for the next release. :)