Marzogh/SPIMemory #18: Arduino Micro support

rnestler, 2016-01-28:
Does this library support the [Arduino Micro](https://www.arduino.cc/en/Main/ArduinoBoardMicro)? I tried to test it with the Micro and a W25Q128FW chip using the TestFlash example and it seemed to work, but its not listed in the [supported boards list](https://github.com/Marzogh/SPIFlash#completely-supported).

Marzogh, 2016-01-28:
Hi Raphael,
The Micro uses the same uC as the Uno, however, I don't own a Micro and so have not tested it with one - if you can confirm that all functions work fine by running Diagnostics.ino to check I'd be happy to add it to the list of boards. :)

rnestler, 2016-01-28:
I had to modify Diagnostics.ino a bit to see the output:
```Arduino
void setup() {
  Serial.begin(115200);
  while(!Serial) {}
```
But some tests seemed to fail:
```Raw
Initialising Flash memory..........

----------------------------------------------------------------------------------------------------------------------------------
                                                           Get ID                                                                 
----------------------------------------------------------------------------------------------------------------------------------
                                                       Winbond W25Q128BV
----------------------------------------------------------------------------------------------------------------------------------
	JEDEC ID: ef6018h
	Manufacturer ID: efh
	Memory Type: 17h
	Capacity: 6018h
	Maximum pages: 65536
----------------------------------------------------------------------------------------------------------------------------------
                                                          Write Data                                                              
----------------------------------------------------------------------------------------------------------------------------------
	Data written without errors
----------------------------------------------------------------------------------------------------------------------------------
                                                          Data Check                                                              
----------------------------------------------------------------------------------------------------------------------------------
	Data Written	||		Data Read		||		Result
----------------------------------------------------------------------------------------------------------------------------------
	35		||		35			||		Pass
	-110		||		-110			||		Pass
	4520		||		4520			||		Pass
	-1250		||		-1250			||		Pass
	876532		||		876532			||		Pass
	-10959		||		-10959			||		Pass
	3.1415		||		3.1415			||		Pass
	123 Test !@#	||		123 Test !@#		||		Pass
	inputStruct	||		outputStruct		||		Pass
	0 - 255		||		0 - 255			||		Pass
----------------------------------------------------------------------------------------------------------------------------------
                                                     Check Other Functions                                                        
----------------------------------------------------------------------------------------------------------------------------------
			Function		||		Result
----------------------------------------------------------------------------------------------------------------------------------
			powerDown		||		Pass
			powerUp			||		Pass
			sectorErase		||		Pass
			chipErase		||		Fail
----------------------------------------------------------------------------------------------------------------------------------
```

rnestler, 2016-01-28:
Also note that I have a *W25Q128FW* attached, not a *W25Q128BV* as the output above suggests.

Marzogh, 2016-01-28:
The chipErase function sometimes throws errors. I need to get that particular bug squashed soon. It usually passes if you hit reset and re-run the sketch a couple of times :)

As for the FW vs BV its just lazy coding on my part I'm afraid. I didn't know a way of ID'ing BV/FW from the JEDEC ID so I just got it to default to BV.

Marzogh, 2016-01-28:
Thanks for checking on Arduino Micro support. ReadMe.md has been updated to reflect the changes.

closed 2016-01-28 by Marzogh