Marzogh/SPIMemory #23: How to get library to compile for ESP8266 (Arduino)

gweep, 2016-06-02:
I get the following error when compiling for the ESP8266 in the Arduino IDE.   

fatal error: util/delay.h: No such file or directory  #include <util/delay.h>

Any thoughts on getting the code to compile?

Marzogh, 2016-06-02:
I haven't actually gotten around to testing this library with the ESP8266 yet and so it is not officially supported.

If the issue is with util/delay.h then, off the top of my head, the following steps should solve that problem.

1. Find the following section right at the beginning of SPIFlash.cpp 
`#if defined (__arm__) && defined (__SAM3X8E__)
 #define _delay_us(us) delayMicroseconds(us)
#else
 #include <util/delay.h>
#endif`

2. Replace it with the code below

`// #if defined (__arm__) && defined (__SAM3X8E__)
 #define _delay_us(us) delayMicroseconds(us)
// #else
//  #include <util/delay.h>
// #endif`

This is a temporary fix for letting the code not throw errors w.r.t util/delay.h when compiling for the ESP8266. I'm not sure if the way I've written the library lets it be compatible with the 8266's SPI library, but I won't know that till I hear back from you :) *Fingers crossed*

gweep, 2016-06-02:
New errors after implementing the suggested changes.  

          expected primary-expression before 'void'    (void)xfer(cmd);

Then added a semi-colon ";" after the "CHIP_SELECT" statement in this sequence.   
#else
	CHIP_SELECT
	(void)xfer(cmd);
 #endif

Resulting in the following error.

'CHIP_SELECT' was not declared in this scope

Marzogh, 2016-06-02:
Which ESP8266 board are you using? I'll see if I can quickly whip up a custom .cpp file for you.

Marzogh, 2016-06-02:
Don't worry about that. just realised there's a pre-processor directive for the ESP-8266 architecture.

Try replacing the code in SPIFlash.cpp with the code below :

```
/* Arduino SPIFlash Library v.2.2.0
 * Copyright (C) 2015 by Prajwal Bhattaram
 * Modified by Prajwal Bhattaram - 24/11/2015
 *
 * This file is part of the Arduino SPIFlash Library. This library is for
 * Winbond NOR flash memory modules. In its current form it enables reading 
 * and writing individual data variables, structs and arrays from and to various locations;
 * reading and writing pages; continuous read functions; sector, block and chip erase;
 * suspending and resuming programming/erase and powering down for low power operation.
 *
 * This Library is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This Library is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License v3.0
 * along with the Arduino SPIFlash Library.  If not, see
 * <http://www.gnu.org/licenses/>.
 */
 
#include "SPIFlash.h"

#if defined (__SAM3X8E__) || defined (ARDUINO_ARCH_ESP8266)
 #define _delay_us(us) delayMicroseconds(us)
#else
 #include <util/delay.h>
#endif

//~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~//
//     Uncomment the code below to run a diagnostic if your flash 	  //
//	   						does not respond   						  //
//																	  //
// 		Error codes will be generated and returned on functions		  //
//~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~//
//#define RUNDIAGNOSTIC												  //
//~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~//
//~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~//
//   Uncomment the code below to increase the speed of the library    //
//	   				by disabling _notPrevWritten()   				  //
//																	  //
// Make sure the sectors being written to have been erased beforehand //
//~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~//
#if defined (__SAM3X8E__)
//#define HIGHSPEED		
#elif defined (ARDUINO_ARCH_AVR) || defined (ARDUINO_ARCH_ESP8266)
//#define HIGHSPEED													  
#endif																  
//~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~//

#define _MANID		 0xEF
#define PAGESIZE	 0x100

#define	MANID		 0x90
#define PAGEPROG     0x02
#define READDATA     0x03
#define FASTREAD	 0x0B
#define WRITEDISABLE 0x04
#define READSTAT1    0x05
#define READSTAT2	 0x35
#define WRITEENABLE  0x06
[... 1792 lines cut ...]
		}
		else if (status1 != status2)
			return true;
	}
	else if (status1 == 255 && status2 == 255)
		return true;
	else if (status1 == 0 && status2 == 0)
		return false;
	return true;
}

//Wakes chip from low power state.
bool SPIFlash::powerUp(void) {

	_cmd(RELEASE, NO_CONTINUE);	
	#if defined (ARDUINO_ARCH_AVR) || defined (ARDUINO_ARCH_ESP8266)
	CHIP_DESELECT
	#endif
	_delay_us(3);						    //Max release enable time according to the Datasheet

	if (_readStat1() == 255)
		return false;
	return true;
}
```

gweep, 2016-06-02:
Thanks,  Looks like that is a big start.  Now getting the following error:     

.../Arduino/libraries/SPIFlash-2.2.0/SPIFlash.cpp: In constructor 'SPIFlash::SPIFlash(uint8_t, bool)':
.../Arduino/libraries/SPIFlash-2.2.0/SPIFlash.cpp:149:13: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
     cs_port = portOutputRegister(digitalPinToPort(csPin));

Am I missing a standard library?

To answer your previous question,  I am using a NodeMCU v1.0 with the Arduino IDE.

Marzogh, 2016-06-02:
Ah. I hoped that wouldn't happen!

Try replacing the section below in your current version of SPIFlash.cpp
```
#if defined (__SAM3X8E__)
 #include <SPI.h>
#elif defined (ARDUINO_ARCH_AVR) || defined (ARDUINO_ARCH_ESP8266)
    #ifdef __AVR_ATtiny85__
        #define CHIP_SELECT   PORTB &= ~cs_mask;
        #define CHIP_DESELECT PORTB |=  cs_mask;
        #define SPIBIT                      \
          USICR = ((1<<USIWM0)|(1<<USITC)); \
          USICR = ((1<<USIWM0)|(1<<USITC)|(1<<USICLK));
        static uint8_t xfer(uint8_t n) {
            USIDR = n;
            SPIBIT
            SPIBIT
            SPIBIT
            SPIBIT
            SPIBIT
            SPIBIT
            SPIBIT
            SPIBIT
            return USIDR;
        }
    #else
        #include <SPI.h>
        #define CHIP_SELECT   *cs_port &= ~cs_mask;
        #define CHIP_DESELECT *cs_port |=  cs_mask;
        #define xfer(n)   SPI.transfer(n)
    #endif
#endif
```
with this:
```
#if defined (__SAM3X8E__)
 #include <SPI.h>
#elif defined (ARDUINO_ARCH_AVR)
    #ifdef __AVR_ATtiny85__
        #define CHIP_SELECT   PORTB &= ~cs_mask;
        #define CHIP_DESELECT PORTB |=  cs_mask;
        #define SPIBIT                      \
          USICR = ((1<<USIWM0)|(1<<USITC)); \
          USICR = ((1<<USIWM0)|(1<<USITC)|(1<<USICLK));
        static uint8_t xfer(uint8_t n) {
            USIDR = n;
            SPIBIT
            SPIBIT
            SPIBIT
            SPIBIT
            SPIBIT
            SPIBIT
            SPIBIT
            SPIBIT
            return USIDR;
        }
    #else
        #include <SPI.h>
        #define CHIP_SELECT   *cs_port &= ~cs_mask;
        #define CHIP_DESELECT *cs_port |=  cs_mask;
        #define xfer(n)   SPI.transfer(n)
    #endif
#elif defined (ARDUINO_ARCH_ESP8266)
        #include <SPI.h>
        #define CHIP_SELECT   digitalWrite(csPin, HIGH);
        #define CHIP_DESELECT digitalWrite(csPin, LOW);
        #define xfer(n)   SPI.transfer(n)
#endif
```
and hopefully that should fix it.

gweep, 2016-06-03:
replaced the section...   still getting the same or similar error.

/Arduino/libraries/SPIFlash-2.2.0/SPIFlash.cpp: In constructor 'SPIFlash::SPIFlash(uint8_t, bool)':

/Arduino/libraries/SPIFlash-2.2.0/SPIFlash.cpp:155:13: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
     cs_port = portOutputRegister(digitalPinToPort(csPin));

added a few spaces to my code so my line numbers may not match yours.

Marzogh, 2016-06-03:
Argh! I missed another section:

just replace:
```
#if defined (__SAM3X8E__)
SPIFlash::SPIFlash(uint8_t cs, bool overflow) {
    csPin = cs;
    pageOverflow = overflow;
}
#elif defined (ARDUINO_ARCH_AVR) || defined (ARDUINO_ARCH_ESP8266)
SPIFlash::SPIFlash(uint8_t cs, bool overflow) {
    csPin = cs;
#ifndef __AVR_ATtiny85__
    cs_port = portOutputRegister(digitalPinToPort(csPin));
#endif
    cs_mask = digitalPinToBitMask(csPin);
    SPI.begin();
    SPI.setDataMode(0);
    SPI.setBitOrder(MSBFIRST);
    //SPI.setClockDivider(SPI_CLOCK_DIV2);
    pageOverflow = overflow;
    pinMode(cs, OUTPUT);
}
#endif
```
with
```
#if defined (__SAM3X8E__)
SPIFlash::SPIFlash(uint8_t cs, bool overflow) {
    csPin = cs;
    pageOverflow = overflow;
}
#elif defined (ARDUINO_ARCH_AVR) || defined (ARDUINO_ARCH_ESP8266)
SPIFlash::SPIFlash(uint8_t cs, bool overflow) {
    csPin = cs;
#ifndef __AVR_ATtiny85__
    cs_port = portOutputRegister(digitalPinToPort(csPin));
#endif
#ifndef ARDUINO_ARCH_ESP8266
    cs_mask = digitalPinToBitMask(csPin);
#endif
    SPI.begin();
    SPI.setDataMode(0);
    SPI.setBitOrder(MSBFIRST);
    //SPI.setClockDivider(SPI_CLOCK_DIV2);
    pageOverflow = overflow;
    pinMode(cs, OUTPUT);
}
#endif
```

Hopefully that'll sort out the assignment issue. :)

gweep, 2016-06-03:
Still getting the same identical error.     Will try reconstructing the .cpp file from the beginning to make sure the changes are correctly incorporated.   May take a day or so.

Marzogh, 2016-06-03:
I haven't managed to get around to actually testing with an ESP8266 board yet, but if you download the version from the [v2.2.1-ESP8266_compat ](https://github.com/Marzogh/SPIFlash/tree/v2.2.1-esp8266_compat)branch, the library compiles without error.

Marzogh, 2016-06-04:
Just finished testing and fixing the library with the ESP8266 - I'm using the Adafruit Huzzah. For the a little while (while I check and see that I haven't broken compatibility with any other platforms) this version of the library will be live on the v2.2.1-esp8266_compat branch.

Marzogh, 2016-06-04:
Just pushed through release v2.3.0 with ESP8266 compatibility sorted and tested. It should come through as an update in the IDE's library manager in 30 mins or so. :)

closed 2016-06-04 by Marzogh

gweep, 2016-06-04:
With my first pass through with v2.3.0 I am getting good compiles and uploads.  the readWriteString example is working.   Now need to incorporate the library into my own code.

THANKS!

Mike

Marzogh, 2016-06-04:
My pleasure!