Marzogh/SPIMemory #30: ATMega2560 problems

robertkendrick, 2016-07-04:
Have ATmega2560 wired up to W25Q64FV winbond chip. Using Hardware SPI pins and SS pin (53) for chipselect
Running XP.

If I use the SPIflash library from RinkyDink website the above setup is reported as working fine.

If I erase the chip using RinkyDink and then switch to your library and run your testflash sketch the only menu item that works is number 1 and it reports the correct chip ID for W25Q64FV.
All other menu items fail or appear to do nothing. ie EraseSector 

EraseChip returns very quickly, the RinkyDink erasechip function takes quite a while.
WriteByte reports data failed. ReadPage reports 256 X 0xFF as you would expect but none of the commands that perform writes to the flash chip are working.

This is most strange given the large amount of positive feedback you have received from users on the internet. Is it me ?? Appreciate if anything obvious springs to mind.
BTW. Have also tried 2.4.0,wip download but results are the same. 
Best Regards

Marzogh, 2016-07-04:
Hi Robert,

Thanks for bringing this up. What you have described sounds weird. I don't have a bare 2560 but I just connected an Arduino Mega2560 board to a W25Q64 and everything works fine. 

Are you using a bare chip? If yes, which arduino board definition are you using? If not, could you please give me the details of your set-up (clock-speed, pin numbers, etc.) so I can try and replicate your issue and fix it.

Cheers!

P.S. eraseChip() for a W25Q64 should take anywhere between 4-8 seconds depending on your clock speed.

Marzogh, 2016-07-05:
Also, Could you uncomment #RUNDIAGNOSTIC in SPIFlash.cpp and then run Diagnostics.ino and let me know what errors are being generated? Thanks! :)

robertkendrick, 2016-07-05:
Thanks for the reply.
I'm using an ordinary Arduino Mega 2560 board. 16Mhz clock. Hardware SPI pins 50-52 (MISO, MOSI, SCK) with SS pin(53) as chip select. Using your 2.4.0.wip download.
Have run your diagnostics. Results attached. This kinda proves the SPI pins are connected correctly as the Chip ID info is displayed correctly in the results. Maybe a strange faulty chip ?? I'll schedule some time to compare again against the RinkyDink library (which seems to be working) and also try an arduino UNO or pro-mini. This will probably turn out to something really silly and not your software.
Regards
[spiflash_diagResults.txt](https://github.com/Marzogh/SPIFlash/files/347563/spiflash_diagResults.txt)

robertkendrick, 2016-07-05:
Hi again :-)
Should have mentioned. The flash chip pins /WP and /HOLD are tied to VCC (3v3). I'm also using a 4 channel level converter to translate the 5v to 3v3 and vice-vera.
Run the RinkyDink test again just by downloading the sketch; no changes to the hardware. Results attached. 
The elapsed time from the message 'Erasing chip again' to 'done' was about 14 secs.
Regards
[RinkyDink_Test_Results.txt](https://github.com/Marzogh/SPIFlash/files/347620/RinkyDink_Test_Results.txt)

Marzogh, 2016-07-05:
Ah. I usually tie WP to ground. May be that will help? (Not sure it will, but we can try hey?) :)

Marzogh, 2016-07-05:
Just to confirm, so it's actually not throwing any errors even after you've uncommented # define RUNDIAGNOSTIC  ?

Marzogh, 2016-07-05:
Fixed! I just got around to sticking a logic analyser on my system and tossing in a level converter (I use a Seeduino Mega that lets me switch its logic levels between 5V and 3V3. So, I've not used a logic converter previously). It was a timing issue. Its been fixed with the latest code I've pushed through to 2.4.0-w.i.p. I've attached the diagnostics readout for the latest version of 2.4.0-w.i.p below. :)
[image]

closed 2016-07-05 by Marzogh

robertkendrick, 2016-07-06:
Hi Marzogh
Your fix did not work for me. The problem remained exactly the same. 
However, I wrote a tiny sketch with your library that just issued an erasechip instruction to the flash memory and examined the communication with a logic analyser.
You issue an WriteEnable instruction (0x06) and then a ReadStatus instruction (0x5) followed by the EraseChip instruction (0xc7).
The ReadStatus instruction returns a value of 0x3e. When I decoded this it is evident that all the Block Protection bits are set !!!! THAT IS WHY ALL MEMORY WRITE FUCTIONS ARE NOT WORKING IN YOUR CODE.
I double checked by looking at the RinkyDink software and indeed he write protects the whole memory array after doing any write functions (WritePage, EraseChip etc) . So this is the source of my problems. Had I used your library first I would not have encountered any issues. So I owe you a big apology. You might want to deal with Block protetction bits in a future release (just in case :-)
Thank you for a very useful library.
Regards

Marzogh, 2016-07-06:
Ah! That explains it all. Thank you for the heads up. I'll add a quick check of the block protection bits before any erase operation as a part of the next release. :)

Marzogh, 2016-07-06:
P.S. No apology needed. You've helped me fix what could've come back to bite me in the future. :)

a93-39a, 2019-07-28:
@Marzogh 

I know this is a dead thread but my diagnostic tool keeps telling me the same thing when I try to load it and this thread has almost identical hardware to mine.

Example warning.

`Initialising..........
Chip Diagnostics initiated.
`

I have waited for some time for it to continue although it seems to lock up every time.

I built my own bidirectional npn transistor based logic level using  a bread board.

I checked the wiring a couple of times.

Could it still be erasing the chip or running some hidden process?

Chip is W25Q64JVSIQ
Board is Arduino MEGA2560
Issue is program lock up after upload to MEGA2560.

Setup Picture
[image]

1. How long should chip diagnostics take?
2. Do you remember off hand the pin-up or pin-down arrangement of the chips hold and write-protect?