Marzogh/SPIMemory #45: ATTiny85 compatibility

JelmerT, 2017-01-15:
I'm trying to compile the library on an ATTiny85 (adafruit trinket), and I'm getting quite some errors.

for example:
> SPIFlash/defines.h:81:12: error: 'SS' was not declared in this scope
 #define CS SS

I can tell there are some checks for __AVR_ATtiny85__ in the library, but I can't make the standard examples work out of the box. Am I missing something? Or has ATTiny support not been tested in a while?

Marzogh, 2017-04-12:
Hi @JelmerT 

Thank you for pointing this out. I have actually been a bit lazy w.r.t testing ATTiny support for the last release, but, I will fix this with the upcoming 2.6.0 release.
Also, my apologies for the late reply. Work has been eating up my time for the last few months and I've only just gotten back into looking at this library.

Cheers!

JelmerT, 2017-04-12:
Thanks @Marzogh ! Sounds great 👍

Marzogh, 2017-04-15:
I'm afraid it looks like a lot has fallen through the gaps with the last release. I'm going to have to push support for the ATTiny85 back to v2.7.0 I'm afraid. v2.6.0 has some important bugfixes that need to be released asap. Sorry! I'm already working on the changes and should have it out before the end of the month.

JelmerT, 2017-04-16:
👍

Marzogh, 2017-04-19:
Hey mate, ATTiny85 compatibility is all good to go as of today's commit on the [ATTiny-compat-w.i.p](https://github.com/Marzogh/SPIFlash/tree/ATTiny-compat-w.i.p) branch. I'm working on polishing up a proper diagnostics example for this so you can compile and run it on the ATtiny85 to check everything's a-ok. For now, feel free to use the tinyFlashTester.ino as a rough test sketch. Once its all sorted and tested with the other boards as well, I'll roll this out into v2.7.0.

JelmerT, 2017-04-22:
Great news! Will test and report back, might take some time though, thanks for this.

Marzogh, 2017-08-02:
Testing works fine at my end - I've tested with a 8MHz 3.3V trinket. So, I've rolled this into the 2.7.0 release along with Arduino Zero compatibility. Please let me know if you face any issues in the future.

closed 2017-08-02 by Marzogh