Marzogh/SPIMemory #49: Doesn't work with Arduino 1.0.x

sej7278, 2017-04-07:
Examples don't compile with IDE 1.0.5 and even using a Makefile and calling avr-g++ directly (and enabling gnu++11 and so on) it won't build.

If I add the ```#define ARDUINO_ARCH_AVR 1``` to readWriteString.ino (should really be ```__AVR__``` to do it automatically) I get a bit further but also need to change:

```(void)xfer(PAGEPROG);``` at line 290 etc. to ```void xfer(PAGEPROG);```

But then I get the error:
```
error: expected initializer before ‘.’ token
     #define xfer(n)   SPI.transfer(n)
```

Which I assume is because SPI.transfer() expects two arguments - a pointer and a value, or lack of trailing semi-colons in various places?

Marzogh, 2017-04-12:
Hi @sej7278 ,

I'm afraid IDE 1.0.5 is from way before my time with Arduino. From what I can see, there have been some pretty big changes to the SPI library over this time. Most importantly the added support for transactions. The number of changes required to make it back compatible appear to be quite large. Is there a particular reason why you would like to keep using IDE 1.0.5?

Cheers!

sej7278, 2017-04-12:
I don't like the direction the 1.5+ ide is going in with the awful boards manager and library manager, plus their licensing causes problems with packaging for most linux distro's.

closed 2017-04-12 by sej7278