Marzogh/SPIMemory #62: Calculate and save a checksum for data files

Marzogh, 2017-04-19:
Once a file system is setup on the Flash chip, the end of every file could possibly have a 16/32bit checksum to test the integrity of the file - without having to re-read the whole thing?

boseji, 2017-05-30:
I would love to have this feature for specific blocks.
Here is some links to a easy to implement [CRC-32](http://excamera.com/sphinx/article-crc.html) from excamera (James Bowman)

Another solution can be to provide a generator function interface used by Hashing algorithms.
For example I need to make checksum of the a particular Page:
``` C
// Somewhere in the code
fn_gen = call PreparePage(Sector, Page); // Get the function pointer
...
// Somewhere Inside the Hashing function
  sum += fn_gen();
...
  return hash_buffer; // Final hashed output
}
```

Hope that this can be part of the next release.
I am currently testing the v3.0.0-work in progress branch for RTL8195A.
Also there is some progress on the RTL8710, but need to merge that older versing to v3.0.0

Thanks again for the super useful code.

Marzogh, 2017-05-30:
Hey mate, thanks for the information. I'm still learning my way around hashing and this will be helpful. 😄 

On a different note, the latest stable pre-release code is always in the dev branch, the other branches keep changing (quite dramatically sometimes) as I test out new features. The next version that is due is v2.7.0 and it comes with Arduino Zero support and (hopefully) fully tested ATTiny support. 😃
The v3.0.0-w.i.p branch has a long way to go before it is stable.

boseji, 2017-06-01:
Thanks for the direction would branch to v2.7.0