Marzogh/SPIMemory #63: Is Cypress f.memory works?

hasifazami, 2017-04-27:
Hi mate, I try using your library for my Cypress S25FL512S f.memory using Arduino MEGA and somehow it display error code 0x03. I edit some common instructions from defines.h code that match for my f.memory but still does not work. Can you give me some advice what to edit if wanna to use another f.memory (i.e Cypress f.memory for my case)? Really appreciate it if you can help me because I am still learning about this. Thank you.

Marzogh, 2017-06-16:
Hi mate, the code on the v3.0.0-w.i.p branch should get you going for now. Please note that this is pre-release code and can change quite dramatically by the time it is ready for release.

haapeng, 2017-06-17:
Hi, I tried run code v3.0.0-w.i.p and this is the results. Seems it can read the flash memory and may be read data, but data write failed. I had attached the test results and my schematic
[Try1.pdf](https://github.com/Marzogh/SPIFlash/files/1082391/Try1.pdf)

Marzogh, 2017-08-10:
Issue #69 had similar problems and its been fixed. I have tested the library code (currently in the development branch - soon to be moved to stable) with the S25FL116K and it works. Since the opcodes don't change between the S25FL116K & the S25FL512S, in theory, the library should work with your chip too. Please let me know if it does.

Marzogh, 2017-08-18:
As of pull request #79 the code on the stable branch works with the Cypress S25FL116K. Since the S25FL512S shares opcodes with the S25FL116K, you should - in theory be able to get the code in the stable branch to work for you. Please let me know how it goes.

Note: Please remember that even though the branch is called 'stable' it is still code that is in testing/development and things may break or change with no warning.

closed 2017-08-18 by Marzogh