Marzogh/SPIMemory #67: In my case it gives 0 for all function call . Please give any solution of it.

kamleshnarola, 2017-05-26:
```
#include<SPIFlash.h>
#include<SPI.h>
#define arrayLen(x) sizeof(x)/sizeof(x[0])
uint32_t strAddr[3], floatAddr[2], byteAddr[4];
String testStr[] = {
  "Test String 0",
  "Test String 1",
  "Test String 2"
};
float testFloat[] = {
  3.1415, 6.283
};
byte testByte[] = {
  3, 245, 84, 100
};

SPIFlash flash;

void getAddresses();
void writeData();

void setup() {
  Serial.begin(9600);
  Serial.print(F("Initialising Flash memory"));
  for (int i = 0; i < 10; ++i)
  {
    Serial.print(F("."));
  }
  Serial.println();
  flash.begin();
  
  Serial.println();
  Serial.println();

  getAddresses();
 
  writeData();
     Serial.printf("\n getMaxPage  = %d\n", flash.getMaxPage());
     
     Serial.printf("\n getCapacity= %d\n", flash.getCapacity());
      
      Serial.printf("\n getManID = %d\n", flash.getManID());
      
 
  //flash.eraseChip();
}

void loop() {

}
// Function to get adresses for various variables
void getAddresses() {
  for (uint8_t i = 0; i < arrayLen(byteAddr); i++) {
    byteAddr[i] = flash.getAddress(sizeof(byte));
    Serial.printf("error= %d\n", flash.error());
    Serial.print("Byte Address ");
    Serial.print(i);
    Serial.print(" : ");
    Serial.println(byteAddr[i]);
  }

  for (uint8_t i = 0; i < arrayLen(floatAddr); i++) {
    floatAddr[i] = flash.getAddress(sizeof(float));
     Serial.printf("error= %d\n", flash.error());
    Serial.print("Float Address ");
    Serial.print(i);
    Serial.print(" : ");
    Serial.println(floatAddr[i]);
  }

  for (uint8_t i = 0; i < arrayLen(strAddr); i++) {
    strAddr[i] = flash.getAddress(flash.sizeofStr(testStr[i]));
  Serial.printf("error= %d\n", flash.error());
    Serial.print("String Address ");
    Serial.print(i);
    Serial.print(" : ");
    Serial.println(strAddr[i]);
  }
}

// Function to write data
void writeData() {
  for (uint8_t i = 0; i < arrayLen(byteAddr); i++) {
    if (flash.writeByte(byteAddr[i], testByte[i])) {
      Serial.printf("error= %d\n", flash.error());
      Serial.print(testByte[i]);
      Serial.print(" written to ");
      Serial.println(byteAddr[i]);
    }
  }

  for (uint8_t i = 0; i < arrayLen(floatAddr); i++) {
    if (flash.writeFloat(floatAddr[i], testFloat[i])) {
     Serial.printf("error= %d\n", flash.error());
      Serial.print(testFloat[i]);
      Serial.print(" written to ");
      Serial.println(floatAddr[i]);
    }
  }

  for (uint8_t i = 0; i < arrayLen(strAddr); i++) {
    if (flash.writeStr(strAddr[i], testStr[i])) {
     Serial.printf("error= %d\n", flash.error());
      Serial.print(testStr[i]);
      Serial.print(" written to ");
      Serial.println(strAddr[i]);
    }
  }
}
```
========================================
out put
===========================================
cüÿInitialising Flash memory..........
error= 5
Byte Address 0 : 0
error= 5
Byte Address 1 : 0
error= 5
Byte Address 2 : 0
error= 5
Byte Address 3 : 0
error= 5
Float Address 0 : 0
error= 5
Float Address 1 : 0
error= 5
String Address 0 : 0
error= 5
String Address 1 : 0
error= 5
String Address 2 : 0

 getMaxPage  = 0

 getCapacity= 0

 getManID = 0

Marzogh, 2017-05-29:
Are you trying to read the NodeMCU module's built-in flash? AFAIK NodeMCU's onboard flash is wired for Dual IO SPI access <[link to NodeMCU's FAQ](https://nodemcu.readthedocs.io/en/master/en/hardware-faq/#why-file-writes-fail-all-the-time-on-devkit-v10)>. This library only supports standard SPI access.

closed 2017-05-30 by Marzogh

kamleshnarola, 2017-05-31:
I am flashing above code from Arduino IDE,Do you have working demo of Arduino for NodeMCU device?

Marzogh, 2017-08-03:
The library works fine with the NodeMCU when used with the Arduino IDE. The problem is that you appear to be trying to read the built-in flash with your code. Is that what you are trying to do?