Marzogh/SPIMemory #70: Add ability to pick between multiple SPI ports for library comms

Marzogh, 2017-06-16:
STM32 Boards have more than one SPI port. The library has been reported to work with port 1. Would it be possible to let the user code pick which port to use through the library?

VitorBoss, 2017-06-20:
Thanks for investigate this, this git could help you https://github.com/RRPRIME/STM32

Marzogh, 2017-06-20:
Cheers, thanks. :)

VitorBoss, 2017-07-21:
Any update? Also, could you add an option to use native SPI routines? As you can see STM32 is working and really fast.
[image]

Marzogh, 2017-08-02:
My apologies for the delay in getting back to you. Work took over my life for a while (in the form of 1251 students who decided to enroll in the First Year genetics course I run!). I've finally got time to look at the library again and I've got the STM32 board - hopefully I'll be able to get this sorted ASAP.

Thanks for running the tests. That is spectacularly fast. Are you using the code from v2.6.0? If yes, would you mind trying the code from the branch v3.0.0-w.i.p? That's the branch I'm going to roll the STM32 support into. Its a complete re-write of the library and so far I've been getting much better I/O speeds from the future v3.0.0 code.

VitorBoss, 2017-08-02:
Seems to work nice, few functions loose speed(my device is running at 168MHz)
[image]

Marzogh, 2017-08-18:
I haven't finished setting up the library to work with the STM32 yet, but the code on the development branch (will go stable once I finish testing the feature across all supported boards) currently supports the ability to use multiple SPI Ports - it's currently been tested with the onboard memory on the Adafruit M0 Feather Express. (It has Cypress 2MBit memory on SPI1.)

Marzogh, 2017-08-31:
I'm afraid I'm having trouble getting the blue pill board to work with Arduino on my Mac. I have tested the library - as of the latest commit to the development branch - with a STM32 Nucleo-F091RC and it works. Could you please test the code in this branch with the Blue Pill boards that you have?

VitorBoss, 2017-08-31:
Yes, I will test and comment later

VitorBoss, 2017-08-31:
Which library are you using for STM32?
I'm using https://github.com/rogerclarkmelbourne/Arduino_STM32 and get a lot of errors

I create PR to fix some errors

Marzogh, 2017-11-04:
Thanks for the info :) I'll roll the the changes to enable the use of Serial properly across multiple boards into the next release. :) (Ref #88)

closed 2017-11-04 by Marzogh