Marzogh/SPIMemory #71: Random Error Code

AMiller10, 2017-07-04:
Hi,

I'm using an Arduino Uno with a Winbond 32mB chip, W25Q256FV.  I've been trying to get it to run with no success.  I have already swapped the chip out and that changed nothing.  I've been running the Flash Diagnostics page and it only works on special cases.  I have to comment out the GetID and diagnose functions and upload.  Then I can uncomment GetID and upload.  Then finally I can uncomment diagnose and upload, but I have to have the powerFuncDiag commented out.  If I don't do all those steps I get:

Error Code 0x04
Error Code 0x04
Error Code 0x04.  

Which says could be a wiring problem, but I'm able to get it to work if I use all the steps above (besides PowerDown, that has never passed).  When I do all of that stuff first I get the screenshot that I attached.  
[image]
 I'm using the sparkfun logic level converter you recommended in the arduino forums and I've checked my wiring 50 times at least.  Would appreciate any help/wondering if there is a bug in the program.  Thank you very much!

Marzogh, 2017-08-02:
My apologies for the delay in getting back to you. Work took over my life for a while (in the form of 1251 students who decided to enroll in the First Year genetics course I run!). I will take a look at this for you and get back to you by next week. 😄

AMiller10, 2017-08-03:
No worries.  But awesome, sounds good.  Thanks a lot!

Marzogh, 2017-08-18:
I'm sorry, but, I have not been able to get my hands on the chip you are using yet. 😒  Could you do me a favour and try running the code form the 'stable' branch and see if that works? 🙏 😃  Its v3.0.0 early alpha level stuff and a lot has changed in the library and I'm hoping I've fixed the bug unwittingly. (I suspect its an address transfer bug) If that doesn't work, I'll just have to wait till AliExpress can get me a couple of the chips via snail mail.

Marzogh, 2017-11-04:
Any luck?

Marzogh, 2017-11-10:
I suspect I may have fixed your problem as well when I fixed #91 😄 Can you confirm that the code in the 'stable' branch runs with no problems?

closed 2017-11-17 by Marzogh