Marzogh/SPIMemory #72: How to handle multiple flash chips?

riklaunim, 2017-07-26:
So the constructor is ```SPIFlash flash(csPin);``` but how to create two constructors for two flash chips with different csPins and start them with .begin() in setup?

Marzogh, 2017-08-02:
Good question. I've never tested that functionality before. The simplest way I can think of achieveing this is something along the lines of

```
#include <SPIFlash.h>
SPIFlash flash1(csPin1);
SPIFlash flash2(csPin2);
SPIFlash flash3(csPin3);
...
...

void setup() {
flash1.begin();
flash2.begin();
flash3.begin();
...
}

void loop() {
...
}
```

I'll see if I can test this sometime in the next week, but if you manage to test this before me, I'd love to know how it goes! 😄

riklaunim, 2017-08-02:
I tried that. But it would hang on the flash2.begin() line. What works is calling SPI.endTransaction() - after working with one device and switching to another one (one SPI device at a time). So if there is .begin() then .end() would also make sense to have (ending the SPI transaction).

Tested on ESP32 with 2 WinBond flash chips and Adafruit SPI OLED display.

Marzogh, 2017-08-02:
Ah. Silly me! That's something I told myself to fix - about four months ago and promptly forgot. Thank you for catching that. I will include a .end() with v3.0.0

Marzogh, 2017-08-18:
All done and good to go. :) The code on the development branch fixes the issue by calling an _endSPI() at the end of the flash.begin() function. The change will be rolled out into v3.0.0.

Thanks for catching this bug.

closed 2017-08-18 by Marzogh