Marzogh/SPIMemory #76: Faster memory access

VitorBoss, 2017-08-07:
That is a idea for enhancement, eliminate all shifts from code, that will increase average speed, just use those macros:
```
#define Lo(param) ((char *)&param)[0] //0x000y
#define Hi(param) ((char *)&param)[1] //0x00y0
#define Higher(param) ((char *)&param)[2] //0x0y00
#define Highest(param) ((char *)&param)[3] //0xy000
#define Low(param) ((int *)&param)[0] //0x00yy
#define Top(param) ((int *)&param)[1] //0xyy00
```
In my tests those macros reduce memory access in a 2000 loop from 200+mS to 2mS.

Marzogh, 2017-08-10:
Hmm... Which shifts have you been replacing in your tests? (and which board are you using?) My testing does not show any difference at all.

VitorBoss, 2017-08-13:
Sorry for delay, I had tested on Arduino Mega on a dumb loop test.

Also, the declared constant PAGESIZE conflicts with some of my libraries, I changed it to SPI_PAGESIZE and solved the problem.

VitorBoss, 2017-08-13:
An example is here:
```
  for (uint8_t i = 0; i < 4; i++) {
    _chip.sfdp += (_nextByte() << (8*i));
  }
```
You could just use this:
```
Lo(_chip.sfdp) = _nextByte();
Hi(_chip.sfdp) = _nextByte();
Higher(_chip.sfdp) = _nextByte();
Highest(_chip.sfdp) = _nextByte();
```

Marzogh, 2017-08-18:
True that. Thanks for pointing it out. 😄 I haven't gotten around to actually using the _getSFDP() function anywhere in the library yet - its sort of sitting there as a placeholder for stuff in the future. 😃

closed 2017-08-18 by Marzogh