Marzogh/SPIMemory #83: Name conflict with SPIFlash library

LowPowerLab, 2017-09-08:
Hi,
I have a library called **SPIFlash** which was created many years ago.
Since you created your own "SPIFlash" with the identical same name, Arduino is causing naming conflicts. I suspect your users have the same issue.
Could you possibly rename your library to something else?

whmountains, 2017-09-26:
Maybe `WinbondSPIFlash`?

Marzogh, 2017-11-04:
Hi @LowPowerLab , my apologies about the naming conflicts. Unfortunately, I suspect changing the name of the library right now will cause updating issues for current users on the Arduino IDE - please correct me if I am wrong. If I can change the name of the library and let users who access & update the library through the library manager to seamlessly upgrade to the new version (with the new name), I'll do that straightaway with the next release.

If that is not the case - if you wouldn't mind waiting a short while - I am working on merging SPI FRAM and SPI EEPROM I/O functions into this library, when I do, I will rename it to SPIMEM, release it anew and pull this version from the Arduino Library manager. Would that be ok?

P.S. @whmountains I'm afraid the library is no longer just for Winbond SPI Flash modules. It has unofficially supported a few other manufacturers for the last couple of versions and will officially support a lot more starting v3.0.0

LowPowerLab, 2017-11-04:
@Marzogh Thanks for replying to this thread and taking action, I am not sure what effect renaming now or later will have. But it will continue to be a problem until we have 2 libraries with the same name. I will leave it up to you how to best move this forward, your cooperation is appreciated.

Marzogh, 2017-11-05:
Cheers mate. I'll check with the folks over at the Arduino repo and see what they have to say about renaming the library. If it doesn't break the upgrade pathway, I'll rename it with the next release (which is only about a week away)

If it breaks the upgrade pathway, I'll keep the current version around for a little bit while I finish up the SPIMemory and then pull my version of SPIFlash from the Arduino Library manager.

LowPowerLab, 2018-01-16:
Ok I see this [arduino issue](https://github.com/arduino/Arduino/issues/6904) was closed, and this issue here is also closed, but the renaming hasn't happened.
Any updates? Shouldn't this be re-opened?

Marzogh, 2018-02-24:
Hi @LowPowerLab , my apologies for the delay in getting back to you. I was away - with not much access to the internet - for a while.

As I mentioned previously, I am working on a different library (SPIMEM) that I will be merging the next release of this library (v3.1.0) into. When v3.1.0 is released within the next couple of weeks, it will carry a deprecation notice and will direct all the users of this library to the new one. Once SPIMEM is released into the wild (which I'm hoping to do before the end of April at the latest), I will ask for this library (SPIFlash) to be pulled from the Arduino Library manager.

LowPowerLab, 2018-02-26:
Thanks for the update. Looking forward to a solution that works for everyone.

mihaigalos, 2018-03-27:
what about using a namespace for at least one of the libs?

Marzogh, 2018-03-27:
We could, but, the name change is underway anyway. I've put out the name change notice in the latest version. 🙂

Also, I'm working on including SPI EEPROM and SPI FRAM into the library over the next few months. So, changing the name to SPIMemory kinda fits. 😀

maxint-rd, 2018-04-15:
Cool! I'm looking forward to try out your new SPIMemory library with the chips I've got laying around (i.e. some various sized Atmel EEPROMS, a 2KB FRAM and some Winbond 4MB and 512KB chips.

closed 2018-05-17 by Marzogh

Marzogh, 2018-05-18:
All done @LowPowerLab you should be able to rename your library to SPIFlash now. :)