Marzogh/SPIMemory #84: 32-bit or 24-bit flash memory

haapeng, 2017-09-25:
Hi,
what I know from ur library support for 32-bit flash memory right?
Is there any possible u make library for 24-bit too?
Thanks mate.

Marzogh, 2017-11-04:
TBH, I haven't ever played around with a 24-bit flash memory chip. Any recommendations of what chips I can start with?

haapeng, 2017-11-06:
yeah maybe u can use M25P80 for example

Marzogh, 2017-11-06:
Thanks. :) I'll get my hands on one and see what I can do. :)

Marzogh, 2017-11-20:
I've finally gotten around to looking at the datasheet. Could you explain what you mean by 24-bit flash memory as opposed to 8-bit or 32-bit? As far as I can see the M25P80 has 8-bit data storage blocks - just like any other flash memory... (I'm confused)

haapeng, 2017-11-21:
Oh sorry for the confusion bro. what i meant was 24-bits or 32-bits address. for example cypress flash memory S25FL256S has 32-bits address but the m25p80 has 24-bits address.  so ur library can handle for both bits address?

Marzogh, 2017-11-21:
It is supposed to handle anything up to 32 bit addresses. 😊

Feel free to test it out and if it doesn’t work let me know and I’ll fix it up for you. I’ve got a few Micron flash memory chips in the mail...

Marzogh, 2017-11-21:
Also, the Winbond flash chip that the library was originally written for uses 23 bit addresses. 👍🏽

Marzogh, 2017-11-28:
@haapeng Mate, have you had a chance to test the library with the m25p80? 🙂

closed 2017-12-27 by Marzogh