Marzogh/SPIMemory #86: How to get the uniqueid?

rmhomecouk, 2017-10-21:
Hello, I'm looking to get the uniqueID out of the chip to use for ID purposes. The specific chip I'm using is the W25Q16JV which I can successfully read/write to. 

The dataheet http://www.winbond.com/resource-files/w25q16jv%20spi%20revf%2005092017.pdf

page 46:

 The Read Unique ID instruction is initiated by driving the /CS pin
low and shifting the instruction code “4Bh” followed by a four bytes of dummy clocks. 

But I have no idea how to add this! Any suggestions?

thanks!

Marzogh, 2017-11-04:
Check the latest code from the 'stable branch'. I've built-in a `flash.getUniqueID()` function into the library that will return the 64-bit Unique ID. If you'd like to print the number via Serial, check out the `printUniqueID()` function in the _Diagnostics_functions_ tab of the included example Diagnostics.ino. If you're going to manipulate it within your code, bear in mind that Arduino does not pay nice with 64-bit integers. You might have to come up with some creative math. 😄

closed 2017-11-04 by Marzogh

rmhomecouk, 2017-11-14:
hey @Marzogh thank you very much for this! awesome work :)

Marzogh, 2017-11-14:
Cheers, mate! Glad I could help.