Marzogh/SPIMemory #91: Can't write data into W25S256FV

thedub2001, 2017-11-05:
Hello!
I have plugged a Winbond W25S256FV and run the diagnostic sketch (FlashDiagnostics.ino in the library example sketch folder).
The output says that reading chip is OK but all writings have failed...
The pins WP, HOLD and RESET are HIGH so the chip should be working.

```
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
                                                                    SPIFlash Library version: 2.7.0
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
                                                                                Get ID
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
			JEDEC ID: ef4019h
			Manufacturer ID: efh
			Memory Type: 40h
			Capacity: 33554432 bytes
			Maximum pages: 131072
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
                                                                               Data Check
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
	Data Written	||	Data Read	||	Result		||	Write Time	||	Read Time	||	Write Time	||	Fast Read Time
			||			||			||			||			||	(No Error Chk)	||
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
	35		||	0		||	Fail		||	40 us		||	32 us		||	32 us		||	32 us
	-110		||	0		||	Fail		||	32 us		||	32 us		||	32 us		||	32 us
	4520		||	0		||	Fail		||	40 us		||	32 us		||	40 us		||	32 us
	-1250		||	0		||	Fail		||	40 us		||	32 us		||	40 us		||	32 us
	876532		||	0		||	Fail		||	40 us		||	40 us		||	40 us		||	40 us
	-10959		||	0		||	Fail		||	40 us		||	32 us		||	40 us		||	32 us
	3.14		||	0.00		||	Fail		||	2.8201 s		||	32 us		||	2.8201 s		||	32 us
	123 Test !@#	||		||	Fail		||	2907.1398 s		||	128 us		||	2907.1398 s		||	144 us
	inputStruct	||	outputStruct	||	Fail		||	56 us		||	32 us		||	40 us		||	32 us
```

thedub2001, 2017-11-06:
Comparing datasheets I have found a important difference in memory writing between chips >= 256Mbit and chips < 256Mbit

### W25Q256FV datasheet ; Section 6.1.5 : Byte Address Modes
The W25Q256FV provides two Address Modes that can be used to specify any byte of data in the memory array. The Byte Address Mode is backward compatible to older generations of serial flash memory that only support up to 128Mbit data.
To address the 256M bit or more data in 3 Byte Address Mode, Extended Address Register must be used in addition to the 3 Byte addresse.

see also section **7.2** of datasheet **Extended Address Register**

[w25q256 datasheet](https://www.winbond.com/resource-files/w25q256fv_revg1_120214_qpi_website_rev_g.pdf)
[w25q128 datasheet](https://www.winbond.com/resource-files/w25q128fv%20rev.l%2008242015.pdf)

Marzogh, 2017-11-07:
Thanks for raising the issue and pointing out the relevant sections of the datasheets. I'll get a patch to you ASAP. However, I'm afraid I won't be able to test the patch as I cannot seem to be able to get my hands on a W25Q256 chip. 😞  If I get the patch to you in the next 48 hours, could you test it for me please?

Also, where did you get your hands on a W25Q256?

thedub2001, 2017-11-07:
I will test it with great interest ! ;)
I found the chips on Aliexpress. Here is the link of the ones I bought (small price, quick delivery) : 
[W25Q256 on Aliexpress](https://fr.aliexpress.com/item/W25Q256-W25Q256FVFIG-25Q256FVFG-SOP-16/32789439851.html) ... I hope they are not counterfeit

Marzogh, 2017-11-08:
I've updated the code in the development branch with a (possible) patch. Could you test the code from there and let me know if it works? Thanks in advance! 👍 

(I've ordered some in from AliExpress and apparently they'll take a month to get to Australia! 😑 )

thedub2001, 2017-11-09:
Really nice work!! 😄 
Here is the serial output : 
```
Initialising Flash memory..........
Chip Diagnostics initiated.

No Chip size defined by user

--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
                                                                    SPIFlash Library version: 3.0.0
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
                                                                                Get ID
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
			JEDEC ID: ef4019h
			Manufacturer ID: efh
			Memory Type: 40h
			Capacity: 33554432 bytes
			Maximum pages: 131072
			Unique ID: 2200514064491264, 64A4A75B1B232300
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

			Erase Chip: 		Data I/O test PASS			Time: 65.938 s
			Erase 64KB Block: 	Data I/O test PASS			Time: 232 us
			Erase 32KB Block: 	Data I/O test PASS			Time: 288 us
			Erase 4KB Sector: 	Data I/O test PASS			Time: 272 us

			Byte: 			Data I/O test PASS			Write Time: 328 us,	Read Time: 136 us
			Char: 			Data I/O test PASS			Write Time: 328 us,	Read Time: 128 us
			Word: 			Data I/O test PASS			Write Time: 344 us,	Read Time: 136 us
			Short: 			Data I/O test PASS			Write Time: 352 us,	Read Time: 128 us
			ULong: 			Data I/O test PASS			Write Time: 376 us,	Read Time: 144 us
			Long: 			Data I/O test PASS			Write Time: 376 us,	Read Time: 152 us
			Float: 			Data I/O test PASS			Write Time: 376 us,	Read Time: 136 us
			Struct: 		Data I/O test PASS			Write Time: 136 us,	Read Time: 136 us
			Byte Array: 		Data I/O test PASS			Write Time: 5.112 ms,	Read Time: 840 us
			String: 		Data I/O test PASS			Write Time: 416 us,	Read Time: 152 us
			Power Down: 		Data I/O test PASS			Time: 168 us
			Power Up: 		Data I/O test PASS			Time: 144 us
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

```

Thanks for your efficiency!
Let me know if you need me to do more test with the W25Q256 chip before you receive it.
Usually the shipping times from Aliexpress are 2 weeks but can last longer...

Marzogh, 2017-11-09:
Fantastic! It's a nice surprise when the very first iteration works! 😀

Thank you for the kind offer. I will take you up on it in a few days - once I've got the the GM version of v3.0.0 ready. 🙂

Thanks again for bringing this bug to my notice. 👍🏽

thedub2001, 2017-11-09:
Yeah it was 65.938 intense seconds 😧 !
I don't close this issue as you can contact me later?

Marzogh, 2017-11-09:
The 65 seconds is comforting. One bug that had been annoying me through the last two releases was one where eraseChip() would fail randomly - especially with high capacity chips that had Chip erase cycles longer than 10 seconds (and, as I later discovered, with Cypress/Spansion chips). I finally nixed the bug with v3.0.0, but, the longest Chip erase cycle I could test it with was 20 seconds on a Cypress chip. Your test shows me my bugfix works for even longer erase cycles. 😁 (I'm waiting for a 1GB Chip to turn up in the mail in the next month - that'll be the acid rest. 🤔)

Keeping this issue open will definitely make it easier to contact you. Thanks. 👍🏽

closed 2017-11-17 by Marzogh