Marzogh/SPIMemory #92: 0x02 error

dunk8888, 2017-11-06:
Hi can you confirm this is how i would set this up.

#include<SPIFlash.h>

int ss = 10;
SPIFlash flash;

void setup() {
 Serial.begin(115200);
flash.begin(MB32); // is this correct
Serial.println("erasing chip");
flash.eraseChip();
delay(0);
flash.begin();
Serial.println(flash.readChar(22,44,false));// is this correct
}

void loop() {
 
}

dunk8888, 2017-11-06:
Hi,how would i write this array of bytes.
 data PROGMEM= [8] {233,23,33,45,15,255,234,255};
Its for an led controler and i need to store verry big patterns like above but 1000s of bytes in size. a quick demo would be great,ime not sure if i would need to increment pages or will it do it its self,if so can you show me that to,much appreciated thanks.
 Duncan

Marzogh, 2017-11-07:
This is how I'd do it:

```
#include<SPIFlash.h>

// int ss = 10; // You don't need this as you are not using a non-default SS pin as your slave select.

SPIFlash flash;
const int arraySize = 8;
const byte arrayData[arraySize] PROGMEM = [8] {233, 23, 33, 45, 15, 255, 234, 255};

void setup() {
  Serial.begin(115200);
  flash.begin(MB32);
  Serial.println("erasing chip");
  flash.eraseChip();
  Serial.println(flash.readChar(22, 44)); // Saying flash.readChar(22, 44, false) is superfluous as the default state of fastRead is false. If however you wanted to fastRead then you'd say flash.readChar(22, 44, true)

  // I'd recommend you do this to write your byte array
  // Use the getAddress function to get an address for your data and save it to a variable. Every time you use the getAddress function it will keep track of the blocks written to and make sure you don't corrupt your data.
  // If you want to write more data, just use the getAddress function again and include your data's size in bytes in brackets as the argument.
  uint32_t arrayAddress = flash.getAddress(arraySize);
  //Now write your byte array
  flash.writeByteArray(arrayAddress, arrayData, arraySize);
}

void loop() {

}
```

If you look at SPIFlash.h and look for the declaration of any read statement you'll see it says something like this - 
` int8_t   readChar(uint32_t address, bool fastRead = false); `
It means that the default value of fastRead is false, so, you only ever have to add the argument for it if you want to fastRead.
`readChar(address, true);` will fastRead a character.
`readChar(address);` is the same as `readChar(address, false);` and will perform a normal read.

fastRead does not really improve read speeds unless you are trying to move massive blocks of data, so make sure your data array is big enough to need it.

Marzogh, 2017-11-07:
Also, why did you title this issue 0x02 error? Are you getting that when you run Diagnostics.ino?

If yes, which flash memory chip are you using?

closed 2017-11-20 by Marzogh