Marzogh/SPIMemory #98: getAddress() isn't considering addresses which have data already written

vss1, 2017-11-16:
The function getAddress() is expected to return the next available address for use.

Say I have addresses 0 to 40 written with data already. Then I expect getAddress() to return 41. But it is returning 0.

**Output of getAddressEx.ino:** 
_Please note that I have data written till address 63
Just added an else condition with a print statement if any write in writeData() doesn't return 1_

The output is as follows:

Initialising Flash memory..........

Byte Address 0 : 0
Byte Address 1 : 1
Byte Address 2 : 2
Byte Address 3 : 3
Float Address 0 : 4
Float Address 1 : 8
String Address 0 : 12
String Address 1 : 30
String Address 2 : 48
Data write failed
Data write failed
Data write failed
Data write failed
Data write failed
Data write failed
Data write failed
Data write failed
Data write failed

The write wouldn't happen because the addresses returned already have data and this is expected.
I am using v2.7.0 of the library with a Winbond external flashchip.

Marzogh, 2017-11-17:
This function was originally written to only be used with fresh - i.e. blank - flash chips. However, I've  realised that many users might want to use this function the way you did. 

I've changed the way the function works - starting from v3.0.0. getAddress() can be called anytime in your sketch to get an address for your data - even if the chip already has data on it, or you have previously allocated addresses to other data. It will find the next available address block/s that will fit your data. 🙂 

Please test out the code from the stable branch and let me know if it works for you. 🙂

Marzogh, 2017-11-17:
Also, I've fixed getAddressEx.ino so it provides more clarity on what exactly it is doing. :)

commit 2ddfb190 (2017-11-17): getAddress() function now works anytime it is called - even if there is preexisting data on the chip. Fixes #98
closed 2017-11-17 by Marzogh