Marzogh/SPIMemory #99: Error 0x04

stanislauhalauko, 2017-11-16:
Dear All,

I work with w25q64 module and ESP32-WROOM-32 via arduino IDE. From few days w25q64 can't start work. I deleted almost all code and I want only begin the module, but all time i recieve error code 0x04. I chech all connection and tried different combination of connection with different pins.  
My connections:
CS -> IO5
CLK ->IO18
DI -> IO19
DO -> IO23

Please note that if I connect csPin to IO2 (as was said in extras in this repository) in ESP32 boatloader tell me 

> flash read err, 1000
Falling back to built-in command interpreter.
OK

And I should choose another pin, e.g. no 5. My code is below.

`[code]
#include<SPIFlash.h>
#include<SPI.h>

#define csPin 5

SPIFlash flash(csPin); 

void setup() {
Serial.begin(115200);
while (!flash.begin()){
  if (flash.error()) {
          Serial.print(flash.error());
     }
}
}

void loop() {
}
[/code]`

If someone know what i do wrong please let me know.

stanislauhalauko, 2017-11-18:
Problem solved by erasing the entire chip.
I connect chip to Arduino Uno, launched "TestFlash.ino" and ran "erases the entire chip". After that error 0x04 disappeared. And it's work with code above. 

But I don't know how fix it in more elegant way.

Marzogh, 2017-11-18:
Hi mate, I know the Adafruit ESP32 Feather I use for testing has a WROOOM-32 module and it works fine with the test code included with the library.

I have attempted to replicate your problem and the only way I can get the chip to lock up in a ‘busy’ state (i.e. error 0x04) is if I do what you have done

```
while (!flash.begin()) {
    if (flash.error()) {
        ........
    }
}
```

I’d recommend that you include a `delay(1000)` at the end of that loop. It seemed to fix the problem for me. (I’m still investigating why this is needed.) Otherwise, just do this instead
```
If (!flash.begin()) {
    Serial.println(flash.error());
}
```

Also to note is the fact that sometimes the flash memory chips just lock up in a ‘busy’ state when a large number of instructions are passed to them in a very small amount of time. The only way I have been able to unfreeze them is by disconnecting and reconnecting them to power. This happens quite regularly with Microchip flash but has happened frequently enough Winbond flash that it can be a problem.

closed 2017-11-20 by Marzogh