Marzogh/SPIMemory #102: SPIFlash::writeByteArray fails it's error check on SAMD

hanyazou, 2017-11-19:
SPIFlash::writeByteArray modify data buffer (storing data from SPI MISO into the buffer) and fail it's read after write verification. Write with error check always fail because of this.

>void SPIFlash::_nextBuf(uint8_t opcode, uint8_t *data_buffer, uint32_t size) {
    // snipped
    #elif defined (ARDUINO_ARCH_SAMD)
       #ifdef ENABLEZERODMA
         spi_write(&(*data_buffer), size);
       #else
         _spi->transfer(&(*data_buffer), size);      // this call destroy the contents of the buffer 
       #endif
     #elif defined (ARDUINO_ARCH_AVR)
       SPI.transfer(&(*data_buffer), size);

Marzogh, 2017-11-19:
Thanks for catching this. I'll check this out ASAP and fix it in the next bugfix release (v3.0.1)

Marzogh, 2017-11-20:
I can't replicate this error on any of the boards or the flash chips I'm using the library with. Both FlashDiagnostics.ino and TestFlash.ino return a SUCCESS after running writeByteArray and I am able to read the contents of the array with no issues at all.

Could you elaborate on the issue you're facing please? What board are you running this on?

hanyazou, 2017-11-20:
Sorry. It does not change the result. According to SPI::transfer() API document, it send out content of the buffer, receive data from target device and store data into the buffer.

https://www.arduino.cc/en/Reference/SPITransfer

I added some debug output to see it's behavior,

>    case PAGEPROG:
    #if defined (ARDUINO_ARCH_SAM)
      _dueSPISendByte(&(*data_buffer), size);
    #elif defined (ARDUINO_ARCH_SAMD)
      #ifdef ENABLEZERODMA
        spi_write(&(*data_buffer), size);
      #else
        mydump("BEFORE", data_buffer, 16);
        _spi->transfer(&(*data_buffer), size);                                                                                                                    
        mydump(" AFTER", data_buffer, 16);
      #endif

Results were

>_spi-&gt;transfer(&(*data_buffer), size);
This function will write a sequence of bytes (0-255) starting from your address of choice
Please enter the address you wish to write to: 1024
BEFORE 20000209: 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
 AFTER 20000209: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Data write failed

>_spi-&gt;transfer(&data_buffer[0], size);
This function will write a sequence of bytes (0-255) starting from your address of choice
Please enter the address you wish to write to: 2048
BEFORE 20000209: 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
 AFTER 20000209: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Data write failed

SPI::transfer re-write the buffer contents. My board is Arduino MKR1000.

Marzogh, 2017-11-22:
I'm unable to replicate your issue on an Adafruit Feather M0 which runs off the same SAMD core as the MKR1000. When I run TestFlash, this is my response.
```
----------------------------------------------------------------------------------------------------------------------------------
                                                       Function 8 : Write Page                                                    
                                                        SPIFlash library test                                                     
----------------------------------------------------------------------------------------------------------------------------------
This function will write a sequence of bytes (0-255) starting from your address of choice
Please enter the address you wish to write to: 44
Values from 0 to 255 have been written starting from the address 44
Type 1 to read the page you have just modified. Type 0 to continue: 1
Would you like your output in decimal or hexadecimal? Please indicate with '1' for HEX or '2' for DEC: 2
Reading address (002c)
000,001,002,003,004,005,006,007,008,009,010,011,012,013,014,015,
016,017,018,019,020,021,022,023,024,025,026,027,028,029,030,031,
032,033,034,035,036,037,038,039,040,041,042,043,044,045,046,047,
048,049,050,051,052,053,054,055,056,057,058,059,060,061,062,063,
064,065,066,067,068,069,070,071,072,073,074,075,076,077,078,079,
080,081,082,083,084,085,086,087,088,089,090,091,092,093,094,095,
096,097,098,099,100,101,102,103,104,105,106,107,108,109,110,111,
112,113,114,115,116,117,118,119,120,121,122,123,124,125,126,127,
128,129,130,131,132,133,134,135,136,137,138,139,140,141,142,143,
144,145,146,147,148,149,150,151,152,153,154,155,156,157,158,159,
160,161,162,163,164,165,166,167,168,169,170,171,172,173,174,175,
176,177,178,179,180,181,182,183,184,185,186,187,188,189,190,191,
192,193,194,195,196,197,198,199,200,201,202,203,204,205,206,207,
208,209,210,211,212,213,214,215,216,217,218,219,220,221,222,223,
224,225,226,227,228,229,230,231,232,233,234,235,236,237,238,239,
240,241,242,243,244,245,246,247,248,249,250,251,252,253,254,255,
----------------------------------------------------------------------------------------------------------------------------------
Please type the next command. Type 0 to get the list of commands

```

I'm not very sure what your mydump() function does. Could you show me what it is doing?

hanyazou, 2017-12-02:
I uploaded my source code on
https://github.com/hanyazou/SPIFlash/commit/fc5177f5af0afd1d8d1c6912bdfc36c0cc9c4f7a

SPI::transfer API document (https://www.arduino.cc/en/Reference/SPITransfer) says that

> the received data is stored in the buffer in-place (the old data is replaced with the data received).

To preserve the contents of data_buffer, I've replaced _spi->transfer(&data_buffer[0], size); with memcpy(buf, data_buffer, size) and _spi->transfer(buf, size); This is the output on my MKR1000 and S25FL123K.

```
----------------------------------------------------------------------------------------------------------------------------------
                                                       Function 8 : Write Page                                                    
                                                        SPIFlash library test                                                     
----------------------------------------------------------------------------------------------------------------------------------                                                     
This function will write a sequence of bytes (0-255) starting from your address of choice
Please enter the address you wish to write to: 0
BEFORE 20000209: 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
 AFTER 20000209: 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
transfered 256 bytes in 585 microsecs
Values from 0 to 255 have been written starting from the address 0
Type 1 to read the page you have just modified. Type 0 to continue: 1
Would you like your output in decimal or hexadecimal? Please indicate with '1' for HEX or '2' for DEC: 1
Reading address (0000)
000102030405060708090a0b0c0d0e0f
101112131415161718191a1b1c1d1e1f
202122232425262728292a2b2c2d2e2f
303132333435363738393a3b3c3d3e3f
404142434445464748494a4b4c4d4e4f
505152535455565758595a5b5c5d5e5f
606162636465666768696a6b6c6d6e6f
707172737475767778797a7b7c7d7e7f
808182838485868788898a8b8c8d8e8f
909192939495969798999a9b9c9d9e9f
a0a1a2a3a4a5a6a7a8a9aaabacadaeaf
b0b1b2b3b4b5b6b7b8b9babbbcbdbebf
c0c1c2c3c4c5c6c7c8c9cacbcccdcecf
d0d1d2d3d4d5d6d7d8d9dadbdcdddedf
e0e1e2e3e4e5e6e7e8e9eaebecedeeef
f0f1f2f3f4f5f6f7f8f9fafbfcfdfeff
----------------------------------------------------------------------------------------------------------------------------------
Please type the next command. Type 0 to get the list of commands
```

Original _spi->transfer(&data_buffer[0], size); is faster than my modified version. But it destroy the contents of data_buffer.

```
----------------------------------------------------------------------------------------------------------------------------------
                                                       Function 8 : Write Page                                                    
                                                        SPIFlash library test                                                     
----------------------------------------------------------------------------------------------------------------------------------
This function will write a sequence of bytes (0-255) starting from your address of choice
Please enter the address you wish to write to: 0
BEFORE 20000209: 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
 AFTER 20000209: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
transfered 256 bytes in 530 microsecs
Data write failed
----------------------------------------------------------------------------------------------------------------------------------
Please type the next command. Type 0 to get the list of commands
```

I've tested on my Feather M0 and S25FL132K too with Arduino IDE version 1.8.5 and Adafruit SAMD Boards by Adafruit version 1.0.20.

This is output of memcpy(buf, data_buffer, size) and _spi->transfer(buf, size) version.
```
----------------------------------------------------------------------------------------------------------------------------------
                                                       Function 8 : Write Page                                                    
                                                        SPIFlash library test                                                     
----------------------------------------------------------------------------------------------------------------------------------
This function will write a sequence of bytes (0-255) starting from your address of choice
Please enter the address you wish to write to: 0
[... 17 lines cut ...]
a0a1a2a3a4a5a6a7a8a9aaabacadaeaf
b0b1b2b3b4b5b6b7b8b9babbbcbdbebf
c0c1c2c3c4c5c6c7c8c9cacbcccdcecf
d0d1d2d3d4d5d6d7d8d9dadbdcdddedf
e0e1e2e3e4e5e6e7e8e9eaebecedeeef
f0f1f2f3f4f5f6f7f8f9fafbfcfdfeff
----------------------------------------------------------------------------------------------------------------------------------
Please type the next command. Type 0 to get the list of commands
```

This is output of original _spi->transfer(&data_buffer[0], size) version.
```
----------------------------------------------------------------------------------------------------------------------------------
                                                       Function 8 : Write Page                                                    
                                                        SPIFlash library test                                                     
----------------------------------------------------------------------------------------------------------------------------------
[... cut ...]  a sequence of bytes (0-255) starting from your address of choice
Please enter the address you wish to write to: 0
BEFORE 20000209: 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
 AFTER 20000209: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
transfered 256 bytes in 493 microsecs
Data write failed
----------------------------------------------------------------------------------------------------------------------------------
Please type the next command. Type 0 to get the list of commands
```

Marzogh, 2017-12-10:
Duh! Silly me. I cocked this one up proper! 😨  Thanks for catching it @hanyazou . 🙂 I've fixed it and pushed the changes through. I'll roll out the bugfix release in an hour or so. 🙂

commit 7799639d (2017-12-10): Fixes #102
commit 6d88fb49 (2017-12-10): Merge pull request #105 from Marzogh/development

Bugfix PR - resolves #102
closed 2017-12-10 by Marzogh