Marzogh/SPIMemory #107: Bug report : can't write ByteArray if address is 0 of multiple of 256

thedub2001, 2017-12-11:
**Hello!**

I'm submitting a bug report, I tried to identify what's going wrong but I cannot fix it 

### Bug report :
**Library version** : 3.0.0
**Branch** : master
**Arduino ID**E : 1.8.4
**OS** : Windows
**Microcontroller** : Arduino-nano (Atmega328P)
**Flash memory module** : W25Q64 freshly erased

**Current behaviour** :
Using the function ```flash.writeByteArray(addr, &pageBuffer[0], SPI_PAGESIZE)``` with ```addr=0 or multiple of 256``` returns ```0``` (write failed)

**Expected behaviour** : 
The function should return ```1``` (write success)

**Code snippet that reproduces the bug** :
```arduino
#include<SPIFlash.h>
uint8_t pageBuffer[256];
String serialCommand;
char printBuffer[128];
uint32_t addr;
uint8_t dataByte;
uint16_t dataInt;
String inputString, outputString;

//Define a flash memory size (if using non-Winbond memory) according to the list in defines.h
//#define CHIPSIZE MB64

#define BAUD_RATE 9600

//SPIFlash flash(SS1, &SPI1);       //Use this constructor if using an SPI bus other than the default SPI. Only works with chips with more than one hardware SPI bus
SPIFlash flash;

void setup() {
  delay(10);
  Serial.begin(BAUD_RATE);
  Serial.print(F("Initialising Flash memory"));
  for (int i = 0; i < 10; ++i)
  {
    Serial.print(F("."));
  }
  Serial.println();

  flash.begin();

  Serial.println();
  Serial.println();
  Serial.println(F("Erasing chip in 10s"));
  
  for (int i=10;i>0;i--){
    Serial.println(i);
    delay(1000);
  }
    Serial.println(F("Erasing chip..."));

  if (flash.eraseChip())
    Serial.println(F("Chip erased"));
  else
    Serial.println(F("Error erasing chip"));
  uint32_t capacity = flash.getCapacity();
  
    for (long k=0;k<capacity;k=k+256) {
      Serial.println();
      addr = k;
      Serial.print(addr,HEX);
      Serial.print(" : ");
      Serial.println(addr);
      
      for (uint16_t i = 0; i < SPI_PAGESIZE; ++i) {
        pageBuffer[i] = i;
      }
      
      if (flash.writeByteArray(addr, &pageBuffer[0], SPI_PAGESIZE)) {
        }
      else {
        Serial.println("failed writing");
      }
    }
}

void loop() {
}
```
I began to investigate and I found that the problem occurs during the errorCheck process of the writeByteArray() function in SPIFlash.cpp : 
```arduino
if (!errorCheck) {
    _endSPI();
    #ifdef RUNDIAGNOSTIC
      _spifuncruntime = micros() - _spifuncruntime;
    #endif
    return true;
  }
  else {
    if (!_notBusy()) {
      return false;
    }
    _currentAddress = _addr;
    CHIP_SELECT
    _nextByte(WRITE, READDATA);
    _transferAddress();
    for (uint16_t j = 0; j < bufferSize; j++) {
      if (_nextByte(READ) != data_buffer[j]) {
        return false;
      }
    }
    _endSPI();
    #ifdef RUNDIAGNOSTIC
      _spifuncruntime = micros() - _spifuncruntime;
    #endif
    return true;
  }
```
**the data_buffer[j] stays to 255 while it should represent the written byte array : [0,1,2,3...]. That's why the function returns ```false``` at the first ocurence of the for loop**

thedub2001, 2017-12-27:
I made some more investigations

Looks like the ```void transfer(void *buf, size_t count)``` function in ```SPI.h``` overwrites the buffer with the return data. 
This function is called by ```_nextBuf``` which is called by ```writeByteArray``` function in ```SPIFlash.cpp```

As I don't really understand the methods for passing variables as parameters by pointer with ```*``` or by reference ```&``` I can only tinker.

The temporary solution I found is using a temporary buffer in which I copy the ```data_buffer``` (in the ``` writeByteArray``` function)
``` arduino
bool SPIFlash::writeByteArray(uint32_t _addr, uint8_t *data_buffer, size_t bufferSize, bool errorCheck) {

(...code...)

  byte tempBuffer[bufferSize];
  for (uint16_t j = 0; j < bufferSize; j++) {
    tempBuffer[j] = data_buffer[j];
  }

  if (bufferSize <= maxBytes) {
    CHIP_SELECT
    _nextByte(WRITE, PAGEPROG);
    _transferAddress();
    _nextBuf(PAGEPROG, &tempBuffer[0], bufferSize);
    CHIP_DESELECT
  }

(...code...)

}
```

Marzogh, 2017-12-27:
Hi mate, Thanks for catching this one. 👍  @hanyazou caught it in #102 and it has been fixed in [v3.0.1](github.com/Marzogh/SPIFlash/releases/tag/v3.0.1). 🙂 

My apologies for the delay in replying - I'm currently away bushwalking in remote Tasmania and rarely have access to the internet...

Marzogh, 2017-12-27:
Duplicate of #102

thedub2001, 2017-12-27:
Bushwalking in remote Tasmania sounds great!! 😀  Enjoy your trip!

Marzogh, 2017-12-27:
Ugh! Silly me. I just realised I had not taken a look at the original issue - about problems writing to address 0 or multiple of 256. I'll look at it as soon as I get back from Tassie.