Marzogh/SPIMemory #108: Reqire whileWait to flash led instead of time out.

dunk8888, 2017-12-14:
Hi,is it possible to flash an led while the flash is busy instead of or with timeout,Can you implement this in the library,would be a great help to me if while a bulk erase is in progress,thanks

Hey there! Thanks for using the SPIFlash library for Arduino. 

**Note: For support questions, please use the [Arduino Forums](http://forum.arduino.cc/index.php?topic=324009.0). This repository's issues are reserved for feature requests and bug reports.**

**I'm submitting a ...**
  - [ ] bug report
  - [ ] feature request
  - [ ] support request => Please do not submit support request here, see note at the top of this template.
  
Do the checklist before filing an issue:

- [ ] Is this something you can **debug and fix**? Send a pull request! Bug fixes and documentation fixes are welcome.
- [ ] Is this an idea for a feature? Post the feature request as an issue here with a title that clearly says "Feature Request".
- [ ] Is this a bug that you cannot fix? Post the bug report as an issue here with a title that cleary says "Bug Report".
  
When opening an issue please include the following details:
-------------------------------------------------------------

- [ ] Do you want to request a *feature* or report a *bug*?
- [ ] Which library version are you using?
- [ ] Which Arduino IDE version are you compiling the library on?
- [ ] Which OS are you compiling your code on? [Linux / Mac / Windows]
- [ ] Which microcontroller platform are you using?
- [ ] Which Flash memory module are you using?
- [ ] What is the current behavior?
- [ ] What is the expected behavior?

--------------------------
###### Bug reports only

- [ ] If this is a bug report - Provide a **minimal code snippet** example that reproduces the bug. Please make sure you wrap any code in the proper code blocks like below 
```
```CODE HERE```
```
- [ ] Provide **screenshots** where appropriate
- [ ] Provide **wiring diagram** where appropriate

-----------------------------
###### Feature Requests only

- [ ] If this is a feature request, what is the motivation / use case for changing the behavior?

----------------------------

- [ ] Other information (e.g. detailed explanation, related issues, suggestions how to fix, links for us to have context etc.)

_Make sure to add **all the information needed to understand the bug/feature** so that someone can help. If the info is missing we'll add the 'Needs more information' label and close the issue until there is enough information._

Marzogh, 2017-12-27:
Hi mate, its not a bad idea at all. :) Thanks for raising this.

Rather than trying to blink an LED - which might prove self-limiting in the end, I think it might be better to give the end-user access to a public variable that tells you if the flash chip is busy. That way, you can write a simple function in your code to blink an LED while the chip is busy. 🙂 Others might want to use this as a signal to execute other functions.

I'll roll this into the next version...

Marzogh, 2018-03-02:
Hi mate, I can't think of a way of doing this that would not interrupt library functions. The only workaround would be this (for your specific case):

Write up a function to blink an LED on the board you are using in your project. Replace the function `_notBusy(uint32_t timeout)` in `FLASHIO.cpp` with this

```
  // Polls the status register 1 until busy flag is cleared or timeout
bool SPIFlash::_notBusy(uint32_t timeout) {
  _delay_us(WINBOND_WRITE_DELAY);
  uint32_t _time = micros();

  do {
    _readStat1();
    if (!(stat1 & BUSY))
    {
      return true;
    }
    _time++;
  } while ((micros() - _time) < timeout);
  if (timeout <= (micros() - _time)) {
    digitalWrite(_LEDPIN_, HIGH);
    _delay_us(_DELAYINMICROSECONDS_);
    digitalWrite(_LEDPIN_, LOW);
 return false;
  }
  return true;
}
```

Make sure you declare your _LEDPIN_ and instantiate it in your code by setting the pinMode to output in `void setup()`. Also, set your blink delay by defining a value (in microseconds) for _DELAYINMICROSECONDS_

 I'm open to ideas if you can figure out a way to do this without slowing the library down enormously. 🙂

closed 2018-03-02 by Marzogh