Marzogh/SPIMemory #109: Error code: 0x03 for winbon 25q64fvsig

duraiyuvaraj, 2017-12-24:
Hi,
Please help me to resolve.
regards
yuvaraj

Marzogh, 2017-12-24:
**Mate, please submit the issue following the template given below:**

> Hey there! Thanks for using the SPIFlash library for Arduino. 
> 
> **Note: For support questions, please use the [Arduino Forums](http://forum.arduino.cc/index.php?topic=324009.0). This repository's issues are reserved for feature requests and bug reports.**
> 
> **I'm submitting a ...**
>   - [ ] bug report
>   - [ ] feature request
>   - [ ] support request => Please do not submit support request here, see note at the top of this template.
>   
> Do the checklist before filing an issue:
> 
> - [ ] Is this something you can **debug and fix**? Send a pull request! Bug fixes and documentation fixes are welcome.
> - [ ] Is this an idea for a feature? Post the feature request as an issue here with a title that clearly says "Feature Request".
> - [ ] Is this a bug that you cannot fix? Post the bug report as an issue here with a title that cleary says "Bug Report".
>   
> When opening an issue please include the following details:
> -------------------------------------------------------------
> 
> - [ ] Do you want to request a *feature* or report a *bug*?
> - [ ] Which library version are you using?
> - [ ] Which Arduino IDE version are you compiling the library on?
> - [ ] Which OS are you compiling your code on? [Linux / Mac / Windows]
> - [ ] Which microcontroller platform are you using?
> - [ ] Which Flash memory module are you using?
> - [ ] What is the current behavior?
> - [ ] What is the expected behavior?
> 
> --------------------------
> ###### Bug reports only
> 
> - [ ] If this is a bug report - Provide a **minimal code snippet** example that reproduces the bug. Please make sure you wrap any code in the proper code blocks like below 
> ```
> ```CODE HERE```
> ```
> - [ ] Provide **screenshots** where appropriate
> - [ ] Provide **wiring diagram** where appropriate
> 
> -----------------------------
> ###### Feature Requests only
> 
> - [ ] If this is a feature request, what is the motivation / use case for changing the behavior?
> 
> ----------------------------
> 
> - [ ] Other information (e.g. detailed explanation, related issues, suggestions how to fix, links for us to have context etc.)
> 
> _Make sure to add **all the information needed to understand the bug/feature** so that someone can help. If the info is missing we'll add the 'Needs more information' label and close the issue until there is enough information._

 I can't do anything with the information you have given me and I'd rather have all the information before I start to figure out what's going wrong.

duraiyuvaraj, 2017-12-24:
I think, I should have submitted to Arduino forum. Thank you for for your reply.

closed 2017-12-24 by duraiyuvaraj