Marzogh/SPIMemory #110: feature request : prevent SPIflash functions from running if chip is in powerDown mode

thedub2001, 2017-12-27:
Hello !
Is it's a good idea to prevent SPIFlash function from running while in powerDown mode ?
What drove me to this question is that SPIFlash function are not working and give no error message if the previous sketch had put the chip in powerDown mode.
This is a non-blocking issue thus not really important.

Marzogh, 2017-12-28:
Why not? That's easy enough to set up. I'll roll it into the next version 🙂

Marzogh, 2018-03-02:
Hi mate, I've updated the code development branch to do as you requested. From v3.1.0 on, when any I/O function is called the library also checks an internal variable to see if the chip is powered down and returns an error if it is. The error codes have been updated to match too. 🙂

Marzogh, 2018-03-02:
As of 9af3d118ec8d331ee6a02ca8da9d3db223a20193 erase functions check to see if chip is powered down as well.

commit 95b43f42 (2018-03-26): Fixes #110