Marzogh/SPIMemory #115: Bug report : W25Q64JV not working.

kucukkose, 2018-02-26:
Hi

I'm using this library with W25Q64BV and Arduino MEGA correctly .
I upgraded my flash memory with W25Q64JV. But now library doesn't work.
What could be the problem?

Marzogh, 2018-02-28:
Hi mate, I have absolutely no idea what the problem is unless you provide me with more information than 'it is not working'. 

When filing a bug report please follow the template that pops up when you open a new issue. That way we can save a whole bunch of time asking and answering basic questions about the problem. 

I have attached the template you will need to follow to the end of my reply. Copy the template into your reply and fill in the details of the problems you are facing. Before you do that though, please make sure you have run FlashDiagnostics.ino with ```#define RUNDIAGNOSTICS``` uncommented in SPIFlash.h. Also please list any error codes that pop up in your Serial output when you run FlashDiagnostics.ino.

------------------------------------ Template begins -------------------------------------------
```
Hey there! Thanks for using the SPIFlash library for Arduino. 

**Note: For support questions, please use the [Arduino Forums](http://forum.arduino.cc/index.php?topic=324009.0). This repository's issues are reserved for feature requests and bug reports.**

**I'm submitting a ...**
  - [ ] bug report
  - [ ] feature request
  - [ ] support request => Please do not submit support request here, see note at the top of this template.
  
Do the checklist before filing an issue:

- [ ] Is this something you can **debug and fix**? Send a pull request! Bug fixes and documentation fixes are welcome.
- [ ] Is this an idea for a feature? Post the feature request as an issue here with a title that clearly says "Feature Request".
- [ ] Is this a bug that you cannot fix? Post the bug report as an issue here with a title that cleary says "Bug Report".
  
When opening an issue please include the following details:
-------------------------------------------------------------

- [ ] Do you want to request a *feature* or report a *bug*?
- [ ] Which library version are you using?
- [ ] Which Arduino IDE version are you compiling the library on?
- [ ] Which OS are you compiling your code on? [Linux / Mac / Windows]
- [ ] Which microcontroller platform are you using?
- [ ] Which Flash memory module are you using?
- [ ] What is the current behavior?
- [ ] What is the expected behavior?

--------------------------
###### Bug reports only

- [ ] If this is a bug report - Provide a **minimal code snippet** example that reproduces the bug. Please make sure you wrap any code in the proper code blocks like below 
```
```CODE HERE```
```
- [ ] Provide **screenshots** where appropriate
- [ ] Provide **wiring diagram** where appropriate

-----------------------------
###### Feature Requests only

- [ ] If this is a feature request, what is the motivation / use case for changing the behavior?

----------------------------

- [ ] Other information (e.g. detailed explanation, related issues, suggestions how to fix, links for us to have context etc.)

_Make sure to add **all the information needed to understand the bug/feature** so that someone can help. If the info is missing we'll add the 'Needs more information' label and close the issue until there is enough information._

```

------------------------------------- Template ends --------------------------------------------

Marzogh, 2018-04-16:
Hi mate, I've finally managed to get my hands on some W25Q64JVs and the current version of the library (v3.1.0) works as it should. Every function passes in Diagnostics.ino. Could you please make sure you are wiring the chip right?

closed 2018-04-16 by Marzogh

kucukkose, 2018-05-26:
Hi Marzogh,

I pulled up pin3 and pin7 and now every thing works fine. 

Thanks for your support.

Marzogh, 2018-05-27:
Fantastic. I'm glad it had a simple solution. 🙂

intensite, 2020-02-17:
Hello gentlemen,

I am not sure if this is the proper place for this, if not please let me know where I should ask.

I am experiencing a similar issues with this chip W25Q64JVSSIQ.  I am using a ESP32 DevkitV4 and the DEV branch of the SPIMemory library.

I have attached my schematics bellow.  

As you can see, I have 10k ohm pull-ups on CS, MOSI and IO3.  (I had to remove the pull-up from MISO as the GPIO12 of the ESP32 apparently has to be low to boot).    The CLK pin is pulled down with a 100k resistor.

Durring the FlashDiagnostics execution, I get the "Check your wiring. Flash chip is non-responsive.
" error message.

I traced it to the SPIFlash::_chipID function call (specifically _getJedecId)

My questions:

1. Is this particular flash chip supported by the library?
2. According to your experience, is my wiring OK?

Best regards,

[image]