Marzogh/SPIMemory #116: BluePill Support request

KingJimmy27, 2018-03-08:
Hey there! Thanks for using the SPIFlash library for Arduino. 
### Please note that starting 01.03.2018 any issue raised here MUST be submitted according to this template or it will be flagged with 'Not enough information'. No action will be taken till all the prerequisite information is provided. If no information is provided for over a month, the issue will be closed.

**Note: For support questions, please use the [Arduino Forums](http://forum.arduino.cc/index.php?topic=324009.0). This repository's issues are reserved for feature requests and bug reports.**

# Issue details

**I'm submitting a ...**
  - [ ] bug report
  - [BluePill compatability ] feature request
  - [ ] support request => Please do not submit support request here, see note at the top of this template.
  
Do the checklist before filing an issue:

- [No ] Is this something you can **debug and fix**? Send a pull request! Bug fixes and documentation fixes are welcome.
- [ ] Is this an idea for a feature? Post the feature request as an issue here with a title that clearly says "Feature Request".
- [ ] Is this a bug that you cannot fix? Post the bug report as an issue here with a title that cleary says "Bug Report".
  
When opening an issue please include the following details:
-------------------------------------------------------------

- [request a feature ] Do you want to request a *feature* or report a *bug*?
- [(v3.0.0 ] Which library version are you using?
- [1.8.5 ] Which Arduino IDE version are you compiling the library on?
- [win 10 ] Which OS are you compiling your code on? [Linux / Mac / Windows]
- [BluePill ] Which microcontroller platform are you using?
- [LC W25Q80B memory module ] Which Flash memory module are you using?
- [FlashDiagnostics example does nor work ] What is the current behavior?
- [FlashDiagnostics ] What is the expected behavior?

--------------------------
###### Bug reports only
- If this is a bug report - 

- [ ] Make sure you have run FlashDiagnostics.ino with ``` #define RUNDIAGNOSTICS ``` uncommented in SPIFlash.h. **List any error codes** that pop up in your Serial output when you run FlashDiagnostics.ino.here:
      Error codes
      ------------
        - 
        -
        -
        -
        -
- [ ] If you have a problem with a particular function, call the flash.error() function (after you have made sure you have started up your Serial port with a ``` Serial.begin(BAUD) ``` ). Provide details of the function, the data given to/ expected from the function and the error code here: (**Please repeat this for every function you have an error with***)
        Function - 
        Data - 
        Error code - 
- [ ] Provide a **minimal code snippet** example that reproduces the bug. Please make sure you wrap any code in the proper code blocks like below 
```
```CODE HERE```
```
- [ ] Provide **screenshots** where appropriate
- [ ] Provide **wiring diagram** where appropriate

-----------------------------
###### Feature Requests only

- [ ] If this is a feature request, what is the motivation / use case for changing the behavior?

----------------------------

- [I am using http://www.visualmicro.com/ to build my projects and not the Arduino ide, however the Visual Micro does use the Arduino software.
I never use a bootloader only a cheap ISP like STM-LINK V2 ] Other information (e.g. detailed explanation, related issues, suggestions how to fix, links for us to have context etc.)

_Make sure to add **all the information needed to understand the bug/feature** so that someone can help. If the info is missing we'll add the 'Needs more information' label and close the issue until there is enough information._

Marzogh, 2018-08-10:
The code in the development branch as of today's commit (703f609781c7d8e866f79f415ef5e254cb324799) works with the BluePill. The next release - v3.3.0 will come with official support for this platform. 🙂

closed 2018-08-10 by Marzogh

FarazArdu, 2018-10-30:
hi, do the library supports stm32 Blue Pill ? thanks