Marzogh/SPIMemory #120: STM32F103RC support for S25FL216K0PMFI011

okwestern, 2018-03-27:
Hey there! Thanks for using the SPIFlash library for Arduino. 
### Please note that starting 01.03.2018 any issue raised here MUST be submitted according to this template or it will be flagged with 'Not enough information'. No action will be taken till all the prerequisite information is provided. If no information is provided for over a month, the issue will be closed.

**Note: For support questions, please use the [Arduino Forums](http://forum.arduino.cc/index.php?topic=324009.0). This repository's issues are reserved for feature requests and bug reports.**

# Issue details

**I'm submitting a ...**
  - [ ] bug report
  - [X] feature request
  - [ ] support request => Please do not submit support request here, see note at the top of this template.
  
Do the checklist before filing an issue:

- [ ] Is this something you can **debug and fix**? Send a pull request! Bug fixes and documentation fixes are welcome.
- [X] Is this an idea for a feature? Post the feature request as an issue here with a title that clearly says "Feature Request".
- [ ] Is this a bug that you cannot fix? Post the bug report as an issue here with a title that cleary says "Bug Report".
  
When opening an issue please include the following details:
-------------------------------------------------------------

- [X] Do you want to request a *feature* or report a *bug*? Feature request
- [X] Which library version are you using? 3.1.0
- [X] Which Arduino IDE version are you compiling the library on? 1.8.5
- [X] Which OS are you compiling your code on? [Linux / Mac / Windows] Mac OS X High Sierra
- [X] Which microcontroller platform are you using? STM32F103RC
- [X] Which Flash memory module are you using? S25FL216K0PMFI011 and S25FL208K0RMFI041
- [X] What is the current behavior? None of the examples work
- [X] What is the expected behavior? Normal behaviour

--------------------------
###### Bug reports only
- If this is a bug report - 

- [ ] Make sure you have run FlashDiagnostics.ino with ``` #define RUNDIAGNOSTICS ``` uncommented in SPIFlash.h. **List any error codes** that pop up in your Serial output when you run FlashDiagnostics.ino.here:
      Error codes
      ------------
        - 
        -
        -
        -
        -
- [ ] If you have a problem with a particular function, call the flash.error() function (after you have made sure you have started up your Serial port with a ``` Serial.begin(BAUD) ``` ). Provide details of the function, the data given to/ expected from the function and the error code here: (**Please repeat this for every function you have an error with***)
        Function - 
        Data - 
        Error code - 
- [ ] Provide a **minimal code snippet** example that reproduces the bug. Please make sure you wrap any code in the proper code blocks like below 
```
```CODE HERE```
```
- [ ] Provide **screenshots** where appropriate
- [ ] Provide **wiring diagram** where appropriate

-----------------------------
###### Feature Requests only

- [X] If this is a feature request, what is the motivation / use case for changing the behavior?
Arduino is moving towards ARM, and adding support for the very cheap and available STM32F103-family would be beneficial for many users.
----------------------------

- [X] Other information (e.g. detailed explanation, related issues, suggestions how to fix, links for us to have context etc.)
A feature request (#116) to support BluePill (STM32F103C8) has already been proposed. Unless STM32F103C8 and STM32F103RC are different, these requests will be solved together.

_Make sure to add **all the information needed to understand the bug/feature** so that someone can help. If the info is missing we'll add the 'Needs more information' label and close the issue until there is enough information._

Marzogh, 2018-03-27:
Hi mate, I'm working on the blue pill at the moment and I'm happy to add this to the list. I have a couple of questions for you:

- Which board are you currently using with the STM32F103RC? I'd like to get my hands on a dev board to work with.

- Also, which board definitions are you using?

okwestern, 2018-03-27:
Hi Marzogh,

 

I've developed my own board from scratch with a lot of onboard peripherals, including Spansion/Cypress SPI flash.

I'm using board definitions from stm32duino.com.

Marzogh, 2018-04-12:
Duplicate of #116

okwestern, 2018-06-06:
Seems it mostly works for F103RC now. A few bugs - I've opened a new issue.

closed 2018-06-06 by okwestern