Marzogh/SPIMemory #122: Bug Report - W25Q128FV Not Responding At All

rsispal, 2018-04-08:
# Issue details

**I'm submitting a ...**
  - [x] bug report
  - [ ] feature request
  - [ ] support request => Please do not submit support request here, see note at the top of this template.
  
Do the checklist before filing an issue:

- [ ] Is this something you can **debug and fix**? Send a pull request! Bug fixes and documentation fixes are welcome.
- [ ] Is this an idea for a feature? Post the feature request as an issue here with a title that clearly says "Feature Request".
- [x] Is this a bug that you cannot fix? Post the bug report as an issue here with a title that cleary says "Bug Report".
  
When opening an issue please include the following details:
-------------------------------------------------------------

- [ ] Do you want to request a *feature* or report a *bug*? ```BUG```
- [ ] Which library version are you using? ```3.1.0```
- [ ] Which Arduino IDE version are you compiling the library on? ```PlatformIO Core 3.5.3a8 and Arduino 1.85```
- [ ] Which OS are you compiling your code on? [Linux / Mac / Windows] ```Mac```
- [ ] Which microcontroller platform are you using? ```Adafruit Feather M0 with ATWINC1500```
- [ ] Which Flash memory module are you using? ```W25Q128FV```
- [ ] What is the current behavior? ```Library does not interact successfully at all```
- [ ] What is the expected behavior? ```The example sketches to at least interact with the flash as per their function```

--------------------------
###### Bug reports only
- If this is a bug report - 

- [ ] Make sure you have run FlashDiagnostics.ino with ``` #define RUNDIAGNOSTICS ``` uncommented in SPIFlash.h. **List any error codes** that pop up in your Serial output when you run FlashDiagnostics.ino.here:
      Error codes
      ------------
        - What error codes? Just wall text of fails is displayed - see attached text file for full output
        -
        -
        -
        -
- [ ] If you have a problem with a particular function, call the flash.error() function (after you have made sure you have started up your Serial port with a ``` Serial.begin(BAUD) ``` ). Provide details of the function, the data given to/ expected from the function and the error code here: (**Please repeat this for every function you have an error with***)
        Function - 
        Data - 
        Error code - 
- [ ] Provide a **minimal code snippet** example that reproduces the bug. Please make sure you wrap any code in the proper code blocks like below 
```
```CODE HERE```
```
- [ ] Provide **screenshots** where appropriate
- [ ] Provide **wiring diagram** where appropriate

-----------------------------
###### Feature Requests only

- [ ] If this is a feature request, what is the motivation / use case for changing the behavior?

----------------------------

- [ ] Other information (e.g. detailed explanation, related issues, suggestions how to fix, links for us to have context etc.)

_Make sure to add **all the information needed to understand the bug/feature** so that someone can help. If the info is missing we'll add the 'Needs more information' label and close the issue until there is enough information._

 ---------------------------
# Description of Issue:

So, I am not sure whether it's something I have missed (probably is)- but how come there's no reference to the HOLD or Chip Select pins? I've looked through the included examples and there doesn't seem to be any reference to them, nor in your pinout diagram in `extras`.  At least for anything other than ESP's.

I have, however, confirmed all my flash chips work, using the following library:
https://github.com/PaulStoffregen/SerialFlash

Happy to help debug the issue as I am able to get them working with the above referenced library, but I do think it's the Chip Select line. Some end-to-end documentation wouldn't go a miss for successfully wiring a chip up to an Arduino for those who aren't experienced.

Marzogh, 2018-04-20:
The Hold pin needs to be pulled up to VCC (if it is active low or vice verse if otherwise).

The Chip Select pin has to be connected to a pin of your choice on your dev board. If you are connecting to a pin that is not the default Slave Select pin for the SPI bus, then you should indicate which pin you have selected as the CS pin when you call the constructor - i.e if you select pin 8, your constructor should read SPIFlash flash(8);

Could you give me the details of your wiring?

Marzogh, 2019-04-20:
It has been over a year and no new information (requested) has been added. I'm closing this issue for now.

closed 2019-04-20 by Marzogh