Marzogh/SPIMemory #125: feature request: Add different CHIPERASE values

dberenguer, 2018-04-13:
SST26VF064B uses 0xC7 as the CHIPERASE command instead of 0x60. I suggest passing some config values to the constructor or maybe dynamically decide the CHIPERASE command based on the type of memory IC.

Marzogh, 2018-04-13:
This is something I'm currently working on. 🙂 The code in the development branch can now identify if a chip supports SFDP. (The SST26 series does). I will be adding the ability to get the opcodes from the chip's SFDP tables by the time the next release rolls around. 🙂

Marzogh, 2018-04-24:
All done and sorted. 😀 The code from the development branch now automatically switches to 0xC7 on the SST26 series and on other ships that support this opcode. I will roll this into v3.2.0.

commit 10c2b863 (2018-04-24): - Fixes #125 . The library automatically uses the alternate chip erase code (0xC7) as required (only on fully supported chips)
- Adds support for a number of flash chips - S25FL032P, AT25SF041,
M25P40 & A25L512A0.
closed 2018-04-24 by Marzogh