Marzogh/SPIMemory #128: Xplained Pro SAMD21 SPIFlash @SPI2 only write fails

StephiB, 2018-04-25:
Hi
# Issue Details
reading from Flash works but no writing possible

**I'm submitting a ...**
  - [ x] bug report
  - [ ] feature request
  - [ ] support request => Please do not submit support request here, see note at the top of this template.
  
Do the checklist before filing an issue:

- [ ] Is this something you can **debug and fix**? Send a pull request! Bug fixes and documentation fixes are welcome.
- [ ] Is this an idea for a feature? Post the feature request as an issue here with a title that clearly says "Feature Request".
- [x ] Is this a bug that you cannot fix? Post the bug report as an issue here with a title that cleary says "Bug Report".
  
When opening an issue please include the following details:
-------------------------------------------------------------

- [ bug] Do you want to request a *feature* or report a *bug*?
- [3.1.0 ] Which library version are you using?
- [1.8.5 ] Which Arduino IDE version are you compiling the library on?
- [Win10 ] Which OS are you compiling your code on? [Linux / Mac / Windows]
- [Xplained Pro SAMD21 ] Which microcontroller platform are you using?
- [AT25DF081A-SSH-T ] Which Flash memory module are you using?
- [all but write does work ] What is the current behavior?
- [ write should work] What is the expected behavior?

--------------------------
###### Bug reports only
- If this is a bug report - 

- [ ] Make sure you have run FlashDiagnostics.ino with ``` #define RUNDIAGNOSTICS ``` uncommented in SPIFlash.h. **List any error codes** that pop up in your Serial output when you run FlashDiagnostics.ino.here:
      Error codes

			JEDEC ID: 1f4501h
			Manufacturer ID: 1fh
			Memory Type: 45h
			Capacity: 8388608 bytes
			Maximum pages: 32768
			Unique ID: 00, 0x00
			Erase Chip: 		Data I/O test PASS			Time: 62 us
			Erase 72KB Section: 	Data I/O test PASS			Time: 213 us
			Erase 64KB Block: 	Data I/O test PASS			Time: 78 us
			Erase 32KB Block: 	Data I/O test PASS			Time: 85 us
			Erase 4KB Sector: 	Data I/O test PASS			Time: 78 us

			Power Down: 		Data I/O test PASS			Time: 60 us
			Power Up: 		Data I/O test PASS			Time: 51 us
                        Write Function has failed errorcheck.
          		Byte: 	Data I/O test FAIL	Write Time: 0 us,	Read Time: 46 us

- [ ] If you have a problem with a particular function, call the flash.error() function (after you have made sure you have started up your Serial port with a ``` Serial.begin(BAUD) ``` ). Provide details of the function, the data given to/ expected from the function and the error code here: (**Please repeat this for every function you have an error with***)
        Function - 
        Data - 
        Error code - 
- [ ] Provide a **minimal code snippet** example that reproduces the bug. Please make sure you wrap any code in the proper code blocks like below 
```
SPIFlash flash(21, &SPI2);       //Use this constructor if using an SPI bus other than the default SPI. Only works with chips with more than one hardware SPI bus

void setup() {
	
	Serial.begin(BAUD_RATE);
	flash.begin(MB(8));
	Serial.println();	
	getID();
	eraseChipTest();
	eraseSectionTest();
	eraseBlock64KTest();
	eraseBlock32KTest();
	eraseSectorTest();
	delay(10);
	powerDownTest();
	powerUpTest();
	Serial.println();
	byteTest();
	charTest();
	wordTest();
	shortTest();
	uLongTest();
	delay(10);
	longTest();
	floatTest();
	structTest();
	arrayTest();
	stringTest();
	printLine();
	if (!flash.functionRunTime()) {
		Serial.println(F("Please uncomment RUNDIAGNOSTIC in SPIFlash.h to see the time taken by each function to run."));
	}
}
```
- [ ] Provide **screenshots** where appropriate
- [ ] Provide **wiring diagram** where appropriate

-----------------------------
###### Feature Requests only

- [ ] If this is a feature request, what is the motivation / use case for changing the behavior?

----------------------------

- [ ] Other information (e.g. detailed explanation, related issues, suggestions how to fix, links for us to have context etc.)

_Make sure to add **all the information needed to understand the bug/feature** so that someone can help. If the info is missing we'll add the 'Needs more information' label and close the issue until there is enough information._

Marzogh, 2018-04-25:
Hi, that definitely shouldn't be happening. Wnfortunately, I don't own an Xplained Pro SAMD21 so I can't test it myself. 

- Does the code run if you try and wire your flash via SPI1?
- Could you please try replacing your library with the version from the development branch and see if that fixes any issues? I had quite a few chips reporting write errors and that bug's been fixed in the development branch code. (It was a timing issue 😅)

StephiB, 2018-04-26:
> Does the code run if you try and wire your flash via SPI1?

Unfortunatly its hardwired on PCB to SPI2, but all of the 3 SPI interfaces are working perfectly with other devices.
Thanks for the hint, will give it a try this evening.

Marzogh, 2018-04-26:
Cheers, keep me posted please. 🙂

StephiB, 2018-04-28:
[AT25DF081A_Command Listing.pdf](https://github.com/Marzogh/SPIFlash/files/1958085/AT25DF081A_Command.Listing.pdf)
Hi, unfortunatly the same behaviour with the development version files 
Seems like an issue with the protection commands 
Looking forward to find a solution for this.
Have a nice Weekend

Marzogh, 2018-05-19:
Hi @StephiB could you confirm that the latest code from the development branch fixes this bug for you?

Marzogh, 2018-05-23:
Oh dear. It might be something to do with the AT25DF081A that I missed. 🙁  I'll look into this over the weekend and let you know.

StephiB, 2018-05-23:
Hi, it works after I played around with the spi speed. Sometimes there is a problem with writing float or string (one out of ten). The last issue is the wrong size, AT25DF081A is a 1MB chip and your library recognized 512kB.
Thanks a lot for efforts.

Marzogh, 2018-05-23:
What speed has it been working at?

StephiB, 2018-05-23:
Seems like it works best at 1MHz

gkavet, 2018-07-20:
Getting a similar issue with SAMD on Adafruit M0 basic proto with Microchip SST26VF016B. The code is just FlashDiagnostics from Examples.

Also curious how StephiB could change the spispeed in SpiMemory. 

```
Initialising..........
Chip Diagnostics initiated.

No Chip size defined by user. Automated identification initiated.
Custom Chipsize defined

SPIMemory Library version: 3.2.1

JEDEC ID: 0xBF2641
Man ID: 0xBF
Memory ID: 0x26
Capacity: 16777216
Max Pages: 4294967295

-----------------------------------------------------------------------------------------------------------------------------
							Testing library code
-----------------------------------------------------------------------------------------------------------------------------
			Function		Test result			     Runtime
-----------------------------------------------------------------------------------------------------------------------------
			Power Down	This function is not supported by the flash memory hardware.
	   FAIL			Not all chips support power down. Check your datasheet.
			Power Up		   PASS				      47 us

			Erase Chip		   PASS				      62 us
			Erase 72KB		   PASS				      203 us
			Erase 64KB		   PASS				      75 us
			Erase 32KB		   PASS				      82 us
			Erase 4KB		   PASS				      75 us
-----------------------------------------------------------------------------------------------------------------------------
			Data type		I/O Result	      Write time	      Read time
-----------------------------------------------------------------------------------------------------------------------------
			Byte			   FAIL			0 us			47 us
			Char			   FAIL			0 us			47 us
			Word			   FAIL			0 us			49 us
			Short			   FAIL			0 us			49 us
			ULong			   FAIL			0 us			54 us
			Long			   FAIL			0 us			55 us
			Float			   FAIL			0 us			54 us
Write Function has failed errorcheck.
If this does not help resolve/clarify this issue, please raise an issue at http://www.github.com/Marzogh/SPIMemory/issues with the details of what your were doing when this error occurred
			Struct			   FAIL			0 us			103 us
			Byte Array		   FAIL	0, 	   FAIL	1, 	   FAIL	2, 	   FAIL	3, 	   FAIL	4, 	   FAIL	5, 	   FAIL	6, 	   FAIL	7, 	   FAIL	8, 	   FAIL	9, 	   FAIL	10, 	   FAIL	11, 	   FAIL	12, 	   FAIL	13, 	   FAIL	14, 	   FAIL	15, 	   FAIL	16, 	   FAIL	17, 	   FAIL	18, 	   FAIL	19, 	   FAIL	20, 	   FAIL	21, 	   FAIL	22, 	   FAIL	23, 	   FAIL	24, 	   FAIL	25, 	   FAIL	26, 	   FAIL	27, 	   FAIL	28, 	   FAIL	29, 	   FAIL	30, 	   FAIL	31, 	   FAIL	32, 	   FAIL	33, 	   FAIL	34, 	   FAIL	35, 	   FAIL	36, 	   FAIL	37, 	   FAIL	38, 	   FAIL	39, 	   FAIL	40, 	   FAIL	41, 	   FAIL	42, 	   FAIL	43, 	   FAIL	44, 	   FAIL	45, 	   FAIL	46, 	   FAIL	47, 	   FAIL	48, 	   FAIL	49, 	   FAIL	50, 	   FAIL	51, 	   FAIL	52, 	   FAIL	53, 	   FAIL	54, 	   FAIL	55, 	   FAIL	56, 	   FAIL	57, 	   FAIL	58, 	   FAIL	59, 	   FAIL	60, 	   FAIL	61, 	   FAIL	62, 	   FAIL	63, 	   FAIL	64, 	   FAIL	65, 	   FAIL	66, 	   FAIL	67, 	   FAIL	68, 	   FAIL	69, 	   FAIL	70, 	   FAIL	71, 	   FAIL	72, 	   FAIL	73, 	   FAIL	74, 	   FAIL	75, 	   FAIL	76, 	   FAIL	77, 	   FAIL	78, 	   FAIL	79, 	   FAIL	80, 	   FAIL	81, 	   FAIL	82, 	   FAIL	83, 	   FAIL	84, 	   FAIL	85, 	   FAIL	86, 	   FAIL	87, 	   FAIL	88, 	   FAIL	89, 	   FAIL	90, 	   FAIL	91, 	   FAIL	92, 	   FAIL	93, 	   FAIL	94, 	   FAIL	95, 	   FAIL	96, 	   FAIL	97, 	   FAIL	98, 	   FAIL	99, 	   FAIL	100, 	   FAIL	101, 	   FAIL	102, 	   FAIL	103, 	   FAIL	104, 	   FAIL	105, 	   FAIL	106, 	   FAIL	107, 	   FAIL	108, 	   FAIL	109, 	   FAIL	110, 	   FAIL	111, 	   FAIL	112, 	   FAIL	113, 	   FAIL	114, 	   FAIL	115, 	   FAIL	116, 	   FAIL	117, 	   FAIL	118, 	   FAIL	119, 	   FAIL	120, 	   FAIL	121, 	   FAIL	122, 	   FAIL	123, 	   FAIL	124, 	   FAIL	125, 	   FAIL	126, 	   FAIL	127, 	   FAIL	128, 	   FAIL	129, 	   FAIL	130, 	   FAIL	131, 	   FAIL	132, 	   FAIL	133, 	   FAIL	134, 	   FAIL	135, 	   FAIL	136, 	   FAIL	137, 	   FAIL	138, 	   FAIL	139, 	   FAIL	140, 	   FAIL	141, 	   FAIL	142, 	   FAIL	143, 	   FAIL	144, 	   FAIL	145, 	   FAIL	146, 	   FAIL	147, 	   FAIL	148, 	   FAIL	149, 	   FAIL	150, 	   FAIL	151, 	   FAIL	152, 	   FAIL	153, 	   FAIL	154, 	   FAIL	155, 	   FAIL	156, 	   FAIL	157, 	   FAIL	158, 	   FAIL	159, 	   FAIL	160, 	   FAIL	161, 	   FAIL	162, 	   FAIL	163, 	   FAIL	164, 	   FAIL	165, 	   FAIL	166, 	   FAIL	167, 	   FAIL	168, 	   FAIL	169, 	   FAIL	170, 	   FAIL	171, 	   FAIL	172, 	   FAIL	173, 	   FAIL	174, 	   FAIL	175, 	   FAIL	176, 	   FAIL	177, 	   FAIL	178, 	   FAIL	179, 	   FAIL	180, 	   FAIL	181, 	   FAIL	182, 	   FAIL	183, 	   FAIL	184, 	   FAIL	185, 	   FAIL	186, 	   FAIL	187, 	   FAIL	188, 	   FAIL	189, 	   FAIL	190, 	   FAIL	191, 	   FAIL	192, 	   FAIL	193, 	   FAIL	194, 	   FAIL	195, 	   FAIL	196, 	   FAIL	197, 	   FAIL	198, 	   FAIL	199, 	   FAIL	200, 	   FAIL	201, 	   FAIL	202, 	   FAIL	203, 	   FAIL	204, 	   FAIL	205, 	   FAIL	206, 	   FAIL	207, 	   FAIL	208, 	   FAIL	209, 	   FAIL	210, 	   FAIL	211, 	   FAIL	212, 	   FAIL	213, 	   FAIL	214, 	   FAIL	215, 	   FAIL	216, 	   FAIL	217, 	   FAIL	218, 	   FAIL	219, 	   FAIL	220, 	   FAIL	221, 	   FAIL	222, 	   FAIL	223, 	   FAIL	224, 	   FAIL	225, 	   FAIL	226, 	   FAIL	227, 	   FAIL	228, 	   FAIL	229, 	   FAIL	230, 	   FAIL	231, 	   FAIL	232, 	   FAIL	233, 	   FAIL	234, 	   FAIL	235, 	   FAIL	236, 	   FAIL	237, 	   FAIL	238, 	   FAIL	239, 	   FAIL	240, 	   FAIL	241, 	   FAIL	242, 	   FAIL	243, 	   FAIL	244, 	   FAIL	245, 	   FAIL	246, 	   FAIL	247, 	   FAIL	248, 	   FAIL	249, 	   FAIL	250, 	   FAIL	251, 	   FAIL	252, 	   FAIL	253, 	   FAIL	254, 	   PASS			0 us			535 us
Write Function has failed errorcheck.
If this does not help resolve/clarify this issue, please raise an issue at http://www.github.com/Marzogh/SPIMemory/issues with the details of what your were doing when this error occurred
			String			   FAIL			0 us			74 us
```

Marzogh, 2019-04-20:
The problem was a bug due to a missing call to `_disableGlobalBlockProtect()` in `begin()` as identified by @jetty840 in #164. This has been fixed in the upcoming [v3.3.0](https://github.com/Marzogh/SPIMemory/tree/v3.3.0) release.

closed 2019-04-20 by Marzogh