Marzogh/SPIMemory #129: The README is not readable anymore

polaroi8d, 2018-04-25:
There are plenty of texts in the READ.ME, I think we can slice this text into a docs folder. Where it could be API documentation, Getting started description, pull request policy or anything else, and on the main READ.ME file there is only a few information about the project and the lib. Any suggestion?

Marzogh, 2018-04-25:
Hi mate, you are right. I've just finished getting the wiki up to date and was thinking of reducing the ReadMe to just basics and point to the wiki for the next release.

I have also been looking at trying to get the repo Doxygen compatible so everything stays up to date with in-line code comments. However I have not used doxygen previously and so it might take me a little time to figure out how it works. 🙂 

Also, I've recently discovered https://readthedocs.org/ and might migrate the docs there someday.

Marzogh, 2018-04-26:
Started the process of getting documentation in order through Sphinx and ReadTheDocs - as of commit 51512a3

polaroi8d, 2018-04-26:
Nice work! If you can shift any of these tasks I can help you in my freetime.

Marzogh, 2018-05-21:
Thanks for the offer 👍 I've got whatever I have done to date on the development branch. Feel free to add to that if you have time and send in a pull request. 🙂 

Just make sure you follow the same page format/design that I've already established - if you have ideas for improving that, feel free to let me know. I'm happy to take any suggestions for improvement on board. 😁