Marzogh/SPIMemory #135: Bug Report/Annoyance: 3.2.0 uses twice the program memory of 3.1.0

Andon-A, 2018-05-17:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
###### ---------------------------- **DO NOT DELETE OR EDIT** anything above this line ----------------------------

Arduino IDE 1.8.5, using alternately SPIFlash 3.1.0 and SPIMemory 3.2.0, under Windows 10.

RUNDIAGNOSTIC and PRINTNAMECHANGEALERT (Lines 36 and 64) in SPIFlash.h are commented out to save space for 3.1.0
RUNDIAGNOSTIC (Line 35) in SPIMemory.h is commented out to save space for 3.2.0

Board settings are Arduino Pro/Pro Mini, ATMega328p, 3.3v/8mHz

The following code compiles to 2784 bytes in 3.1.0:
```#include <SPIFlash.h>
SPIFlash flash(5);

void setup() {
  flash.begin();
}

void loop() {

}```

Replacing `SPIFlash.h` with `SPIMemory.h` for using with 3.2.0 compiles with 5636 bytes, over double that of 3.1.0.

This is of particular interest to me as the [program I am using it in](https://github.com/PhoenixbornTech/DigiBadge_Mini) compiles to precisely 30720 bytes with 3.1.0 - The maximum a 328p can utilize. For ease of users compiling their own code, I would like to be able to give simple instructions on what needs to be commented out. For 3.1.0, this is "Comment out lines 36 and 64 in SPIFlash.h" - For 3.2.0 this "Simple" memory saver doesn't work.

I haven't had time to investigate the *cause* of the space usage increase (I'm flying out to Seattle tomorrow morning), but I feel such incredible balooning seems to be a bug.

If this is expected performance, then this can be treated as a feature request for space saving measures.

Marzogh, 2018-05-17:
Ugh! That's something that I missed with the name change issue.

I will fix this for you ASAP and get a bugfix release out within a week with the option of turning off the new features. That should take the memory footprint back down. Thanks for catching it.

Marzogh, 2018-05-17:
Commit e24a2bc fixes this issue and reduces the footprint even further. If you make sure that #define RUNDIAGNOSTIC and #USES_SFDP are commented out, your code is currently compiling ti 30342 bytes instead of the previous 30720 bytes. Feel free to use the code from the development branch for now. I have a few more changes to make before the code goes stable and I push out v 3.2.1

closed 2018-05-21 by Marzogh

Andon-A, 2018-05-22:
I've been away pretty much since I posted this. Awesome to come back to! Thanks a lot.

Marzogh, 2018-05-23:
Anytime! v3.2.1 is currently out in the wild and should fix the problem you had. 🙂👍🏽

commit e24a2bcf (2018-05-17): Fixes #135 : The addition of the SFDP checking to _chipID resulted in a sudden (very large) increase in compiled code size. As of the current commit on the development branch, SFDP checking is an user controlled option. To get the library to work with SFDP compatible flash memory that is not officially supported, the user must uncomment '//#define USES_SFDP' in 'SPIMemory.h'.
PR #137 (2018-05-21): v3.2.1 is ready to go stable
[image]
###### ---------------------------- **DO NOT DELETE OR EDIT** anything above this line ----------------------------

# Pull request details

* **Please check if the PR fulfills these requirements**
- [x] The commit message/s explain/s all the changes clearly
- [x] Tests for the changes have been added (for bug fixes / features)
- [x] Docs have been added / updated (for bug fixes / features)

* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)
- [x] Bug fix: Fixes #135 & #128 (Fixed by @constiko in #136)
- [x] Added feature [... cut]