Marzogh/SPIMemory #139: Whether SPIMemory supports  SST25VF080B Flash  ??

BharatGopani, 2018-05-29:
I am using Arduino M0 Pro board with SPIMemory library with SST25VF080B Flash. But following diagnostic message was observed while writing string to Flash.I am using  readWriteString example.

I would be gratfult if you could suggest for the same.

Following message observed on the PC terminal.

Chip Diagnostics initiated.

The Flash chip does not support SFDP.
Custom Chipsize defined
Write Function has failed errorcheck.
If this does not help resolve/clarify this issue, please raise an issue at http://www.github.com/Marzogh/SPIMemory/issues with the details of what your were doing when this error occurred
Written st

Marzogh, 2018-05-29:
Also, I do not have an SST25VF080B at hand, but from all I can see from the datasheet, it should work the same as the SST25VF064C which is officially supported. For a list of officially supported chips, please refer to the [ReadMe.md](https://github.com/Marzogh/SPIMemory/blob/master/README.md) file that comes with the library.

BharatGopani, 2018-05-30:
Hi Marzogh,

Thanks for the immediate reply. PFB my reply.
 
**I'm submitting a  bug report.
  - [Y ] bug report   :
  - [N] feature request
  - [ N] support request => Please do not submit support request here, see note at the top of this template.
  
Do the checklist before filing an issue:

- [ N] Is this something you can **debug and fix**? Send a pull request! Bug fixes and documentation fixes are welcome.
- [N ] Is this an idea for a feature? Post the feature request as an issue here with a title that clearly says "Feature Request".
- [Y ] Is this a bug that you cannot fix? Post the bug report as an issue here with a title that cleary says "Bug Report".
  
When opening an issue please include the following details:
-------------------------------------------------------------

- [ ] Do you want to request a *feature* or report a *bug*?   : report a bug
- [ ] Which library version are you using?   :   3.2.0
       Where can I find version 3.2.1 ??
- [ ] Which Arduino IDE version are you compiling the library on?  : 1.8.5
- [ ] Which OS are you compiling your code on? [Linux / Mac / Windows] : Windows
- [ ] Which microcontroller platform are you using?  : SAMD21G18
- [ ] Which Flash memory module are you using?     : SST25VF080B
- [ ] What is the current behavior?                           : It's giving error checking while writing a string.
- [ ] What is the expected behavior?                        : It must not give this error.

--------------------------
###### Bug reports only

- [ ] If this is a bug report - Provide a **minimal code snippet** example that reproduces the bug. Please make sure you wrap any code in the proper code blocks like below 

CODE HERE : 

#include<SPIMemory.h>
uint32_t strAddr;

#define ARDUINO_SAMD_ZERO
//#define SERIAL_PORT_USBVIRTUAL

#define CAPACITY  4096

//#if defined(ARDUINO_SAMD_ZERO) && defined(SERIAL_PORT_USBVIRTUAL)
// Required for Serial on Zero based boards
//#define Serial SERIAL_PORT_USBVIRTUAL
//#endif

#if defined (SIMBLEE)
#define BAUD_RATE 250000
#define RANDPIN 1
#else
#define BAUD_RATE 115200
#define RANDPIN A0
#endif

//SPIFlash flash(SS1, &SPI1);       //Use this constructor if using an SPI bus other than the default SPI. Only works with chips with more than one hardware SPI bus
SPIFlash flash;

bool readSerialStr(String &inputStr);

void setup() {
  Serial.begin(115200);
  delay(200);
//#if defined (ARDUINO_SAMD_ZERO) || (__AVR_ATmega32U4__)
//  while (!Serial) ; // Wait for Serial monitor to open
//#endif

  flash.begin(CAPACITY);

  randomSeed(analogRead(RANDPIN));
  strAddr = random(0, flash.getCapacity());
  String inputString = "This is a test String";
  flash.writeStr(strAddr, inputString);
  Serial.print(F("Written string: "));

  //Serial.println("World");
  Serial.println(inputString);
  Serial.print(F("To address: "));
  Serial.println(strAddr);
  String outputString = "";
  
  if (flash.readStr(strAddr, outputString)) {
    Serial.print(F("Read string: "));
    Serial.println(outputString);
    Serial.print(F("From address: "));
    Serial.println(strAddr);
  }
  while (!flash.eraseSector(strAddr));
}

void loop() {

}

//Reads a string from Serial
bool readSerialStr(String &inputStr) {
  if (!Serial)
    Serial.begin(115200);
  while (Serial.available()) {
    inputStr = Serial.readStringUntil('\n');
    Serial.println(inputStr);
    return true;
  }
  return false;
}

- [ ] Provide **screenshots** where appropriate
- [ ] Provide **wiring diagram** where appropriate

-----------------------------
###### Feature Requests only

- [ ] If this is a feature request, what is the motivation / use case for changing the behavior?

----------------------------

- [ ] Other information (e.g. detailed explanation, related issues, suggestions how to fix, links for us to have context etc.)

_Make sure to add **all the information needed to understand the bug/feature** so that someone can help. If the info is missing we'll add the 'Needs more information' label and close the issue until there is enough information._

BharatGopani, 2018-05-30:
Hi

Thanks for the immediate reply.

Current I am using SPIMemory version 3.2.0.  & Arduino IDE 1.8.5 and
Arduino Mo Pro board.

From where I can find SPIMemory version 3.2.1  ??

Regards
Bharat Gopani

Marzogh, 2018-05-30:
Hi Bharat,

The example files do not need y user modification beyond declaring chip size or changing the SPI Port in use. Here's a list of issues with your modifications to readWriteString.ino.

### Issue 1:
```
#define ARDUINO_SAMD_ZERO                   // You do not need to define this
//#define SERIAL_PORT_USBVIRTUAL         // You do not need to define this
```
Both the defines below are not user defines. They are pre-processor directives that are internal to
the Arduino IDE and are generated from the board definition flies that are selected when you pick the
board you are uploading your code to.

### Issue 2:
```
#define CAPACITY  4096 // Get rid of this definition. You don't need a #define here.
```
The capacity definition has to be done properly. If you refer to [this section of the Read Me](https://github.com/Marzogh/SPIMemory/blob/master/README.md#begin_chipsize) you will see that there is a pre-defined format for to define chip size. [The wiki gives you more information.](https://github.com/Marzogh/SPIMemory/wiki/Chipsize). Also refer to Issue 4 below.

### Issue 3:
```
//#if defined(ARDUINO_SAMD_ZERO) && defined(SERIAL_PORT_USBVIRTUAL) /* Do not comment out */
// Required for Serial on Zero based boards /* Do not comment out */
//#define Serial SERIAL_PORT_USBVIRTUAL /* Do not comment out */
//#endif /* Do not comment out */
```

```
//#if defined (ARDUINO_SAMD_ZERO) || (__AVR_ATmega32U4__) /* Do not comment out */
//  while (!Serial) ; // Wait for Serial monitor to open /* Do not comment out */
//#endif /* Do not comment out */
```
Do not comment these sections out. They prevent Serial comms errors that tend to happen with SAMD21 boards like the Arduino Zero if you don't wait for comms to be established

### Issue 4:
```
  flash.begin(CAPACITY);
```
Replace CAPACITY here with a value the library understands (refer to Issue 2 above). You are using the SST25VF080B. The datasheet states that this chip has a capacity of 8Mb (8 Megabits not 8 MegaBytes). 8 bits = 1 Byte, so, in this scenario your chip's capacity is 1 MB (MegaByte) i.e. MB(1). So, replace the word CAPACITY with MB(1)
<hr>

### Here's the code as it should be

```
#include<SPIMemory.h>

uint32_t strAddr;

#if defined(ARDUINO_SAMD_ZERO) && defined(SERIAL_PORT_USBVIRTUAL)
// Required for Serial on Zero based boards
#define Serial SERIAL_PORT_USBVIRTUAL
#endif

#if defined (SIMBLEE)
#define BAUD_RATE 250000
#define RANDPIN 1
#else
#define BAUD_RATE 115200
#define RANDPIN A0
#endif

//SPIFlash flash(SS1, &SPI1);       //Use this constructor if using an SPI bus other than the default SPI. Only works with chips with more than one hardware SPI bus
SPIFlash flash;

bool readSerialStr(String &inputStr);

void setup() {
  Serial.begin(BAUD_RATE);
#if defined (ARDUINO_SAMD_ZERO) || (__AVR_ATmega32U4__)
  while (!Serial) ; // Wait for Serial monitor to open
#endif

  flash.begin(MB(1));

  randomSeed(analogRead(RANDPIN));
  strAddr = random(0, flash.getCapacity());
  String inputString = "This is a test String";
  flash.writeStr(strAddr, inputString);
  Serial.print(F("Written string: "));
  Serial.println(inputString);
  Serial.print(F("To address: "));
  Serial.println(strAddr);
  String outputString = "";
  if (flash.readStr(strAddr, outputString)) {
    Serial.print(F("Read string: "));
    Serial.println(outputString);
    Serial.print(F("From address: "));
    Serial.println(strAddr);
  }
  while (!flash.eraseSector(strAddr));
}

void loop() {

}

//Reads a string from Serial
bool readSerialStr(String &inputStr) {
  if (!Serial)
    Serial.begin(115200);
  while (Serial.available()) {
    inputStr = Serial.readStringUntil('\n');
    Serial.println(inputStr);
    return true;
  }
  return false;
}
```

Marzogh, 2018-05-30:
To update to v3.2.1 of SPIMemory from the library manager:
 - Go to Sketch -> Include Library -> Manage Libraries...
 - Search for SPIMemory
 - Set the dropdown menu to v3.2.1
[image]

 - Click on 'Update' or 'Install'

Marzogh, 2018-05-30:
**If the updated code above does not fix your issues** please do this:

After you have installed v3.2.1
  - Uncomment `#define RUNDIAGNOSTIC` in `SPIMemory.h` and save the file.
  - Run `FlashDiagnostics.ino` from the library examples.
  - Copy the Serial response into your reply to this message.

BharatGopani, 2018-05-30:
Hi,
Installed version 3.2.1 & updated code as per your suggestions but issue persists.
I uncommented #define RUNDIAGNOSTIC in SPIMemory.h and run FlashDiagnostics.ino .
PFB serial response. 

Initialising..........
Chip Diagnostics initiated.

No Chip size defined by user. Automated identification initiated.
Custom Chipsize defined

SPIMemory Library version: 3.2.1

JEDEC ID: 0xBF258E
Man ID: 0xBF
Memory ID: 0x25
Capacity: 1048576
Max Pages: 4294967295

-----------------------------------------------------------------------------------------------------------------------------
							Testing library code
-----------------------------------------------------------------------------------------------------------------------------
			Function		Test result			     Runtime
-----------------------------------------------------------------------------------------------------------------------------
			Power Down	This function is not supported by the flash memory hardware.
	   FAIL			Not all chips support power down. Check your datasheet.
			Power Up		   PASS				      46 us

			Erase Chip		   PASS				      34.56 ms
			Erase 72KB		   PASS				      35.66 ms
			Erase 64KB		   PASS				      17.85 ms
			Erase 32KB		   PASS				      17.86 ms
			Erase 4KB		   PASS				      17.77 ms
-----------------------------------------------------------------------------------------------------------------------------
			Data type		I/O Result	      Write time	      Read time
-----------------------------------------------------------------------------------------------------------------------------
			Byte			   PASS			103 us			51 us
			Char			   PASS			109 us			47 us
			Word			   FAIL			0 us			49 us
			Short			   FAIL			0 us			49 us
			ULong			   FAIL			0 us			55 us
			Long			   FAIL			0 us			55 us
			Float			   FAIL			0 us			55 us
Write Function has failed errorcheck.
If this does not help resolve/clarify this issue, please raise an issue at http://www.github.com/Marzogh/SPIMemory/issues with the details of what your were doing when this error occurred
			Struct			   FAIL			0 us			108 us
			Byte Array		   FAIL	0, 	   FAIL	1, 	   FAIL	2, 	   FAIL	3, 	   FAIL	4, 	   FAIL	5, 	   FAIL	6, 	   FAIL	7, 	   FAIL	8, 	   FAIL	9, 	   FAIL	10, 	   FAIL	11, 	   FAIL	12, 	   FAIL	13, 	   FAIL	14, 	   FAIL	15, 	   FAIL	16, 	   FAIL	17, 	   FAIL	18, 	   FAIL	19, 	   FAIL	20, 	   FAIL	21, 	   FAIL	22, 	   FAIL	23, 	   FAIL	24, 	   FAIL	25, 	   FAIL	26, 	   FAIL	27, 	   FAIL	28, 	   FAIL	29, 	   FAIL	30, 	   FAIL	31, 	   FAIL	32, 	   FAIL	33, 	   FAIL	34, 	   FAIL	35, 	   FAIL	36, 	   FAIL	37, 	   FAIL	38, 	   FAIL	39, 	   FAIL	40, 	   FAIL	41, 	   FAIL	42, 	   FAIL	43, 	   FAIL	44, 	   FAIL	45, 	   FAIL	46, 	   FAIL	47, 	   FAIL	48, 	   FAIL	49, 	   FAIL	50, 	   FAIL	51, 	   FAIL	52, 	   FAIL	53, 	   FAIL	54, 	   FAIL	55, 	   FAIL	56, 	   FAIL	57, 	   FAIL	58, 	   FAIL	59, 	   FAIL	60, 	   FAIL	61, 	   FAIL	62, 	   FAIL	63, 	   FAIL	64, 	   FAIL	65, 	   FAIL	66, 	   FAIL	67, 	   FAIL	68, 	   FAIL	69, 	   FAIL	70, 	   FAIL	71, 	   FAIL	72, 	   FAIL	73, 	   FAIL	74, 	   FAIL	75, 	   FAIL	76, 	   FAIL	77, 	   FAIL	78, 	   FAIL	79, 	   FAIL	80, 	   FAIL	81, 	   FAIL	82, 	   FAIL	83, 	   FAIL	84, 	   FAIL	85, 	   FAIL	86, 	   FAIL	87, 	   FAIL	88, 	   FAIL	89, 	   FAIL	90, 	   FAIL	91, 	   FAIL	92, 	   FAIL	93, 	   FAIL	94, 	   FAIL	95, 	   FAIL	96, 	   FAIL	97, 	   FAIL	98, 	   FAIL	99, 	   FAIL	100, 	   FAIL	101, 	   FAIL	102, 	   FAIL	103, 	   FAIL	104, 	   FAIL	105, 	   FAIL	106, 	   FAIL	107, 	   FAIL	108, 	   FAIL	109, 	   FAIL	110, 	   FAIL	111, 	   FAIL	112, 	   FAIL	113, 	   FAIL	114, 	   FAIL	115, 	   FAIL	116, 	   FAIL	117, 	   FAIL	118, 	   FAIL	119, 	   FAIL	120, 	   FAIL	121, 	   FAIL	122, 	   FAIL	123, 	   FAIL	124, 	   FAIL	125, 	   FAIL	126, 	   FAIL	127, 	   FAIL	128, 	   FAIL	129, 	   FAIL	130, 	   FAIL	131, 	   FAIL	132, 	   FAIL	133, 	   FAIL	134, 	   FAIL	135, 	   FAIL	136, 	   FAIL	137, 	   FAIL	138, 	   FAIL	139, 	   FAIL	140, 	   FAIL	141, 	   FAIL	142, 	   FAIL	143, 	   FAIL	144, 	   FAIL	145, 	   FAIL	146, 	   FAIL	147, 	   FAIL	148, 	   FAIL	149, 	   FAIL	150, 	   FAIL	151, 	   FAIL	152, 	   FAIL	153, 	   FAIL	154, 	   FAIL	155, 	   FAIL	156, 	   FAIL	157, 	   FAIL	158, 	   FAIL	159, 	   FAIL	160, 	   FAIL	161, 	   FAIL	162, 	   FAIL	163, 	   FAIL	164, 	   FAIL	165, 	   FAIL	166, 	   FAIL	167, 	   FAIL	168, 	   FAIL	169, 	   FAIL	170, 	   FAIL	171, 	   FAIL	172, 	   FAIL	173, 	   FAIL	174, 	   FAIL	175, 	   FAIL	176, 	   FAIL	177, 	   FAIL	178, 	   FAIL	179, 	   FAIL	180, 	   FAIL	181, 	   FAIL	182, 	   FAIL	183, 	   FAIL	184, 	   FAIL	185, 	   FAIL	186, 	   FAIL	187, 	   FAIL	188, 	   FAIL	189, 	   FAIL	190, 	   FAIL	191, 	   FAIL	192, 	   FAIL	193, 	   FAIL	194, 	   FAIL	195, 	   FAIL	196, 	   FAIL	197, 	   FAIL	198, 	   FAIL	199, 	   FAIL	200, 	   FAIL	201, 	   FAIL	202, 	   FAIL	203, 	   FAIL	204, 	   FAIL	205, 	   FAIL	206, 	   FAIL	207, 	   FAIL	208, 	   FAIL	209, 	   FAIL	210, 	   FAIL	211, 	   FAIL	212, 	   FAIL	213, 	   FAIL	214, 	   FAIL	215, 	   FAIL	216, 	   FAIL	217, 	   FAIL	218, 	   FAIL	219, 	   FAIL	220, 	   FAIL	221, 	   FAIL	222, 	   FAIL	223, 	   FAIL	224, 	   FAIL	225, 	   FAIL	226, 	   FAIL	227, 	   FAIL	228, 	   FAIL	229, 	   FAIL	230, 	   FAIL	231, 	   FAIL	232, 	   FAIL	233, 	   FAIL	234, 	   FAIL	235, 	   FAIL	236, 	   FAIL	237, 	   FAIL	238, 	   FAIL	239, 	   FAIL	240, 	   FAIL	241, 	   FAIL	242, 	   FAIL	243, 	   FAIL	244, 	   FAIL	245, 	   FAIL	246, 	   FAIL	247, 	   FAIL	248, 	   FAIL	249, 	   FAIL	250, 	   FAIL	251, 	   FAIL	252, 	   FAIL	253, 	   FAIL	254, 	   PASS			0 us			544 us
Write Function has failed errorcheck.
If this does not help resolve/clarify this issue, please raise an issue at http://www.github.com/Marzogh/SPIMemory/issues with the details of what your were doing when this error occurred
			String			   FAIL			0 us			78 us

BharatGopani, 2018-05-31:
Hi

I am grateful if you could suggest for the same.

waiting for your reply.

Regards
Bharat Gopani

BharatGopani, 2018-06-06:
Hi

If you don't know solution then just say I don't know. 

Not replying is not good thing. So request you to reply.

Regards
Bharat Gopani

mkielak, 2018-07-05:
I have same tests result with SST25VF020B. I initialize flash with flash.begin(KB(256)) but only erasing tests pass, byte and char. Other tests fails.

mkielak, 2018-07-07:
Ok, after many hours i found the cause of this issue. If you look into documentation of SST25 family and compare it to SST26 you will find out, that Byte-Program command (0x02) saves **1 or more** bytes in SST26 but same command (0x02) saves **only 1** byte in SST25. You need to use special command for SST25 called AAI-Word-Program (0xAD). I found some library which has support for both SST25 and SST26, maybe this would help - https://gitbucket.pgollor.de/kalle/MySensors/tree/extended-gateway/drivers/SPIFlash

BharatGopani, 2018-07-14:
> gateway/drivers/SPIFlash

Thanks.

Marzogh, 2018-08-09:
Hi guys, My apologies for the long delay in replying. I was away for health reasons and am slowly transitioning back into normality.

@mkielak thanks for identifying the issue. I will look at it and incorporate any changes required into the next version.

@BharatGopani I hope @mkielak 's solution has helped. If not, I will keep you posted on any updates to SPIMemory over the next week - which should hopefully fix this problem.

closed 2019-04-20 by Marzogh

max6464, 2019-11-25:
Hi. First I'd like to say thanks for all the hard work on this. Just wondering if anyone has had any further success with this issue? I am currently experiencing the exact same thing on v3.4. Char and Byte pass, but every other data type fails.

Thanks in advance.