Marzogh/SPIMemory #143: read Strings missing characters / SAMD21 and SST26VF064B

WalterPrech, 2018-06-29:
[image] [image]
[image] [image] [image]
[image] [image]
###### ---------------------------- **DO NOT DELETE OR EDIT** anything above this line ----------------------------
###### ---------------------------- **DO NOT DELETE OR EDIT** anything above this line ----------------------------

Hello!

I'm submitting a bug report, I tried to identify what's going wrong but I cannot fix it...
Bug report :

Library version : 3.2.1
Branch : master
Arduino IDE : 1.8.4
OS : Mac OS X 10.13.5
Microcontroller : SAMD21
Flash memory module : SST26VF064B
Current behaviour :
In my application I write a log strings to a fix addresses and then I want to read this strings and export them to csv. While reading the strings, there are missing pieces of the string.
To verify this behaviour, I edited the example sketch readWriteString and in the first step I erase the sector (125000) and write the string "This is a test String". Reading after writing the string gives the correct string back.
In the second step I disabled the eraseSector and the writing to this sector and only read the string, I got the result: " is a test String"

`SPIFlash flash(12);
void setup() {
  Serial.begin(BAUD_RATE);
#if defined (ARDUINO_SAMD_ZERO) || (__AVR_ATmega32U4__)
  while (!Serial) ; // Wait for Serial monitor to open
#endif

  flash.begin();

  randomSeed(analogRead(RANDPIN));
  strAddr = 125000;
  //flash.eraseSector(strAddr);
  /*String inputString = "This is a test String";
  flash.writeStr(strAddr, inputString);
  Serial.print(F("Written string: "));
  Serial.println(inputString);
  Serial.print(F("To address: "));
  Serial.println(strAddr);*/
  String outputString = "";
  if (flash.readStr(strAddr, outputString)) {
    Serial.print(F("Read string: "));
    Serial.println(outputString);
    Serial.print(F("From address: "));
    Serial.println(strAddr);
  }
  //while (!flash.eraseSector(strAddr));
}`

Output of the first run with activated #RUNDIAGNOSTICS and eraseSector and writeStr:
`Chip Diagnostics initiated.

No Chip size defined by user. Automated identification initiated.
Chip identified. This chip is fully supported by the library.
run disableGlobalBlockProtect!
Written string: This is a test String
To address: 125000
Read string: This is a test String
From address: 125000`

Output of the second run with only readStr:
`Read string:      is a test String
From address: 125000`

joeywun, 2018-07-06:
I have also encountered the same problem. I found that if I read from the same address again, the first two characters are missing. I was thinking if there is something wrong in my sketch. I tried to find it but I couldn't. So I go back and use the example "readWriteString" and modified it a little by introducing a second test string. Then in the main loop, I read from first and second addresses alternately. When I test it, not only that the first two characters are missing, the last three characters are missing on my second loop. Then the loop just freeze. 

#include<SPIMemory.h>

uint32_t strAddr1;
uint32_t strAddr2;

#if defined(ARDUINO_SAMD_ZERO) && defined(SERIAL_PORT_USBVIRTUAL)
// Required for Serial on Zero based boards
#define Serial SERIAL_PORT_USBVIRTUAL
#endif

#if defined (SIMBLEE)
#define BAUD_RATE 250000
#define RANDPIN 1
#else
#define BAUD_RATE 115200
#define RANDPIN A0
#endif

//SPIFlash flash(SS1, &SPI1);       //Use this constructor if using an SPI bus other than the default SPI. Only works with chips with more than one hardware SPI bus
SPIFlash flash;

bool readSerialStr(String &inputStr);

void setup() {
  Serial.begin(BAUD_RATE);
#if defined (ARDUINO_SAMD_ZERO) || (__AVR_ATmega32U4__)
  while (!Serial) ; // Wait for Serial monitor to open
#endif

  flash.begin();

  randomSeed(analogRead(RANDPIN));
  strAddr1 = random(0, flash.getCapacity());
  strAddr2 = random(0, flash.getCapacity());
  String inputString1 = "This is test String 1";
  String inputString2 = "This is test String 2";
  
  flash.writeStr(strAddr1, inputString1);
  flash.writeStr(strAddr2, inputString2);
  Serial.print(F("Written string: "));
  Serial.println(inputString1);
  Serial.print(F("To address: "));
  Serial.println(strAddr1);

  Serial.print(F("Written string: "));
  Serial.println(inputString2);
  Serial.print(F("To address: "));
  Serial.println(strAddr2);
  
}

void loop() {

 String outputString="";

  if (flash.readStr(strAddr1, outputString)) {
    Serial.print(F("Read string: "));
    Serial.println(outputString);
    Serial.print(F("From address: "));
    Serial.println(strAddr1);
    outputString = "";
  }
  delay(100);
  if (flash.readStr(strAddr2, outputString)) {
    Serial.print(F("Read string: "));
    Serial.println(outputString);
    Serial.print(F("From address: "));
    Serial.println(strAddr2);
     outputString = "";
  }
  delay(100);
}

//Reads a string from Serial
bool readSerialStr(String &inputStr) {
  if (!Serial)
    Serial.begin(115200);
  while (Serial.available()) {
    inputStr = Serial.readStringUntil('\n');
    Serial.println(inputStr);
    return true;
  }
  return false;
}

The result looks like this:-

[image]

Library version : 3.2.1
Arduino IDE : 1.8.4
OS : Win7 Home Premium
Microcontroller : Arduino Mega 2560
Flash memory module : W25Q128

Marzogh, 2018-08-09:
Uh oh! Thanks for catching this guys. I'll look into it right away.

My apologies for the long delay in replying. I was away for health reasons and am slowly transitioning back into normality.

Marzogh, 2018-08-10:
This is fixed as of commit 703f609781c7d8e866f79f415ef5e254cb324799 . Thanks again for catching this.

closed 2018-08-10 by Marzogh