Marzogh/SPIMemory #145: Division by zero in the SPI Flash::getMaxPage method

lebfr, 2018-07-21:
[image] [image]
[image] [image] [image]
[image] [image]
###### ---------------------------- **DO NOT DELETE OR EDIT** anything above this line ----------------------------
###### ---------------------------- **DO NOT DELETE OR EDIT** anything above this line ----------------------------

Hello!

The _pageSize variable is always zero if SFDP discovery is disabled.
-------------------------------------------
file: SPIMemory.h
#define RUNDIAGNOSTIC
--------------------------------------------
file: SPIflash.cpp
//Returns maximum number of pages
uint32_t SPIFlash::getMaxPage(void) {
        Serial.print("_pageSize: ");
        Serial.println(_pageSize); // --> We get zero in the output
	return (_chip.capacity / _pageSize);
}
--------------------------------------------------
Output from SPIFlash::getMaxPage method:

Max Pages: _pageSize: 0
4294967295

We get an incorrect number of Max Pages.

------------------------------------------------------
Full output from sketch:
Initialising..........
Chip Diagnostics initiated.

No Chip size defined by user. Automated identification initiated.
Chip identified. This chip is fully supported by the library.

SPIMemory Library version: 3.2.1

JEDEC ID: 0xEF4016
Man ID: 0xEF
Memory ID: 0x40
Capacity: 4194304
Max Pages: _pageSize: 0
4294967295
Unique ID: 283840304344352, 0x671887C3351C2C00
-----------------------------------------------------------------------------------------------------------------------------
							Testing library code
-----------------------------------------------------------------------------------------------------------------------------
			Function		Test result			     Runtime
-----------------------------------------------------------------------------------------------------------------------------
			Power Down		   FAIL			Not all chips support power down. Check your datasheet.
			Power Up		   PASS				      64 us

			Erase Chip		   PASS				      8.895 s
			Erase 72KB		   PASS				      228.60 ms
			Erase 64KB		   PASS				      175.69 ms
			Erase 32KB		   PASS				      120.51 ms
			Erase 4KB		   PASS				      36.85 ms
-----------------------------------------------------------------------------------------------------------------------------
			Data type		I/O Result	      Write time	      Read time
-----------------------------------------------------------------------------------------------------------------------------
			Byte			   PASS			164 us			68 us
			Char			   PASS			160 us			60 us
			Word			   PASS			164 us			64 us
			Short			   PASS			164 us			64 us
			ULong			   PASS			204 us			72 us
			Long			   PASS			176 us			68 us
			Float			   PASS			196 us			72 us
			Struct			   PASS			312 us			100 us
			Byte Array		   PASS			2.66 ms			444 us
			String			   PASS			204 us			76 us
-----------------------------------------------------------------------------------------------------------------------------

lebfr, 2018-07-21:
`uint32_t SPIFlash::getMaxPage(void) {
	#ifdef USES_SFDP
		return (_chip.capacity / _pageSize);
	#else
		return (_chip.capacity / SPI_PAGESIZE);
   }`

Marzogh, 2018-08-09:
Uh oh! good catch. I'll fix this up right away and roll it into the next release.

My apologies for the long delay in replying. I was away for health reasons and am slowly transitioning back into normality.

Marzogh, 2018-08-09:
Actually, now that I've just gone back and looked at what I'd been working on before I fell ill, this issue has been fixed as of the 31st of May in the code currently in the development branch and will be rolled out into the next release. 🙂 

Commit 5bb3f0fabb4b4772fe6d4baf7ef71bf3534f655d closes #145

closed 2018-08-09 by Marzogh