Marzogh/SPIMemory #146: Support for FRAM chips ?

rajibgupta, 2018-08-07:
Since FRAM chips are more stable and comes with more read/write cycles, I would like to prefer that.

Does this library support any FRAM chip ?

Marzogh, 2018-08-09:
Hi mate, I'm working on FRAM support into the next major release. In the meantime, feel free to keep an eye on the code in the development branch. As of today it supports the FM25W256 from Cypress. The documentation for this is not ready yet so you'll have to dig through the library to figure out how it works.

I've tried to keep it simple and identical to the API that's used to access SPI Flash memory, so you won't need to make big changes to your pre-existing code - but be warned, it is a work in progress.

Do not depend on code in the _development_ branch for production ready stuff as new commits might break existing functionality. Once I've got the API properly baked in, I'll roll the code into the _stable_ branch while I work on documentation. 😄

Marzogh, 2019-04-20:
v3.3.0 has been released today and is the first version to officially support FRAM chips. 🙂

closed 2019-04-20 by Marzogh