Marzogh/SPIMemory #147: Winbond w25n01GV support

akareembedri, 2018-08-07:
Hi,
I'm using the Winbond w25n01GV in a project and I'm having a hard time running initial tests on it. I'm using your TestFlash example. I managed to obtain the correct JEDEC ID from it after making a few modifications to the library, but I couldn't write anything to memory as the error check always flags the write operation. The read functions seems to work well but I have no way to verify it since I can't write to memory.
Please let me know if your library can support the Winbond w25n01GV and I would appreciate any adivce on how to run the write/read instructions.

Thank you! 

[image] [image]
[image] [image] [image]
[image] [image]
###### ---------------------------- **DO NOT DELETE OR EDIT** anything above this line ----------------------------
###### ---------------------------- **DO NOT DELETE OR EDIT** anything above this line ----------------------------

Hey there! Thanks for using the SPIFlash library for Arduino.
### Please note that starting 01.03.2018 any issue raised here MUST be submitted according to this template or it will be flagged with 'Not enough information'. No action will be taken till all the prerequisite information is provided. If no information is provided for over a month, the issue will be closed.

**Note: For support questions, please use the [Arduino Forums](http://forum.arduino.cc/index.php?topic=324009.0). This repository's issues are reserved for feature requests and bug reports.**

# Issue details

**I'm submitting a ...**
  - [ ] bug report
  - [ ] feature request
  - [ ] support request => Please do not submit support request here, see note at the top of this template.

Do the checklist before filing an issue:

- [ ] Is this something you can **debug and fix**? Send a pull request! Bug fixes and documentation fixes are welcome.
- [ ] Is this an idea for a feature? Post the feature request as an issue here with a title that clearly says "Feature Request".
- [ ] Is this a bug that you cannot fix? Post the bug report as an issue here with a title that cleary says "Bug Report".

When opening an issue please include the following details:
-------------------------------------------------------------

- [ ] Do you want to request a *feature* or report a *bug*?
- [ ] Which library version are you using?
- [ ] Which Arduino IDE version are you compiling the library on?
- [ ] Which OS are you compiling your code on? [Linux / Mac / Windows]
- [ ] Which microcontroller platform are you using?
- [ ] Which Flash memory module are you using?
- [ ] What is the current behavior?
- [ ] What is the expected behavior?

--------------------------
###### Bug reports only
- If this is a bug report -

- [ ] Make sure you have run FlashDiagnostics.ino with ``` #define RUNDIAGNOSTICS ``` uncommented in SPIFlash.h. **List any error codes** that pop up in your Serial output when you run FlashDiagnostics.ino.here:
      Error codes
      ------------
        -
        -
        -
        -
        -
- [ ] If you have a problem with a particular function, call the flash.error() function (after you have made sure you have started up your Serial port with a ``` Serial.begin(BAUD) ``` ). Provide details of the function, the data given to/ expected from the function and the error code here: (**Please repeat this for every function you have an error with***)
        Function -
        Data -
        Error code -
- [ ] Provide a **minimal code snippet** example that reproduces the bug. Please make sure you wrap any code in the proper code blocks like below
```
```CODE HERE```
```
- [ ] Provide **screenshots** where appropriate
- [ ] Provide **wiring diagram** where appropriate

-----------------------------
###### Feature Requests only

- [ ] If this is a feature request, what is the motivation / use case for changing the behavior?

----------------------------

- [ ] Other information (e.g. detailed explanation, related issues, suggestions how to fix, links for us to have context etc.)

_Make sure to add **all the information needed to understand the bug/feature** so that someone can help. If the info is missing we'll add the 'Needs more information' label and close the issue until there is enough information._

Marzogh, 2018-08-09:
Hi mate, unfortunately I do not have any of the Winbond W25N series flash memory chips with me and I won't be able to test the library with them till I get some. I'm going to try and get my hands on some and I'll keep you posted.

In the meantime, could you try this and let me know what the results are please?

```
//Change any function that looks like this

flash.writeShort(addressToWriteto, dataToWrite);
//or
flash.writeByteArray(addressToWriteTo, bufferToWriteFrom, bufferSize);

//to this

flash.writeShort(addressToWriteto, dataToWrite, NOERRCHK);
//or
flash.writeByteArray(addressToWriteTo, bufferToWriteFrom, bufferSize, NOERRCHK);
```

This will disable error checking on the write functions and will let us know if the problem is with the actual act of writing to the chip (in which case I'll have to get my hands on one before I can do anything else) or if the `errorCheck()` function is broken (which might be easier to fix without the actual chip in hand)

davidraclet, 2018-08-30:
Hi,
Just by running the example 'TestFlash', I confirm JEDEC function doesn't work. 

Maybe the architecture of this memory (1Gbit) is a little bit different from the smaller tested. According specifications :

> The W25N01GV 1G-bit memory array is organized into 65,536 programmable pages of 2,048-bytes each.
> The entire page can be programmed at one time using the data from the 2,048-Byte internal buffer. Pages
> can be erased in groups of 64 (128KB block erase). The W25N01GV has 1,024 erasable blocks.

The diagram in the specs shows each page capacity is 2KB, divided in 4 sectors of 512 Bytes.

The page buffer is not 256 bytes but 2048 and erase function seems to work only on one block (= 64 pages = 128KBytes). The example initializes 
`uint8_t pageBuffer[256];`
Do you think this architecture is working with your library ?

Thank you.
David

TimoteoMendes, 2018-09-07:
Hello Marzogh,
Any news?
I have been performing some tests with this and the 2Gb version, a stack of 2x 1Gb.
If I could help in the development it would be nice as I also need a good solution to use this memories.
For now because the ATMEGA328 is a bit limited I decided to use the STM32 with the duino version from Roger.
The commands in this memory are a bit different, here the list I have been tested and worked out:
//set instructions
#define W_EN   0x06  //write enable
#define W_DE  0x04  //write disable
#define R_SR 0x05  //read status reg
#define W_SR  0x01  //write status reg
#define DEV_RST 0xFF //reset the device ~500us
#define DIE_ST 0xC2 //command to change DIE 0x00 ou 0x01 depois
#define PAGE_PGM  0x02  //page program
#define BLK_E_128K 0xD8  //block erase 128KB
#define PROGRAM_EXEC 0x10  //grava o buffer (2048 ou 2112) para a pagina
#define LOAD_PROGRAM 0x02  //escrever no buffer de 1 a 2048 Bytes

#define PDWN    0xB9  //power down
#define R_JEDEC_ID  0x9F  //read JEDEC ID = Manuf+ID (suggested)
//read instructions
#define READ    0x03
#define READ_PAGE 0x13
#define FAST_READ 0x0B

Thanks,
Tim

sslupsky, 2019-10-14:
I had a quick look at the datasheet and these parts are NAND flash.  Does this library support NAND flash devices?

Montvydas, 2024-11-18:
From the codebase I don't see this supporting NAND flashes. These use inherently different addresses and different command sets e.g. writing into status register requires sending command to "set feature" followed by the status register, followed by the value instead of directly accessing the status register and writing value. Although you should still be able to read Manufacturer ID and chip ID without issues, since that part uses the same address.