Marzogh/SPIMemory #149: Add Support for MX25R3235F

ForrestErickson, 2018-08-08:
[image] [image]
[image] [image] [image]
[image] [image]
###### ---------------------------- **DO NOT DELETE OR EDIT** anything above this line ----------------------------
###### ---------------------------- **DO NOT DELETE OR EDIT** anything above this line ----------------------------

Hey there! Thanks for using the SPIFlash library for Arduino.
### Please note that starting 01.03.2018 any issue raised here MUST be submitted according to this template or it will be flagged with 'Not enough information'. No action will be taken till all the prerequisite information is provided. If no information is provided for over a month, the issue will be closed.

**Note: For support questions, please use the [Arduino Forums](http://forum.arduino.cc/index.php?topic=324009.0). This repository's issues are reserved for feature requests and bug reports.**

# Issue details

**I'm submitting a ...**
  - [ ] bug report
  - [ ] feature request
  - [x ] support request => Please do not submit support request here, see note at the top of this template.

I see no note at the top of the template regarding support requests.

Do the checklist before filing an issue:

- [ ] Is this something you can **debug and fix**? Send a pull request! Bug fixes and documentation fixes are welcome.
- [ ] Is this an idea for a feature? Post the feature request as an issue here with a title that clearly says "Feature Request".
- [ ] Is this a bug that you cannot fix? Post the bug report as an issue here with a title that cleary says "Bug Report".

When opening an issue please include the following details:
-------------------------------------------------------------

- [ ] Do you want to request a *feature* or report a *bug*?  I thinks support, but perhaps feature. New chip support.
- [ ] Which library version are you using?  v.3.2.1
- [ ] Which Arduino IDE version are you compiling the library on?  Windows 1.8.5
- [ ] Which OS are you compiling your code on? [Linux / Mac / Windows]  Windows 7 64 bit.
- [ ] Which microcontroller platform are you using?  Arduino Uno
- [ ] Which Flash memory module are you using?  MX25R3235F (This chip is internatl to the Texas Instruments CC3220MOD on a WiFi Launchpad.
- [ ] What is the current behavior?  I try to read the Manufacturier ID and get inconsistant results. I get C4, 80, 88. The FlashDiagnostics throws many errors (see below). 
- [ ] What is the expected behavior?  Was expecting to read Manufacturer ID of "C2" hex.

--------------------------
###### Bug reports only
- If this is a bug report -

- [ ] Make sure you have run FlashDiagnostics.ino with ``` #define RUNDIAGNOSTICS ``` uncommented in SPIFlash.h. **List any error codes** that pop up in your Serial output when you run FlashDiagnostics.ino.here:
      Error codes
      ------------
Chip Diagnostics initiated.

Error code: 0x10
Error code: 0x02

SPIMemory Library version: 3.2.1

JEDEC ID: 0x8040A0
Man ID: 0x80
Memory ID: 0x40
Capacity: 0
Max Pages: 4294967295

-----------------------------------------------------------------------------------------------------------------------------
							Testing library code
-----------------------------------------------------------------------------------------------------------------------------
			Function		Test result			     Runtime
-----------------------------------------------------------------------------------------------------------------------------
			Power Down		   PASS				      108 us
			Power Up		   PASS				      96 us

			Erase Chip		   PASS				      6.30 ms
			Erase 72KB	Error code: 0x01
	   FAIL	
			Erase 64KB	Error code: 0x01
[... 40 lines cut ...]
        -
        -
        -
        -
- [ ] If you have a problem with a particular function, call the flash.error() function (after you have made sure you have started up your Serial port with a ``` Serial.begin(BAUD) ``` ). Provide details of the function, the data given to/ expected from the function and the error code here: (**Please repeat this for every function you have an error with***)
        Function -
        Data -
        Error code -
- [ ] Provide a **minimal code snippet** example that reproduces the bug. Please make sure you wrap any code in the proper code blocks like below
```
```CODE HERE```
```
- [ ] Provide **screenshots** where appropriate
- [ ] Provide **wiring diagram** where appropriate

-----------------------------
###### Feature Requests only

- [ ] If this is a feature request, what is the motivation / use case for changing the behavior?

----------------------------

- [ ] Other information (e.g. detailed explanation, related issues, suggestions how to fix, links for us to have context etc.)

_Make sure to add **all the information needed to understand the bug/feature** so that someone can help. If the info is missing we'll add the 'Needs more information' label and close the issue until there is enough information._

ForrestErickson, 2018-08-08:
Regarding the Feature Request, and motivation. This part is in an interesting Texas Instrument Launchpad for developing wireless IOT applications.

Marzogh, 2019-04-20:
I'm afraid I do not have the CC3220MOD and will probably not be able to get my hands on one in the near future. I've added the 'help wanted' tag to this issue and will leave it open to see if anyone else can help.

My apologies for the late reply - I've been away for the last year-ish due to health issues.