Marzogh/SPIMemory #151: writeAnything bug

400081, 2018-08-12:
[image] [image]
[image] [image] [image]
[image] [image]
###### ---------------------------- **DO NOT DELETE OR EDIT** anything above this line ----------------------------

# Issue details

**I'm submitting a ...**
  - [x] bug report

Do the checklist before filing an issue:

- [x] Is this a bug that you cannot fix? Post the bug report as an issue here with a title that cleary says "Bug Report".

When opening an issue please include the following details:
-------------------------------------------------------------

- [x] Do you want to request a *feature* or report a *bug*?  - bug
- [x] Which library version are you using? - 3.2.1
- [x] Which Arduino IDE version are you compiling the library on? - 1.8.5
- [x] Which OS are you compiling your code on? [Linux / Mac / Windows] - Windows
- [x] Which microcontroller platform are you using? - arduino
- [x] Which Flash memory module are you using? - W25Q16BV
- [x] What is the current behavior? - writeAnything - Structure of size 12, spans memory boundary is not written

- [ ] Provide a **minimal code snippet** example that reproduces the bug. Please make sure you wrap any code in the proper code blocks like below
```#include<SPIMemory.h>

SPIFlash flash;

typedef struct {
  uint32_t a;
  uint32_t b;
  uint32_t c;
} Test;

void setup() {
  Serial.begin(115200);
  flash.begin();
  uint32_t addr = 250;
  Test data_w;
  Test data_r;
  flash.eraseBlock32K(addr);
    data_w.a = 1;
    data_w.b = 2;
    data_w.c = 3;
    flash.writeAnything(addr,data_w);
    flash.readAnything(addr, data_r);
    Serial.print(data_r.a);Serial.print("\t");
    Serial.print(data_r.b);Serial.print("\t");
    Serial.println(data_r.c);
}

void loop() {}   ```

Marzogh, 2018-08-16:
Thanks for catching this. Page boundary bugs are turning out to be the bane of my existence 😉 

I'll roll the fix into the next release.

taHC81, 2019-03-13:
Hi all, i'm getting failed (Write Function has failed errorcheck) on structTest() with Spansion/Cypress S25FL164K, running on ATSAMD21E18A. Any suggestions for further troubleshooting? Thanks in advance.

commit cd9d4339 (2019-04-19): Merge pull request #165 from jacky4566/master

Update SPIFlash.h
Fixes #151
closed 2019-04-19 by Marzogh