Marzogh/SPIMemory #152: [Bug Report] String data corrupted when written in a loop by later define strings

jeraymond, 2018-09-09:
[image] [image]
[image] [image] [image]
[image] [image]
###### ---------------------------- **DO NOT DELETE OR EDIT** anything above this line ----------------------------
###### ---------------------------- **DO NOT DELETE OR EDIT** anything above this line ----------------------------

# Issue details

**I'm submitting a ...**
  - [x] bug report
  - [ ] feature request
  - [ ] support request => Please do not submit support request here, see note at the top of this template.

Do the checklist before filing an issue:

- [ ] Is this something you can **debug and fix**? Send a pull request! Bug fixes and documentation fixes are welcome.
- [ ] Is this an idea for a feature? Post the feature request as an issue here with a title that clearly says "Feature Request".
- [x] Is this a bug that you cannot fix? Post the bug report as an issue here with a title that cleary says "Bug Report".

When opening an issue please include the following details:
-------------------------------------------------------------

- [ ] Do you want to request a *feature* or report a *bug*?
Bug
- [ ] Which library version are you using?
3.2.1
- [ ] Which Arduino IDE version are you compiling the library on?
1.8.4
- [ ] Which OS are you compiling your code on? [Linux / Mac / Windows]
macOS
- [ ] Which microcontroller platform are you using?
ATmega32u4 (Adafruit 32u4 feather)
- [ ] Which Flash memory module are you using?
[W25Q128JVSIM-ND](https://www.digikey.com/product-detail/en/winbond-electronics/W25Q128JVSIM/W25Q128JVSIM-ND/6819721)
- [ ] What is the current behavior?
When String data is written inside of a loop with `flash.writeStr()`, it becomes corrupted by the first string defined that follows outside of the loop. The next defined String overwrites the first part of the written data. Performing the same write out of a loop writes the data as expected.

- [ ] What is the expected behavior?
No data corruption.

--------------------------
###### Bug reports only
- If this is a bug report -

- [ ] Make sure you have run FlashDiagnostics.ino with ``` #define RUNDIAGNOSTICS ``` uncommented in SPIFlash.h. **List any error codes** that pop up in your Serial output when you run FlashDiagnostics.ino.here:
      Error codes
      ------------
        - None. The results were clean showing PASS for all Test results.

- [ ] If you have a problem with a particular function, call the flash.error() function (after you have made sure you have started up your Serial port with a ``` Serial.begin(BAUD) ``` ). Provide details of the function, the data given to/ expected from the function and the error code here: (**Please repeat this for every function you have an error with***)

I printed the error codes after the write and read operations and they both showed `0`. See also the **Program output**.
        
- [ ] Provide a **minimal code snippet** example that reproduces the bug. Please make sure you wrap any code in the proper code blocks like below
```c++
#include <SPIMemory.h>

SPIFlash flash(A5);

void setup() {
    Serial.begin(115200);
    while(!Serial);

    if (!flash.begin()) {
        Serial.println("Flash initialization failed");
        return;
    }

    uint32_t addr = 0;
    bool success = false;
    while (!success) {
        success = flash.eraseSector(addr);
        Serial.print("Erase error="); Serial.println(flash.error());
    }
    Serial.println("Erased sector");

    // The data written via flash.writeString() gets corrupted by
    // String corruptor that follows this single iteration loop. If the
    // code inside this loop is moved outside of the loop no corruption takes
    // place. 
    for (int i = 0; i < 1; i++) {
        String writeString = "This is the string data that is expected.";
        Serial.print("Writing: "); Serial.println(writeString);
        success = flash.writeStr(addr, writeString);
        Serial.print("Write error="); Serial.println(flash.error());
        if (success) {
            Serial.println("Writing to flash succeeded");
        } else {
            Serial.println("Writing string to flash failed");
        }
    }

    String corrputor = "How does this get written?";

    String outputString = "";
    if (flash.readStr(addr, outputString)) {
        Serial.print(F("Read string: "));
        Serial.println(outputString);
        Serial.print(F("From address: "));
        Serial.println(addr);
    }
    Serial.print("Read error="); Serial.println(flash.error());

}

void loop() {
   
}
```

**Program output**

```
Chip Diagnostics initiated.

No Chip size defined by user. Automated identification initiated.
Chip identified. This chip is fully supported by the library.
Erase error=0
Erased sector
Writing: This is the string data that is expected.
Write error=0
Writing to flash succeeded
Read string: How does this get written?   s expected.
From address: 0
Read error=0

```

----------------------------

- [ ] Other information (e.g. detailed explanation, related issues, suggestions how to fix, links for us to have context etc.)

_Make sure to add **all the information needed to understand the bug/feature** so that someone can help. If the info is missing we'll add the 'Needs more information' label and close the issue until there is enough information._

rambo, 2018-10-13:
Just a note, I personally found the String class to be very flaky, leading to weird crashes and what not when used in loops or functions called multiple times, whether initialized once and modified or initialized multiple times.

Using good old fashioned C-strings, ie char arrays and `writeCharArray` / `readCharArray` I've found to be reliable.

jeraymond, 2018-10-13:
I switched to a [different library](https://github.com/PaulStoffregen/SerialFlash) and it’s been rock solid.

Marzogh, 2019-04-20:
@jeraymond This has been fixed in v3.3.0 which I should be rolling out in a few hours. Running your code with v3.3.0 produces the following output.

```
Chip Diagnostics initiated.

No Chip size defined by user. Checking library support.
Chip identified. This chip is fully supported by the library.
Erase error=0
Erased sector
Writing: This is the string data that is expected.
Write error=0
Writing to flash succeeded
Read string: This is the string data that is expected.
From address: 0
Read error=0

```

P.S @rambo I agree with you about the String class. I've had just about enough with them and the latest version of writeStr() actually breaks down the String into a char array for writing. Its rock solid and doesn't slow the process down by much. Personally, I think the stability is worth the tiny delay. 🙂 

P.P.S My apologies for the delay - I have been away for a while due to health issues.

closed 2019-04-20 by Marzogh