Marzogh/SPIMemory #155: not the same result after each run with flashDiagnostic

cawena, 2018-09-29:
[image] [image]
[image] [image] [image]
[image] [image]
###### ---------------------------- **DO NOT DELETE OR EDIT** anything above this line ----------------------------
###### ---------------------------- **DO NOT DELETE OR EDIT** anything above this line ----------------------------

Hey there! Thanks for using the SPIFlash library for Arduino.
### Please note that starting 01.03.2018 any issue raised here MUST be submitted according to this template or it will be flagged with 'Not enough information'. No action will be taken till all the prerequisite information is provided. If no information is provided for over a month, the issue will be closed.

**Note: For support questions, please use the [Arduino Forums](http://forum.arduino.cc/index.php?topic=324009.0). This repository's issues are reserved for feature requests and bug reports.**

# Issue details

**I'm submitting a ...**
  - [X ] bug report
  - [ ] feature request
  - [ ] support request => Please do not submit support request here, see note at the top of this template.

Do the checklist before filing an issue:

- [ ] Is this something you can **debug and fix**? Send a pull request! Bug fixes and documentation fixes are welcome.
- [ ] Is this an idea for a feature? Post the feature request as an issue here with a title that clearly says "Feature Request".
- [X] Is this a bug that you cannot fix? Post the bug report as an issue here with a title that cleary says "Bug Report".

When opening an issue please include the following details:
-------------------------------------------------------------

- [bug] Do you want to request a *feature* or report a *bug*?
- [3.2.1] Which library version are you using?
- [1.8.7] Which Arduino IDE version are you compiling the library on?
- [macOS 10.13.6] Which OS are you compiling your code on? [Linux / Mac / Windows]
- [Atmel 2560] Which microcontroller platform are you using?
- [W25Q16BV] Which Flash memory module are you using?
- [Test Sketch flashDiagnostics from your examples return different results] What is the current behavior?
- [always the same result] What is the expected behavior?

--------------------------
###### Bug reports only
- If this is a bug report -

- [ ] Make sure you have run FlashDiagnostics.ino with ``` #define RUNDIAGNOSTICS ``` uncommented in SPIFlash.h. **List any error codes** that pop up in your Serial output when you run FlashDiagnostics.ino.here:
      Error codes
      ------------
Initialising..........

SPIMemory Library version: 3.2.1

JEDEC ID: 0xEF4014
Man ID: 0xEF
Memory ID: 0x40
Capacity: 1048576
Max Pages: 4294967295
Unique ID: 3641690526130432, 0x63D4269926312100
-----------------------------------------------------------------------------------------------------------------------------
							Testing library code
-----------------------------------------------------------------------------------------------------------------------------
			Function		Test result			     Runtime
-----------------------------------------------------------------------------------------------------------------------------
			Power Down		   PASS				      0 us
			Power Up		   PASS				      0 us

			Erase Chip		   PASS				      0 us
			Erase 72KB		   PASS				      0 us
			Erase 64KB		   PASS				      0 us
			Erase 32KB		   PASS				      0 us
			Erase 4KB		   PASS				      0 us
-----------------------------------------------------------------------------------------------------------------------------
			Data type		I/O Result	      Write time	      Read time
-----------------------------------------------------------------------------------------------------------------------------
			Byte			   FAIL				      0 us
[... 30 lines cut ...]
			Erase 72KB		   PASS				      0 us
			Erase 64KB		   PASS				      0 us
			Erase 32KB		   PASS				      0 us
			Erase 4KB		   PASS				      0 us
-----------------------------------------------------------------------------------------------------------------------------
			Data type		I/O Result	      Write time	      Read time
-----------------------------------------------------------------------------------------------------------------------------
			Byte			   PASS				      0 us
			Char			   PASS				      0 us
			Word			   PASS				      0 us
			Short			   PASS				      0 us
			ULong			   PASS				      0 us
			Long			   FAIL				      0 us
			Float			   PASS				      0 us
			Struct			   FAIL				      0 us
			Byte Array		   FAIL	4, 	   FAIL	5, 	   FAIL	6, 	   FAIL	7, 	   FAIL	8, 	   FAIL	9, 	   FAIL	10, 	   FAIL	11, 	   FAIL	12, 	   FAIL	13, 	   FAIL	14, 	   FAIL	15, 	   FAIL	16, 	   FAIL	17, 	   FAIL	18, 	   FAIL	19, 	   FAIL	20, 	   FAIL	21, 	   FAIL	22, 	   FAIL	23, 	   FAIL	24, 	   FAIL	25, 	   FAIL	26, 	   FAIL	27, 	   FAIL	28, 	   FAIL	29, 	   FAIL	30, 	   FAIL	31, 	   FAIL	32, 	   FAIL	33, 	   FAIL	34, 	   FAIL	35, 	   FAIL	36, 	   FAIL	37, 	   FAIL	38, 	   FAIL	39, 	   FAIL	40, 	   FAIL	41, 	   FAIL	42, 	   FAIL	43, 	   FAIL	44, 	   FAIL	45, 	   FAIL	46, 	   FAIL	47, 	   FAIL	48, 	   FAIL	49, 	   FAIL	50, 	   FAIL	51, 	   FAIL	52, 	   FAIL	53, 	   FAIL	54, 	   FAIL	55, 	   FAIL	56, 	   FAIL	57, 	   FAIL	58, 	   FAIL	59, 	   FAIL	60, 	   FAIL	61, 	   FAIL	62, 	   FAIL	63, 	   FAIL	64, 	   FAIL	65, 	   FAIL	66, 	   FAIL	67, 	   FAIL	68, 	   FAIL	69, 	   FAIL	70, 	   FAIL	71, 	   FAIL	72, 	   FAIL	73, 	   FAIL	74, 	   FAIL	75, 	   FAIL	76, 	   FAIL	77, 	   FAIL	78, 	   FAIL	79, 	   FAIL	80, 	   FAIL	81, 	   FAIL	82, 	   FAIL	83, 	   FAIL	84, 	   FAIL	85, 	   FAIL	86, 	   FAIL	87, 	   FAIL	88, 	   FAIL	89, 	   FAIL	90, 	   FAIL	91, 	   FAIL	92, 	   FAIL	93, 	   FAIL	94, 	   FAIL	95, 	   FAIL	96, 	   FAIL	97, 	   FAIL	98, 	   FAIL	99, 	   FAIL	100, 	   FAIL	101, 	   FAIL	102, 	   FAIL	103, 	   FAIL	104, 	   FAIL	105, 	   FAIL	106, 	   FAIL	107, 	   FAIL	108, 	   FAIL	109, 	   FAIL	110, 	   FAIL	111, 	   FAIL	112, 	   FAIL	113, 	   FAIL	114, 	   FAIL	115, 	   FAIL	116, 	   FAIL	117, 	   FAIL	118, 	   FAIL	119, 	   FAIL	120, 	   FAIL	121, 	   FAIL	122, 	   FAIL	123, 	   FAIL	124, 	   FAIL	125, 	   FAIL	126, 	   FAIL	127, 	   FAIL	128, 	   FAIL	129, 	   FAIL	130, 	   FAIL	131, 	   FAIL	132, 	   FAIL	133, 	   FAIL	134, 	   FAIL	135, 	   FAIL	136, 	   FAIL	137, 	   FAIL	138, 	   FAIL	139, 	   FAIL	140, 	   FAIL	141, 	   FAIL	142, 	   FAIL	143, 	   FAIL	144, 	   FAIL	145, 	   FAIL	146, 	   FAIL	147, 	   FAIL	148, 	   FAIL	149, 	   FAIL	150, 	   FAIL	151, 	   FAIL	152, 	   FAIL	153, 	   FAIL	154, 	   FAIL	155, 	   FAIL	156, 	   FAIL	157, 	   FAIL	158, 	   FAIL	159, 	   FAIL	160, 	   FAIL	161, 	   FAIL	162, 	   FAIL	163, 	   FAIL	164, 	   FAIL	165, 	   FAIL	166, 	   FAIL	167, 	   FAIL	168, 	   FAIL	169, 	   FAIL	170, 	   FAIL	171, 	   FAIL	172, 	   FAIL	173, 	   FAIL	174, 	   FAIL	175, 	   FAIL	176, 	   FAIL	177, 	   FAIL	178, 	   FAIL	179, 	   FAIL	180, 	   FAIL	181, 	   FAIL	182, 	   FAIL	183, 	   FAIL	184, 	   FAIL	185, 	   FAIL	186, 	   FAIL	187, 	   FAIL	188, 	   FAIL	189, 	   FAIL	190, 	   FAIL	191, 	   FAIL	192, 	   FAIL	193, 	   FAIL	194, 	   FAIL	195, 	   FAIL	196, 	   FAIL	197, 	   FAIL	198, 	   FAIL	199, 	   FAIL	200, 	   FAIL	201, 	   FAIL	202, 	   FAIL	203, 	   FAIL	204, 	   FAIL	205, 	   FAIL	206, 	   FAIL	207, 	   FAIL	208, 	   FAIL	209, 	   FAIL	210, 	   FAIL	211, 	   FAIL	212, 	   FAIL	213, 	   FAIL	214, 	   FAIL	215, 	   FAIL	216, 	   FAIL	217, 	   FAIL	218, 	   FAIL	219, 	   FAIL	220, 	   FAIL	221, 	   FAIL	222, 	   FAIL	223, 	   FAIL	224, 	   FAIL	225, 	   FAIL	226, 	   FAIL	227, 	   FAIL	228, 	   FAIL	229, 	   FAIL	230, 	   FAIL	231, 	   FAIL	232, 	   FAIL	233, 	   FAIL	234, 	   FAIL	235, 	   FAIL	236, 	   FAIL	237, 	   FAIL	238, 	   FAIL	239, 	   FAIL	240, 	   FAIL	241, 	   FAIL	242, 	   FAIL	243, 	   FAIL	244, 	   FAIL	245, 	   FAIL	246, 	   FAIL	247, 	   FAIL	248, 	   FAIL	249, 	   FAIL	250, 	   FAIL	251, 	   FAIL	252, 	   FAIL	253, 	   FAIL	254, 	   FAIL	255, 	   PASS				      0 us
			String			   FAIL				      0 us
-----------------------------------------------------------------------------------------------------------------------------
To see function runtimes ncomment RUNDIAGNOSTIC in SPIMemory.h.

        -
- [ ] Provide a **minimal code snippet** example that reproduces the bug. Please make sure you wrap any code in the proper code blocks like below
'''
it's the example sketch flashDiagnostics
'''

cawena, 2018-09-29:
uncomment RUNDIAGNOSTIC in SPIMemory.h

```

Initialising..........
Chip Diagnostics initiated.

No Chip size defined by user. Automated identification initiated.
Chip identified. This chip is fully supported by the library.

SPIMemory Library version: 3.2.1

JEDEC ID: 0xEF4014
Man ID: 0xEF
Memory ID: 0x40
Capacity: 1048576
Max Pages: 4294967295
Unique ID: 3641690526130432, 0x63D4269926312100
-----------------------------------------------------------------------------------------------------------------------------
							Testing library code
-----------------------------------------------------------------------------------------------------------------------------
			Function		Test result			     Runtime
-----------------------------------------------------------------------------------------------------------------------------
			Power Down		   PASS				      88 us
			Power Up		   PASS				      80 us

			Erase Chip		   PASS				      1.163 s
			Erase 72KB		   PASS				      117.19 ms
			Erase 64KB		   PASS				      88.92 ms
			Erase 32KB		   PASS				      66.56 ms
			Erase 4KB		   PASS				      27.66 ms
-----------------------------------------------------------------------------------------------------------------------------
			Data type		I/O Result	      Write time	      Read time
-----------------------------------------------------------------------------------------------------------------------------
			Byte			   PASS			184 us			76 us
			Char			   FAIL			0 us			68 us
			Word			   FAIL			0 us			72 us
			Short			   FAIL			0 us			68 us
Error code: 0x07
			ULong			   FAIL			0 us			84 us
			Long			   FAIL			204 us			76 us
			Float			   FAIL			0 us			72 us
Error code: 0x0A
			Struct			   FAIL			0 us			128 us
			Byte Array		   FAIL	0, 	   FAIL	1, 	   FAIL	2, 	   FAIL	3, 	   FAIL	4, 	   FAIL	5, 	   FAIL	6, 	   FAIL	7, 	   FAIL	8, 	   FAIL	9, 	   FAIL	10, 	   FAIL	11, 	   FAIL	12, 	   FAIL	13, 	   FAIL	14, 	   FAIL	15, 	   FAIL	16, 	   FAIL	17, 	   FAIL	18, 	   FAIL	19, 	   FAIL	20, 	   FAIL	21, 	   FAIL	22, 	   FAIL	23, 	   FAIL	24, 	   FAIL	25, 	   FAIL	26, 	   FAIL	27, 	   FAIL	28, 	   FAIL	29, 	   FAIL	30, 	   FAIL	31, 	   FAIL	32, 	   FAIL	33, 	   FAIL	34, 	   FAIL	35, 	   FAIL	36, 	   FAIL	37, 	   FAIL	38, 	   FAIL	39, 	   FAIL	40, 	   FAIL	41, 	   FAIL	42, 	   FAIL	43, 	   FAIL	44, 	   FAIL	45, 	   FAIL	46, 	   FAIL	47, 	   FAIL	48, 	   FAIL	49, 	   FAIL	50, 	   FAIL	51, 	   FAIL	52, 	   FAIL	53, 	   FAIL	54, 	   FAIL	55, 	   FAIL	56, 	   FAIL	57, 	   FAIL	58, 	   FAIL	59, 	   FAIL	60, 	   FAIL	61, 	   FAIL	62, 	   FAIL	63, 	   FAIL	64, 	   FAIL	65, 	   FAIL	66, 	   FAIL	67, 	   FAIL	68, 	   FAIL	69, 	   FAIL	70, 	   FAIL	71, 	   FAIL	72, 	   FAIL	73, 	   FAIL	74, 	   FAIL	75, 	   FAIL	76, 	   FAIL	77, 	   FAIL	78, 	   FAIL	79, 	   FAIL	80, 	   FAIL	81, 	   FAIL	82, 	   FAIL	83, 	   FAIL	84, 	   FAIL	85, 	   FAIL	86, 	   FAIL	87, 	   FAIL	88, 	   FAIL	89, 	   FAIL	90, 	   FAIL	91, 	   FAIL	92, 	   FAIL	93, 	   FAIL	94, 	   FAIL	95, 	   FAIL	96, 	   FAIL	97, 	   FAIL	98, 	   FAIL	99, 	   FAIL	100, 	   FAIL	101, 	   FAIL	102, 	   FAIL	103, 	   FAIL	104, 	   FAIL	105, 	   FAIL	106, 	   FAIL	107, 	   FAIL	108, 	   FAIL	109, 	   FAIL	110, 	   FAIL	111, 	   FAIL	112, 	   FAIL	113, 	   FAIL	114, 	   FAIL	115, 	   FAIL	116, 	   FAIL	117, 	   FAIL	118, 	   FAIL	119, 	   FAIL	120, 	   FAIL	121, 	   FAIL	122, 	   FAIL	123, 	   FAIL	124, 	   FAIL	125, 	   FAIL	126, 	   FAIL	127, 	   FAIL	128, 	   FAIL	129, 	   FAIL	130, 	   FAIL	131, 	   FAIL	132, 	   FAIL	133, 	   FAIL	134, 	   FAIL	135, 	   FAIL	136, 	   FAIL	137, 	   FAIL	138, 	   FAIL	139, 	   FAIL	140, 	   FAIL	141, 	   FAIL	142, 	   FAIL	143, 	   FAIL	144, 	   FAIL	145, 	   FAIL	146, 	   FAIL	147, 	   FAIL	148, 	   FAIL	149, 	   FAIL	150, 	   FAIL	151, 	   FAIL	152, 	   FAIL	153, 	   FAIL	154, 	   FAIL	155, 	   FAIL	156, 	   FAIL	157, 	   FAIL	158, 	   FAIL	159, 	   FAIL	160, 	   FAIL	161, 	   FAIL	162, 	   FAIL	163, 	   FAIL	164, 	   FAIL	165, 	   FAIL	166, 	   FAIL	167, 	   FAIL	168, 	   FAIL	169, 	   FAIL	170, 	   FAIL	171, 	   FAIL	172, 	   FAIL	173, 	   FAIL	174, 	   FAIL	175, 	   FAIL	176, 	   FAIL	177, 	   FAIL	178, 	   FAIL	179, 	   FAIL	180, 	   FAIL	181, 	   FAIL	182, 	   FAIL	183, 	   FAIL	184, 	   FAIL	185, 	   FAIL	186, 	   FAIL	187, 	   FAIL	188, 	   FAIL	189, 	   FAIL	190, 	   FAIL	191, 	   FAIL	192, 	   FAIL	193, 	   FAIL	194, 	   FAIL	195, 	   FAIL	196, 	   FAIL	197, 	   FAIL	198, 	   FAIL	199, 	   FAIL	200, 	   FAIL	201, 	   FAIL	202, 	   FAIL	203, 	   FAIL	204, 	   FAIL	205, 	   FAIL	206, 	   FAIL	207, 	   FAIL	208, 	   FAIL	209, 	   FAIL	210, 	   FAIL	211, 	   FAIL	212, 	   FAIL	213, 	   FAIL	214, 	   FAIL	215, 	   FAIL	216, 	   FAIL	217, 	   FAIL	218, 	   FAIL	219, 	   FAIL	220, 	   FAIL	221, 	   FAIL	222, 	   FAIL	223, 	   FAIL	224, 	   FAIL	225, 	   FAIL	226, 	   FAIL	227, 	   FAIL	228, 	   FAIL	229, 	   FAIL	230, 	   FAIL	231, 	   FAIL	232, 	   FAIL	233, 	   FAIL	234, 	   FAIL	235, 	   FAIL	236, 	   FAIL	237, 	   FAIL	238, 	   FAIL	239, 	   FAIL	240, 	   FAIL	241, 	   FAIL	242, 	   FAIL	243, 	   FAIL	244, 	   FAIL	245, 	   FAIL	246, 	   FAIL	247, 	   FAIL	248, 	   FAIL	249, 	   FAIL	250, 	   FAIL	251, 	   FAIL	252, 	   FAIL	253, 	   FAIL	254, 	   PASS			0 us			436 us
Error code: 0x0A
			String			   FAIL			0 us			88 us
-----------------------------------------------------------------------------------------------------------------------------

```

Marzogh, 2019-04-19:
Will take a look at this ASAP

Marzogh, 2019-06-01:
I am unable to replicate this issue on my Arduino Mega with the W25Q80BV, running on macOS 10.14.5 and Arduino IDE 1.8.9, using the latest stable code of the library (the stable branch) and will therefore consider this issue closed. Please let me know if you face any issues with the upcoming release.

closed 2019-06-01 by Marzogh