Marzogh/SPIMemory #159: WriteWord not working property

rafacabelo, 2018-11-28:
[image] [image]
[image] [image] [image]
[image] [image]
###### ---------------------------- **DO NOT DELETE OR EDIT** anything above this line ----------------------------
###### ---------------------------- **DO NOT DELETE OR EDIT** anything above this line ----------------------------

Hey there! Thanks for using the SPIFlash library for Arduino.
### Please note that starting 01.03.2018 any issue raised here MUST be submitted according to this template or it will be flagged with 'Not enough information'. No action will be taken till all the prerequisite information is provided. If no information is provided for over a month, the issue will be closed.

**Note: For support questions, please use the [Arduino Forums](http://forum.arduino.cc/index.php?topic=324009.0). This repository's issues are reserved for feature requests and bug reports.**

# Issue details

**I'm submitting a ...**
  - [x] bug report
  - [ ] feature request
  - [ ] support request => Please do not submit support request here, see note at the top of this template.

Do the checklist before filing an issue:

- [ ] Is this something you can **debug and fix**? Send a pull request! Bug fixes and documentation fixes are welcome.
- [ ] Is this an idea for a feature? Post the feature request as an issue here with a title that clearly says "Feature Request".
- [x] Is this a bug that you cannot fix? Post the bug report as an issue here with a title that cleary says "Bug Report".

When opening an issue please include the following details:
-------------------------------------------------------------

- [x ] Do you want to request a *feature* or report a *bug*?
bug
- [ x] Which library version are you using?
3.2.1
- [x ] Which Arduino IDE version are you compiling the library on?
1.8.5
- [ x] Which OS are you compiling your code on? [Linux / Mac / Windows]
windows 8.1
- [x ] Which microcontroller platform are you using?
atmega328p
- [x ] Which Flash memory module are you using?
W25Q64FVSSIG
- [ ] What is the current behavior?
- [ ] What is the expected behavior?

--------------------------
###### Bug reports only
- If this is a bug report -

- [ ] Make sure you have run FlashDiagnostics.ino with ``` #define RUNDIAGNOSTICS ``` uncommented in SPIFlash.h. **List any error codes** that pop up in your Serial output when you run FlashDiagnostics.ino.here:
      Error codes
      ------------
Initialising..........
Chip Diagnostics initiated.

No Chip size defined by user. Automated identification initiated.
Chip identified. This chip is fully supported by the library.

SPIMemory Library version: 3.2.1

JEDEC ID: 0xEF4017
Man ID: 0xEF
Memory ID: 0x40
Capacity: 8388608
Max Pages: 4294967295
Unique ID: 3341979874572544, 0x6274224353453900
-----------------------------------------------------------------------------------------------------------------------------
							Testing library code
-----------------------------------------------------------------------------------------------------------------------------
			Function		Test result			     Runtime
-----------------------------------------------------------------------------------------------------------------------------
			Power Down		   FAIL			Not all chips support power down. Check your datasheet.
			Power Up		   PASS				      80 us

			Erase Chip		   PASS				      21.400 s
			Erase 72KB		   PASS				      208.94 ms
			Erase 64KB		   PASS				      129.05 ms
			Erase 32KB		   PASS				      88.70 ms
			Erase 4KB		   PASS				      51.32 ms
-----------------------------------------------------------------------------------------------------------------------------
			Data type		I/O Result	      Write time	      Read time
-----------------------------------------------------------------------------------------------------------------------------
			Byte			   PASS			200 us			68 us
			Char			   PASS			196 us			68 us
			Word			   PASS			204 us			68 us
			Short			   PASS			200 us			68 us
			ULong			   PASS			220 us			72 us
			Long			   PASS			212 us			72 us
			Float			   PASS			220 us			72 us
			Struct			   PASS			328 us			104 us
			Byte Array		   PASS			2.78 ms			448 us
Error code: 0x06
			String			   FAIL			0 us			88 us
-----------------------------------------------------------------------------------------------------------------------------

- [ ] If you have a problem with a particular function, call the flash.error() function (after you have made sure you have started up your Serial port with a ``` Serial.begin(BAUD) ``` ). Provide details of the function, the data given to/ expected from the function and the error code here: (**Please repeat this for every function you have an error with***)
        Function -
        Data -
        Error code -
- [x] Provide a **minimal code snippet** example that reproduces the bug. Please make sure you wrap any code in the proper code blocks like below
```
#include<SPIMemory.h>
SPIFlash flash(10); // csPin

void setup()
{
  Serial.begin(250000);

  flash.begin(); // auto detects chip used // 2502/484

  Serial.println(flash.getCapacity());

  uint16_t previous_millis = millis();
  uint8_t count = 0;

  flash.eraseSector(0);

  for (uint32_t k = 0; k < 4096; k++)
  {
	flash.writeWord(k, 345);
	flash.error();
	Serial.print(flash.error());
	count++;
	if (count >= 52)
	{
		Serial.println();
		count = 0;
	}
	k++;
  }
  Serial.println();
  
  count = 0;
  for (uint32_t k = 0; k < 4096; k++)
  {
    Serial.print(flash.readWord(k));
	 flash.error();
	 Serial.print(";");
	 Serial.print(flash.error());
    Serial.print(";");
    count++;
	 count++;
	 k++;
    if (count >= 52)
    {
      Serial.println();
		count = 0;
    }
  }
  
  Serial.println();
  Serial.print(millis() - previous_millis);
  Serial.println(" ok");

}

void loop()
{

}
```
- [ ] Provide **screenshots** where appropriate
- [ ] Provide **wiring diagram** where appropriate

-----------------------------
###### Feature Requests only

- [ ] If this is a feature request, what is the motivation / use case for changing the behavior?

----------------------------

- [x ] Other information (e.g. detailed explanation, related issues, suggestions how to fix, links for us to have context etc.)

The issue is related to WriteWord function. When I use this funciton the first byte is not write. This happen intermitently but is more often to couple address. Per my code example the data 345 should be written as 89, 1 but the function is having trouble to write the byte 1. Its writes 89, 255. Sometimes 89, 0.

_Make sure to add **all the information needed to understand the bug/feature** so that someone can help. If the info is missing we'll add the 'Needs more information' label and close the issue until there is enough information._

Marzogh, 2019-06-02:
I am unable to replicate this error with > v3.3.0 code. Unless you find this happening again, I will consider this issues closed. 🙂

closed 2019-06-02 by Marzogh

rafacabelo, 2020-06-03:
After a year of troubleshooting, I have isolated the issue to the voltage level at WRITE pin dropping bellow the datasheet value to be considerated as HIGH. As expected only during the write of second byte the issue would occur. Adding a 10uF cap at VCC solve the issue. Thanks!