Marzogh/SPIMemory #161: Support for more chips

thagh05t, 2019-02-01:
[image] [image]
[image] [image] [image]
[image] [image]
###### ---------------------------- **DO NOT DELETE OR EDIT** anything above this line ----------------------------
###### ---------------------------- **DO NOT DELETE OR EDIT** anything above this line ----------------------------

Hey, I am looking to help support other flash chips. I use a lot of Micronix and notice you do not have support for them here so would like to contribute.

What do I need to do in order to help get more chips supported?

Marzogh, 2019-04-19:
Hi mate,
Thanks for the offer to support more chips, that'd be fantastic! I'd recommend that you look through the library (the code in the development branch specifically) and adapt it to make it work with the chips you use. Once you have it working, submit a pull request against the development branch. I'll make sure your changes haven't broken anything pre-existing and roll the changes into the next release 🙂 

Cheers!

P.S. My apologies for the late reply. I was away for a little around a year due to health issues, but I'm back now. 😄

closed 2019-04-19 by Marzogh

thagh05t, 2019-04-19:
Hey,

Glad to hear from you. I am not too great at C++ but I have been working on
a few things. I like that this project can help others interface with
popular EEPROM fairly Easy though. 25 series chips are super popular and I
have been working with them for quite some time. Since there was no recent
development I decided to start working on my own 25 series library though.
I probably do not know as much as you do on the subject, but it appears I
am able to write to most popular brands of most sizes so far. I have also
implemented an xmodem upload/download feature.

If you are interested in a bit of help furthering what you already have,
please let me know I am on board. I would like to touch base with you on
best practices though, as it seams you have testing and a process flow much
different than mine. Also, I have cataloged most SPI chips into a few
formats that can be consumed as a quick lookup based on JEDEC ID, this cuts
down on identification and operational compatibility, but is a trade-off
with space. I am not currently using it, but I have added it to an API I
have written for UI's to use when interfacing with the devices I have been
working on.