Marzogh/SPIMemory #163: Automated capacity recognition improvement

LoicGRENON, 2019-03-17:
Hey !

In wanting to use this library with an SST26VF016B memory chip, I had to look deep into how the memory capacity automated recognition works to add support for this chip.
I noticed that you use for this the Device ID from the JEDEC-ID Read instruction.

I think this is not enough since components from different manufacturers can have the same device ID.
We should also use manufacturer ID and Device type to identify the chip (The whole JEDEC-ID responses values, in fact).

https://github.com/Marzogh/SPIMemory/blob/341c29a7ebe31339726da035b3a62bd5088f0a55/src/SPIFlash.h#L197

For that, the _capID array can be of type uint32_t with values like 0xAABBCCDD where :

- AA will be a reserved byte with 00 value,
- BB will be the manufacturer ID,
- CC will be the device type,
- DD will be the device ID

Obviously, the line 566 of SPIFlashIO.cpp should also be modified accordingly.
https://github.com/Marzogh/SPIMemory/blob/341c29a7ebe31339726da035b3a62bd5088f0a55/src/SPIFlashIO.cpp#L566

I can do this modification and make a pull request but I would like to have your opinion before that.

Regards,
Loïc

Marzogh, 2019-04-19:
Hi Loïc,

Thank you for offering to do this. I would greatly appreciate any help and be happy to take you up on it. 🙂 

My apologies for the late reply. I've been away for the last year due to health issues - but I'm back now.

Cheers!

Marzogh, 2019-06-03:
I've started the process of cleaning this up as of commit 7367cef . Should have a cleaner and faster system in place for v3.5.0

Re: #181

Marzogh, 2019-06-03:
Final update to this issue 🙂  @LoicGRENON 

`_chipID()` in `flash.begin()` has been completely re-written from scratch by commit 32341cc. In previous versions of the library, this function took between **200 to 250 ms** to execute. It now takes **450 to 500 µS** (i.e. it is ~500x faster).

commit 32341ccd (2019-06-03): Updated change log and speed comparisons doc. Fixes #163 Resolves #181
closed 2019-06-03 by Marzogh