Marzogh/SPIMemory #184: Bug report - *W25Q64FV writing random characters *

03sarath, 2019-06-23:
Hey,

This is Sarath, Thanks for your work.
So I have issues in using W25Q64FV, I'm using TestFlash example code, I'm trying to write a string to a particular address, but it is writing random characters along with string.

For Example:
I'm trying to Write:  

{"L":"FRONT","I":3,"C":0,"Q":"1.4 Hey this is Device A from Delhi Second ID 1"}

But getting written:  

{"L":"FRONT","I"⸮⸮⸮⸮C":0,"Q":"1.4 Hey this is Device A from Delhi Second ID 1"}

I really don't know why this random "⸮⸮" characters are getting written.

Marzogh, 2019-06-24:
Hi SArath, could you please give me all the details that I've requested be filled in when filing a bug report? They should've turned up pre-filled when you created a new issue. I've pasted a copy below for you to use. Without this information, I spend a lot of time asking you questions for basic information.

Cheers!

<hr>

##### Any bug report raised here MUST be submitted according to this template or it will be flagged with 'Not enough information'. No action will be taken till all the prerequisite information is provided. If no information is provided for over a month after the 'Not enough information' label is applied, the issue will be closed.

## Pre-bug report checklist:

Do this checklist before filing an bug report:
- [ ] Is this something you can **debug and fix**? Send a pull request! Bug fixes and documentation fixes are welcome.
- [ ] Is this an idea for a feature? Post it as a Feature request. NOT a bug report.
- [ ] Is this a bug that you cannot fix? Go ahead with filing a bug report below

<hr>

## Bug Report

#### Describe the bug

_Include a clear and concise description of what the bug is._

- Make sure you have run FlashDiagnostics.ino with ``` #define RUNDIAGNOSTICS ``` uncommented in SPIFlash.h. Paste the **relevant sections** of the output from your Serial console when you run FlashDiagnostics.ino.here:
```
```FlashDiagnostics output```
```
- If you have a problem with a particular function, call the `flash.error()` function (after you have made sure you have started up your Serial port with a ``` Serial.begin(BAUD) ``` ). Provide details of the function, the data given to/ expected from the function and the error code here: (**Please repeat this for every function you have an error with**)
        - Function: (e.g. writeByte())
       -  Data: (e.g. 3.14 or Struct)
        - Error code: (e.g. 0x0A)

#### To Reproduce
Provide a **minimal code snippet** example that reproduces the bug (If you're using one of the examples that came with the library, just tell us which one instead of pasting the entire example's code). Please make sure you wrap any code in the proper code blocks like below
```
```CODE HERE```
```

#### Expected behavior
A clear and concise description of what you expected to happen.

#### Screenshots
If applicable, add screenshots to help explain your problem.

#### Wiring diagram
If applicable, add a wiring diagram to help explain your problem.

#### Software environment (please complete the following information):
 - Library version: [e.g. v3.3.0]
 - Arduino IDE version: [e.g. 1.8.9]
 - OS: [e.g. Linux]
 - Micro controller platform: [e.g. Feather M0]
 - Flash/Fram memory module: [e.g. SST25VF064C]

#### Additional context
Add any other context about the problem here.

<hr>

###### DO NOT DELETE OR EDIT anything below this

<hr>

<sub> <b> Note 1: _Make sure to add **all the information needed to understand the bug** so that someone can help. If any essential information is missing we'll add the 'Needs more information' label and close the issue until there is enough information._ </b></sub>

<sub> <b> Note 2: For support questions (for example, tutorials on how to use the library), please use the [Arduino Forums](http://forum.arduino.cc/index.php?topic=324009.0). This repository's issues are reserved for feature requests and bug reports. </b></sub>

<hr>

[image] [image] [image] [image] [image] [image] [image]

03sarath, 2019-06-24:
Hey @Marzogh Marzogh.

Sorry, It's my bad I have skipped your form. Here I have an atteched bug report as per your requirement. Thanks in advance.

### **Pre-bug report checklist:**

Do this checklist before filing an bug report:

 **Is this a bug that you cannot fix? Go ahead with filing a bug report below**

### **Bug Report**

_**Describe the bug**_
So I have issues in using **W25Q64FV** Winbond, I'm using **TestFlash** example code, I'm trying to write a string to a particular address(i.e to address 4500), but some characters are missed while writing.

**For Example:**
_I'm trying to Write:_

**{"L":"FRONT","I":3,"C":0,"Q":"1.4 Hey this is Device A from Delhi Second ID 1"}**

_But getting written: Some characters are missed while writing_

**{"L":"FRONT","I"⸮⸮⸮⸮C":0,"Q":"1.4 Hey this is Device A from Delhi Second ID 1"}**

I really don't know why some characters are missed while writing,```⸮⸮⸮``` are getting added instead. But some times it can able to write without missing any characters, but it fails much time. I have Installed the latest SPIFlash library from Arduino IDE. But when I'm running FlashDiagnostics.ino it displays ```SPIMemory Library version: < 2.5.0``` in my Serial monitor console, I have attached screenshots for your reference

- Make sure you have run FlashDiagnostics.ino with ```#define RUNDIAGNOSTICS``` uncommented in SPIFlash.h. Paste the relevant sections of the output from your Serial console when you run FlashDiagnostics.ino.here:

**_Note: But I can find ```#define RUNDIAGNOSTICS``` in SPIMemory.h file_**
Here I have attached Flash Diagnostics report from the serial monitor console.

``` Initialising..........
Chip Diagnostics initiated.

No Chip size defined by user. Checking library support.
Chip identified. This chip is fully supported by the library.

SPIMemory Library version: < 2.5.0

JEDEC ID: 0xEF4017
Man ID: 0xEF
Memory ID: 0x40
Capacity: 8388608
Max Pages: 32768
Unique ID: 1999905419506688, 0x64486367441D3400
-----------------------------------------------------------------------------------------------------------------------------
							Testing library code
-----------------------------------------------------------------------------------------------------------------------------
			Function		Test result			     Runtime
-----------------------------------------------------------------------------------------------------------------------------
			Power Down		   PASS				      55 us
			Power Up		   PASS				      47 us

			Erase Chip		   PASS				      19.373 s
			Erase 72KB		   PASS				      221.48 ms
			Erase 64KB		   PASS				      151.78 ms
			Erase 32KB		   PASS				      96.94 ms
			Erase 4KB		   PASS				      49.55 ms
-----------------------------------------------------------------------------------------------------------------------------
			Data type		I/O Result	      Write time	      Read time
-----------------------------------------------------------------------------------------------------------------------------
			Byte			   PASS			112 us			52 us
			Char			   PASS			120 us			48 us
			Word			   PASS			140 us			56 us
			Short			   PASS			140 us			56 us
			ULong			   PASS			152 us			60 us
			Long			   PASS			144 us			56 us
			Float			   PASS			156 us			56 us
			Struct			   PASS			324 us			108 us
			Byte Array		   PASS			2.48 ms			548 us
			String			   PASS			460 us			176 us
-----------------------------------------------------------------------------------------------------------------------------
```

**To Reproduce**
Provide a **minimal code snippet** example that reproduces the bug (If you're using one of the examples that came with the library, just tell us which one instead of pasting the entire example's code). Please make sure you wrap any code in the proper code blocks like below

Note: I'm using ```TestFlash.ino``` from example code without any changes, Trying to use ```writeStr``` to write a string to a particular address, basically, I'm trying to write JSON string (I.e ```{"L":"FRONT","I":3,"C":0,"Q":"1.4 Hey this is Device A from Delhi Second ID 1"}``` but some characters are misses while writing i.e  ``` {"L":"FRONT","I"⸮⸮⸮⸮C":0,"Q":"1.4 Hey this is Device A from Delhi Second ID 1"}```   )

**Expected behavior**

Trying to write a string JSON i.e ```{"L":"FRONT","I":3,"C":0,"Q":"1.4 Hey this is Device A from Delhi Second ID 1"}``` without missing characters.

**Screenshots**

Using ```TestFlash.ino``` example from laibrary.
**Step 1:**
[image]

Step2:
Accessing Function 6 to write a string to a particular address
[image]

Step 3:
Assigning address 4500 to write a string.
[image]

Step 4:
Writing the JSON string i.e ```{"Loc":"BACK","SL:NO":1.3,"ImgID":3,"CON":"OPEN CONDITION","Qtion":"Hey this is Device A from location Delhi ID 24A612334"}``` to address 4500

[image]

Step 5:
Result: ```Data Write Failed```
[image]

Step 6:
Accessing address 4500 by using function **7 Read String** to check.
**Result:** Missing Character has been found.
[image]

**Software environment (please complete the following information):**
Library version: [e.g. v3.4.0]  _Note:Installed v3.4.0 but it displayed 2.5.0 in serial moniter_
Arduino IDE version: [e.g. 1.8.9]
OS: [e.g. Windows 8.1,64 bit]
Micro controller platform: [e.g. SAMD21 (i.e ARDUINO MKR WAN 1300)]
Flash/Fram memory module: [e.g. W25Q64FV]

Marzogh, 2019-06-25:
Cheers! Thank you for the update with the information. I will look into this over the next couple of days and get back to you asap. :)

Marzogh, 2019-07-13:
Hi Sarath, thanks again for catching this one - it was page boundary error that seemed to crop up under very specific circumstances. I've fixed it in the code in the development branch and will push it stable once I finish testing in an hour or so. The bug fix will be rolled out officially when v3.5.0 is released in August. In the meantime, please feel free to use the code from the stable branch. 🙂 Cheers!

closed 2019-07-13 by Marzogh

03sarath, 2019-11-06:
Hey @Marzogh,

I hope you doing, I couldn't update from a long time, We got the issue again, I have tried the code from a stable branch [https://github.com/Marzogh/SPIMemory/tree/0318bbcb1146aeb9f3012b74e0a09d5d6ba82a6e](url)

According to the example` readWriteString.ino `

I tried to write a string that you have in line:49 `String inputString = R"({"L":"BACK","S":1.3,"I":3,"C":"OPEN CONDITION","Q":"Hey, this is Device A from location Delhi ID 24A612334"})";`, it is successfully writing to address 4500, But when I try to write the same string to address 4900 & other address, it's not writing the string.

**Here I have attached the screenshot.**
[image]

**readWriteString.ino** 

```
/*
  |~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~|
  |                                                             readWriteString.ino                                                               |
  |                                                               SPIMemory library                                                                |
  |                                                                   v 3.5.0                                                                     |
  |~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~|
  |                                                                    Marzogh                                                                    |
  |                                                                  13.07.2019                                                                   |
  |~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~|
  |                                                                                                                                               |
  |                        This program shows the method of reading a string from the console and saving it to flash memory                       |
  |                                                                                                                                               |
  |~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~|
*/
#include<SPIMemory.h>

uint32_t strAddr;

#if defined(ARDUINO_SAMD_ZERO) && defined(SERIAL_PORT_USBVIRTUAL)
// Required for Serial on Zero based boards
#define Serial SERIAL_PORT_USBVIRTUAL
#endif

#if defined (SIMBLEE)
#define BAUD_RATE 250000
#define RANDPIN 1
#else
#define BAUD_RATE 115200
#define RANDPIN A0
#endif

//SPIFlash flash(SS1, &SPI1);       //Use this constructor if using an SPI bus other than the default SPI. Only works with chips with more than one hardware SPI bus
SPIFlash flash(11);

bool readSerialStr(String &inputStr);

void setup() {
  Serial.begin(BAUD_RATE);
#if defined (ARDUINO_ARCH_SAMD) || (__AVR_ATmega32U4__) || defined(ARCH_STM32) || defined(NRF5)
  while (!Serial) ; // Wait for Serial monitor to open
#endif

  flash.begin();

   randomSeed(analogRead(RANDPIN));
  //strAddr = random(0, flash.getCapacity());
  strAddr = 4900;
  //String inputString = "This is a test String";
  String inputString = R"({"L":"BACK","S":1.3,"I":3,"C":"OPEN CONDITION","Q":"Hey, this is Device A from location Delhi ID 24A612334"})";
    //String inputString ="{\"L\":\"FRONT\",\"I\":1,\"C\":\"STOPPED CONDITION\",\"Q\":\" 1.2 Inspect the TELESCOPIC GUARDS are available\"}";
  //char inArray[6] = {'a', 'b', 'c', 'd', 'e', 'f'};
  //if (flash.writeCharArray(strAddr, inArray, 6)) {
  if (flash.writeStr(strAddr, inputString)) {
    Serial.print(F("Written data: "));
  }
  else {
    Serial.print(F("Unable to write data: "));
  }
  Serial.print(inputString);
  /*for (uint8_t i = 0; i < 6; i++) {
    Serial.print(inArray[i]);
    Serial.print(", ");
  }*/
  Serial.print(F(" to address: 0x"));
  Serial.println(strAddr, HEX);

  String outputString = "";
  char outArray[6];
  //if (flash.readCharArray(strAddr, outArray, 6)) {
  if (flash.readStr(strAddr, outputString)){
    Serial.print(F("Read data: "));
    Serial.print(outputString);
    /*for (uint8_t i = 0; i < 6; i++) {
      Serial.print(outArray[i]);
      Serial.print(", ");
    }*/
    Serial.print(F(" from address: 0x"));
    Serial.println(strAddr, HEX);
  }
  else {
    Serial.println("Unable to read String");
  }
  while (!flash.eraseSector(strAddr));
}

void loop() {

}

//Reads a string from Serial
bool readSerialStr(String &inputStr) {
  if (!Serial)
    Serial.begin(115200);
  while (Serial.available()) {
    inputStr = Serial.readStringUntil('\n');
    Serial.println(inputStr);
    return true;
  }
  return false;
}
```

Trying to figure out what's wrong with this, Let me know if I need to do much more experiments. :)

jpmeijers, 2019-11-29:
I think I am seeing the same issue. The test example also regularly fails on the last string test.

commit d5d75ea9 (2019-07-13): Formatting fixes and Fixes #184 page boundary error with writing long Strings.
commit 0318bbcb (2019-07-13): Merge pull request #185 from Marzogh/development

Fixes #184