Marzogh/SPIMemory #198: Bug report - HSPI Support for ESP32 does't seem to work

intensite, 2020-02-23:
##### Any bug report raised here MUST be submitted according to this template or it will be flagged with 'Not enough information'. No action will be taken till all the prerequisite information is provided. If no information is provided for over a month after the 'Not enough information' label is applied, the issue will be closed.

## Pre-bug report checklist:

Do this checklist before filing an bug report:
- [ ] Is this something you can **debug and fix**? Send a pull request! Bug fixes and documentation fixes are welcome.
- [ ] Is this an idea for a feature? Post it as a Feature request. NOT a bug report.
- [x] Is this a bug that you cannot fix? Go ahead with filing a bug report below

<hr>

## Bug Report

#### Describe the bug

The library doesn't work with HARDWARE SPI HSPI on a ESP32 chip with a W25Q64JV flash chip.
I have a custom made PCB which was designed to use the HSPI pins, if I can't make it work, the complete PCB batch will need to be trashed :-(

I am however able to run the  FlashDiagnostics test program when I use the VSPI Pins on a bread board, which confirms the Flash chip is detected and supported. (See output bellow)

**Default pins(IOMUX pins) for ESP32**
| SPI | MOSI | MISO | CLK | CS |
| ------------ | -------------| ------------ | -------------| ------------ |
| VSPI | GPIO 23 | GPIO 19 | GPIO 18 | GPIO 5 |
| HSPI | GPIO 13 | GPIO 12 | GPIO 14 | GPIO 15 |

I am calling the ```SPIFlash flash(15, new SPIClass(HSPI));``` Constructor as per documentation.

I have noticed that the in the function  ```bool SPIFlash::begin(uint32_t flashChipSize)```, the _nonStdSPI values are passed as -1 to the ```SPIClass::begin(int8_t sck, int8_t miso, int8_t mosi, int8_t ss)``` function.  This should not be a problem since this block of code should set the proper pin numbers:

```
if(sck == -1 && miso == -1 && mosi == -1 && ss == -1) {
        _sck = (_spi_num == VSPI) ? SCK : 14;
        _miso = (_spi_num == VSPI) ? MISO : 12;
        _mosi = (_spi_num == VSPI) ? MOSI : 13;
        _ss = (_spi_num == VSPI) ? SS : 15;
       
        Serial.println(_spi_num);  // I placed that here!
    } 
```

Un fortunately the _spi_num variable that I set to HSPI in the constructor gets reset to VSPI (from HSPI:2 to VSPI:3) and thus the pin assignment is not correct.

**The following is the output of FlashDiagnostics.ino when using HSPI**

```
15:51:07.903 -> Initialising..........
15:51:07.903 -> Chip Diagnostics initiated.
15:51:07.903 -> 
15:51:07.903 -> Inside SPIClass::begin
15:51:07.903 -> 3
15:51:07.903 -> Check your wiring. Flash chip is non-responsive.
15:51:07.903 -> If this does not help resolve/clarify this issue, please raise an issue at http://www.github.com/Marzogh/SPIMemory/issues with the details of what your were doing when this error occurred
15:51:07.938 -> Unable to identify chip. Are you sure this chip is supported?
15:51:07.938 -> If this does not help resolve/clarify this issue, please raise an issue at http://www.github.com/Marzogh/SPIMemory/issues with the details of what your were doing when this error occurred
15:51:07.938 -> 
15:51:07.938 -> SPIMemory Library version: < 2.5.0
15:51:07.938 -> 
15:51:07.938 -> No comms. Check wiring. Is chip supported? If unable to fix, raise an issue on Github```
```
**The following is the output of FlashDiagnostics.ino when using VSPI**
```
Initialising..........
Library instantiated in: 5.00 µS

Chip Diagnostics initiated.

Inside SPIClass::begin
3
No Chip size defined by user. Automatic ID initiated. 
SFDP ID not requested
Checking library support.
Chip identified. This chip is fully supported by the library.

SPIMemory Library version: 3.5.0

flash.begin() runtime: 13614.00 µS

JEDEC ID: 0xEF4017
Man ID: 0xEF
Memory ID: 0x40
Capacity: 8388608
Max Pages: 32768
Unique ID: 1986556344219648, 0x68C0C00331393800
-----------------------------------------------------------------------------------------------------------------------------
							Testing library code
-----------------------------------------------------------------------------------------------------------------------------
			Function		Test result			     Runtime
-----------------------------------------------------------------------------------------------------------------------------
			Power Down		   PASS				      3.03 ms
			Power Up		   PASS				      21 us

			Erase Chip		   PASS				      14.070 s
			Erase 72KB		   PASS				      211.82 ms
			Erase 64KB		   PASS				      159.68 ms
			Erase 32KB		   PASS				      112.42 ms
			Erase 4KB		   PASS				      36.62 ms
-----------------------------------------------------------------------------------------------------------------------------
			Data type		I/O Result	      Write time	      Read time
-----------------------------------------------------------------------------------------------------------------------------
			Byte			   PASS			6.05 ms			3.02 ms
			Char			   PASS			6.05 ms			3.02 ms
			Word			   PASS			6.05 ms			3.02 ms
			Short			   PASS			6.05 ms			3.02 ms
			ULong			   PASS			6.06 ms			3.02 ms
			Long			   PASS			6.06 ms			3.02 ms
			Float			   PASS			6.06 ms			3.02 ms
			Struct			   PASS			12.17 ms			3.05 ms
			Byte Array		   PASS			13.04 ms			3.33 ms
			String			   FAIL			0 us			3.09 ms
			Power Down		   PASS				      3.03 ms			Power Up		   PASS				      18 us
-----------------------------------------------------------------------------------------------------------------------------

```

#### To Reproduce
Provide a **minimal code snippet** example that reproduces the bug (If you're using one of the examples that came with the library, just tell us which one instead of pasting the entire example's code). Please make sure you wrap any code in the proper code blocks like below

```I am using the default FlashDiagnostics.ino example sketch```

#### Expected behavior
Hardware SPI should work sas well as VSPI using default pins(IOMUX pins) defined above.

#### Wiring diagram

[image]

#### Software environment (please complete the following information):
 - Library version: v3.4.0  (Also tried the DEV branch
 - Arduino IDE version: 1.8.10 and Visual STudio Code
 - OS: Windows 10
 - Micro controller platform: ESP32 DevKit V4 with Arduino core for the ESP32 V1.0.4
 - Flash/Fram memory module: W25Q64JV

#### Additional context
Add any other context about the problem here.

Thank you for your help!

<hr>

###### DO NOT DELETE OR EDIT anything below this

<hr>

<sub> <b> Note 1: _Make sure to add **all the information needed to understand the bug** so that someone can help. If any essential information is missing we'll add the 'Needs more information' label and close the issue until there is enough information._ </b></sub>

<sub> <b> Note 2: For support questions (for example, tutorials on how to use the library), please use the [Arduino Forums](http://forum.arduino.cc/index.php?topic=324009.0). This repository's issues are reserved for feature requests and bug reports. </b></sub>

<hr>

[image] [image] [image] [image] [image] [image] [image]

intensite, 2020-02-24:
### Update 2020-02-24

I found the following piece of code that does a test on SPI Flash for ESP32:
https://github.com/kriswiner/ESP32/blob/master/SPI/SPIFlash_ESP32.ino

This piece of code doesn't use any library except the default ```#include <SPI.h>```

To my surprise, after changing the line: 
```SPI.begin(18, 19, 23, 5); // sck, miso, mosi, ss (ss can be any GPIO)```  
to 
```SPI.begin(14, 12, 13, 15); // sck, miso, mosi, ss (ss can be any GPIO)```

The all the tests (Write / Read / Erase / Read again) worked flawlessly.

It is difficult for me to debug, as my only tool is **Serial.println**,  but I suspect the chip identification in _getJedecId (SPIFlashIO.cpp).  I might be totally wrong so I don't want to waste your time with false tracs. 

I will continue investigation on my side.

mariusburakovas, 2020-04-20:
Hello, 
quick fix is to change BEGIN_SPI in SPIFlash.cpp to SPI.begin(14, 12, 13, 15).

rtek1000, 2021-06-26:
It worked here ESP32(HSPI) + W25Q64:

[Attached file:]

```C++
#elif defined (ARDUINO_ARCH_SAMD) || defined (ARCH_STM32) || defined(ARDUINO_ARCH_ESP32)
SPIFlash::SPIFlash(uint8_t cs, SPIClass *spiinterface) {
#if defined (ARDUINO_ARCH_ESP32)
SPIClass SPI2(HSPI);
SPIClass SPI3(VSPI);
#endif

  _spi = spiinterface;  //Sets SPI interface - if no user selection is made, this defaults to SPI
  if (_spi == &SPI) {
    _SPIInUse = STDSPI;
#if defined (ARDUINO_ARCH_ESP32)
  } else if (_spi == &SPI2) { // ESP32: HSPI, SPI.begin(14, 12, 13, 15);
    _nonStdSPI.sck = 14;
    _nonStdSPI.miso = 12;
    _nonStdSPI.mosi = 13;
    _nonStdSPI.ss  = 15;
    _SPIInUse = ALTSPI;
  } else if (_spi == &SPI3) { // ESP32: VSPI, SPI.begin(18, 19, 23, 5);
    _nonStdSPI.sck = 18;
    _nonStdSPI.miso = 19;
    _nonStdSPI.mosi = 23;
    _nonStdSPI.ss  = 5;
    _SPIInUse = ALTSPI;
#endif
  } else {
    _SPIInUse = ALTSPI;
  }
  csPin = cs;
  pinMode(csPin, OUTPUT);
  CHIP_DESELECT
}
```

Usage (HSPI):
```C++
SPIClass SPI2(HSPI);

SPIFlash flash(15, &SPI2);       //Use this constructor if using an SPI bus other than the default SPI. Only works with chips with more than one hardware SPI bus

void setup() {
```

(Note: VSPI not tested)
Usage (VSPI):
```C++
SPIClass SPI3(VSPI);

SPIFlash flash(5, &SPI3);       //Use this constructor if using an SPI bus other than the default SPI. Only works with chips with more than one hardware SPI bus

void setup() {
```

REF.: https://github.com/espressif/arduino-esp32/issues/1219

REF.: https://blog.bachi.net/?p=10562

> ESP32 integrates four SPI peripherals.
> 
> SPI0 and SPI1 are used internally to access the ESP32’s attached flash memory and thus are currently not open to users. They share one signal bus via an arbiter.
> SPI2 and SPI3 are general purpose SPI controllers, sometimes referred to as HSPI and VSPI, respectively. They are open to users. SPI2 and SPI3 have independent signal buses with the same respective names. Each bus has three CS lines to drive up to three SPI slaves.

[SPIFlash.cpp.zip](https://github.com/Marzogh/SPIMemory/files/6720833/SPIFlash.cpp.zip)

issacarbrun, 2023-03-29:
I tried using the VSPI but withou success. Still trying figure out what's the problem