Marzogh/SPIMemory #199: Bug report - Not working by default with Teensy 4.0 and W25Q64JV

bleckers, 2020-03-01:
## Pre-bug report checklist:

Do this checklist before filing an bug report:
- [ ] Is this something you can **debug and fix**? Send a pull request! Bug fixes and documentation fixes are welcome.
- [ ] Is this an idea for a feature? Post it as a Feature request. NOT a bug report.
- [X] Is this a bug that you cannot fix? Go ahead with filing a bug report below

<hr>

## Bug Report

#### SPI transaction in this library doesn't doesn't perform correctly with Teensy 4.0

Chip doesn't identify and flash operations fail when using the Teensy 4.0 and W25Q64JV. However this same setup does work with the SerialFlash library - https://github.com/PaulStoffregen/SerialFlash/

Output from SPIMemory:
```
```

Initialising..........

Chip Diagnostics initiated.

Unable to identify chip. Are you sure this chip is supported?

If this does not help resolve/clarify this issue, please raise an issue at http://www.github.com/Marzogh/SPIMemory/issues with the details of what your were doing when this error occurred

SPIMemory Library version: < 2.5.0

JEDEC ID: 0xF7200B

Man ID: 0xF7

Memory ID: 0x20

Capacity: 0

Max Pages: 0

Unique ID: 2085529283271680, 0x343E1BC720191C00

-----------------------------------------------------------------------------------------------------------------------------

							Testing library code

-----------------------------------------------------------------------------------------------------------------------------

			Function		Test result			     Runtime

-----------------------------------------------------------------------------------------------------------------------------

			Erase Chip	Unable to Enable Writing to chip.

Please make sure the HOLD & WRITEPROTECT pins are pulled up to VCC

If this does not help resolve/clarify this issue, please raise an issue at http://www.github.com/Marzogh/SPIMemory/issues with the details of what your were doing when this error occurred

	   FAIL	

			Erase 72KB	*constructor_of_choice*.begin() was not called in void setup()

	   FAIL	

			Erase 64KB	*constructor_of_choice*.begin() was not called in void setup()

	   FAIL	

			Erase 32KB	*constructor_of_choice*.begin() was not called in void setup()

	   FAIL	
[... 215 lines cut ...]

			Struct			   PASS			84 us			13 us

			Byte Array		   PASS			647 us			85 us

			String			   PASS			124 us			20 us

-----------------------------------------------------------------------------------------------------------------------------

```
```

<hr>

###### DO NOT DELETE OR EDIT anything below this

<hr>

<sub> <b> Note 1: _Make sure to add **all the information needed to understand the bug** so that someone can help. If any essential information is missing we'll add the 'Needs more information' label and close the issue until there is enough information._ </b></sub>

<sub> <b> Note 2: For support questions (for example, tutorials on how to use the library), please use the [Arduino Forums](http://forum.arduino.cc/index.php?topic=324009.0). This repository's issues are reserved for feature requests and bug reports. </b></sub>

<hr>

[image] [image] [image] [image] [image] [image] [image]

intensite, 2020-03-01:
I am experiencing similar issue with ESP32 and the exact same flash chip.  It also fail at the chip identification.  However, it passes the tests with SerialFlash.

After some investigation, it looks as if all functions of esp32-hal-spi that uses the *_spi member property of the SPIClass class fails or return 0 with something like:
```
    if(!spi) {
        return 0;
    }
```
Since I don't have a JTAG device, it is difficult to troubleshoot with Serial.print and without watch and breakpoints.  I have filed issue #198 which might be related.

bleckers, 2020-03-01:
Does running flash.setClock(50000000); before flash.begin(); solve your issues intensite?

intensite, 2020-03-01:
Thanks but nope.  I tried almost every increments between 10,000,000 and 104,000,000 without success.  I spent the best part of the week trying to make it work.  I am afraid I have reached the end of my rope.  I hope to hear from the author soon.